Koenkk/zigbee2mqtt #27236: zb-gw03 - error loop c20251a8bd7e

kotborealis, 2025-04-27:
### What happened?

Sometimes zigbee2mqtt gets error `c20251a8bd7e` from ezsp and goes into loop.
The only way to make it work is to restart zigbee2mqtt.
```
zigbee2mqtt  | [2025-04-26 23:51:05] ESC[34mdebugESC[39m:       zh:ezsp:uart: <-- [c20251a8bd7ec20251a8bd7ec20251a8bd7e]
zigbee2mqtt  | [2025-04-26 23:51:05] ESC[34mdebugESC[39m:       zh:ezsp:uart: <-- Error c20251a8bd7e
zigbee2mqtt  | [2025-04-26 23:51:05] ESC[34mdebugESC[39m:       zh:ezsp:uart: Uart reseting
zigbee2mqtt  | [2025-04-26 23:51:05] ESC[34mdebugESC[39m:       zh:ezsp:uart: --> Write reset
zigbee2mqtt  | [2025-04-26 23:51:05] ESC[34mdebugESC[39m:       zh:ezsp:uart: --> [1ac038bc7e]
zigbee2mqtt  | [2025-04-26 23:51:05] ESC[34mdebugESC[39m:       zh:ezsp:uart: -?- waiting reset
zigbee2mqtt  | [2025-04-26 23:51:05] ESC[34mdebugESC[39m:       zh:ezsp:uart: <-- Error c20251a8bd7e
zigbee2mqtt  | [2025-04-26 23:51:05] ESC[34mdebugESC[39m:       zh:ezsp:uart: Uart reseting
zigbee2mqtt  | [2025-04-26 23:51:05] ESC[34mdebugESC[39m:       zh:ezsp:uart: --> Write reset
zigbee2mqtt  | [2025-04-26 23:51:05] ESC[34mdebugESC[39m:       zh:ezsp:uart: --> [1ac038bc7e]
zigbee2mqtt  | [2025-04-26 23:51:05] ESC[34mdebugESC[39m:       zh:ezsp:uart: -?- waiting reset
zigbee2mqtt  | [2025-04-26 23:51:05] ESC[34mdebugESC[39m:       zh:ezsp:uart: <-- Error c20251a8bd7e
zigbee2mqtt  | [2025-04-26 23:51:05] ESC[34mdebugESC[39m:       zh:ezsp:uart: Uart reseting
zigbee2mqtt  | [2025-04-26 23:51:05] ESC[34mdebugESC[39m:       zh:ezsp:uart: --> Write reset
zigbee2mqtt  | [2025-04-26 23:51:05] ESC[34mdebugESC[
```

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

6.7.9.0 build 405

### Adapter

EZSP v8

### Setup

zigbee2mqtt running in a docker container, x86-64.
Zigbee hub is attached using uart-tcp socket - it's a zb-gw03.

Koenkk, 2025-04-27:
Assuming this device is similar to zb-gw04, it might be due to instability of the UART chip: https://github.com/Koenkk/zigbee2mqtt.io/pull/3598#issuecomment-2727789186 (this is also the reason why zb-gw04 is not recommended). Not something that is fixable by Zigbee2MQTT.

kotborealis, 2025-04-27:
It was very much stable until I've added some malfunctioning device or updated z2m - both happened roughly at the same time and I cannot tell what exactly triggered this behavior. 

Also I believe I'm not using the UART chip - zb-gw03 is a standalone gateway, flashed with tasmota and connected to the LAN.

@Koenkk, I want to tinker myself with the ezsp, to try and fix this. Could you give me some pointers - like where to look for the handling of the mentioned error?

Koenkk, 2025-04-27:
Regarding the `ezsp` driver:

> You are using the old `ezsp` driver, it's recommended to use the new `ember` drivers ([info](https://github.com/Koenkk/zigbee2mqtt/discussions/21462))

kotborealis, 2025-04-27:
To use it I would need to upgrade coordinator firmware. Seems risky but I'll give it a shot

kotborealis, 2025-05-01:
For history: upgraded zb-gw03 using [this guide](https://github.com/Koenkk/zigbee2mqtt/discussions/22546#discussioncomment-10654027), switched to ember driver. 
If network stabilizes I'll close this issue.

kotborealis, 2025-05-04:
Update: after a few days with the ember driver network stabilized.

closed 2025-05-04 by kotborealis