Koenkk/zigbee2mqtt #27238: [Wrong device]: Tuya IH-K663 detected as Loginovo ZG-101ZL

ellnic, 2025-04-27:
### Link

https://www.zigbee2mqtt.io/devices/IH-K663.html

### Model

IH-K663

### Description

Smart Button

### Vendor

Tuya

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/IH-K663.png

### Database entry

{"id":18,"type":"EndDevice","ieeeAddr":"0xa4c138cde5e4c37e","nwkAddr":43671,"manufId":4742,"manufName":"_TZ3000_ja5osu5g","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,1,3,4,6,4096,57345],"outClusterList":[25,10,3,4,6,8,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TS004F","manufacturerName":"_TZ3000_ja5osu5g","powerSource":3,"zclVersion":3,"appVersion":145,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017"}},"genOnOff":{"attributes":{"tuyaOperationMode":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":145,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1745731976005}

noctwm, 2025-05-04:
I also have this device (or very similar to it). In my case, it was sold as eWelink ZB-K1 (it was printed on the device packaging). Looks exactly as this one: https://www.zigbee2mqtt.io/devices/CK-TLSR8656-SS5-01(7000).html

Database entry:
{"id":10,"type":"EndDevice","ieeeAddr":"0xa4c138c1d5a16794","nwkAddr":63260,"manufId":4742,"manufName":"_TZ3000_ja5osu5g","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,1,3,4,6,4096,57345],"outClusterList":[25,10,3,4,6,8,4096],"clusters":{"genOnOff":{"attributes":{"tuyaOperationMode":1}},"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":145,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1746364973323}

ellnic, 2025-05-04:
@noctwm There are tons of Tuya clones of these since the original Xiaomi Mi ones with the Mi logo in the centre. I imagine this will be an ongoing task. Interestingly, I noticed that on one of mine, they have removed the reset/pair pin hole button on the back (probably to save costs). Instead you hold the main button for about 10 seconds.