Koenkk/zigbee2mqtt #27239: [Wrong device]: Or new device? Tuya Smart Knob detected as different Tuya model?

ellnic, 2025-04-27:
### Link

https://www.aliexpress.com/item/1005008066891727.html

### Model

Unknown

### Description

Smart Knob

### Vendor

Tuya

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/ZG-101Z-D.png

### Database entry

{"id":83,"type":"EndDevice","ieeeAddr":"0xa4c138f463fff318","nwkAddr":41903,"manufId":4417,"manufName":"_TZ3000_402vrq2i","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[1,3,4,6,4096,0],"outClusterList":[3,4,5,6,8,4096,25,10],"clusters":{"genBasic":{"attributes":{"65503":"��@/\u0011��@/\u0012��@/\u0016","65506":55,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_402vrq2i","zclVersion":3,"appVersion":66,"modelId":"TS004F","powerSource":3}},"genOnOff":{"attributes":{"tuyaOperationMode":1}},"manuSpecificTuya_3":{"attributes":{}},"genPowerCfg":{"attributes":{"batteryVoltage":25,"batteryPercentageRemaining":2}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002e14549d","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002e14549d","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0}],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1272899879},"lastSeen":1744616953776}

### Notes

This is an odd one. The product looks ZG-101Z_D but identifies as ERS-10TZBVK-AA. However the ZG-101Z_D does not expose `action_step_size, action_transition_time, action_rate, operation_mode` where as the device I have seems to expose and does indeed work with those functions:

[image]

I can definitely switch operation modes and see the steps etc when turning it.

Would this be better as a new device issue? No model is listed on Ali, and the manufacturer is iHseno, there is only one other iHseno entry in the Z2M DB: https://www.zigbee2mqtt.io/devices/TY_24G_Sensor_V2.html

Koenkk, 2025-04-28:
Could you check if this is fixed in the de branch?

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

ellnic, 2025-04-29:
Unless I've messed something up on my end, there's no change. 

I fired up another instance of Z2M to test, but I double checked the version and it's definitely correct: `2.2.1-dev commit: [97dbe1e8]`

The image / model shown are still the same:

[image]

Koenkk, 2025-05-01:
Fixed!

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `ZG-101Z_D.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

ellnic, 2025-05-01:
Thanks! No need to add the image, it's already in the DB: https://www.zigbee2mqtt.io/images/devices/ZG-101Z-D.png :)

ellnic, 2025-05-01:
Oh.. Sorry to be the bearer of bad news, but I have just updated my test instance, and the identification is still the same as before:

[image]

What can I do/get on my end for you to troubleshoot this?

Koenkk, 2025-05-03:
Whoops ,I did not merge https://github.com/Koenkk/zigbee-herdsman-converters/pull/9269 yet, update again once it is merged.

ellnic, 2025-05-04:
No worries! That's done it - it's detected correctly and all functions work. Thank you :) ❤ 

The product image isn't showing though - I think it's because your specified image has an underscore not dash before the `D` at the end. I uploaded a better image for this product recently: https://www.zigbee2mqtt.io/images/devices/ZG-101Z-D.png

Do you want me to submit a new PR with a file with underscore or would it be better to use the same image?

Koenkk, 2025-05-04:
@ellnic if you can make a Pr to make a copy of the old one would be great

ellnic, 2025-05-04:
Done! Thanks again for sorting this :)

closed 2025-05-05 by Koenkk