Koenkk/zigbee2mqtt #27246: The Zigbee devices do not respond to the commands sent.

bensmens, 2025-04-28:
### What happened?

Hello,

With my Home Assistant Yellow, my NodOn FP heating modules regularly stop responding to actions sent. I tried to switch the comfort mode to off from the Z2Mqtt interface, but without success. After refreshing the module's page, I can see it is still in comfort mode.
The only solution I have is to restart Z2Mqtt.

I don't see any errors in the logs, but when it doesn't work, I don't see the following lines:

- zh:ezsp: sendZclFrameToEndpointInternal ...
- zh:ezsp:ezsp: ==> sendUnicast: ...

Here’s a log when the command is taken into account.

[2025-04-28 15:01:53] [34mdebug[39m: 	z2m:mqtt: Received MQTT message on 'ha/Module FP sdb/set' with data '{"pilot_wire_mode":"off"}'
[2025-04-28 15:01:53] [34mdebug[39m: 	z2m: Publishing 'set' 'pilot_wire_mode' to 'Module FP sdb'
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:controller:endpoint: ZCL command 0x385b44fffec56c44/1 customClusterNodOnPilotWire.setMode({"mode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"manufacturerCode":4747,"writeUndiv":false})
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp: sendZclFrameToEndpointInternal 0x385b44fffec56c44:62239/1 (0,0,1), timeout=10000
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:ezsp: ==> sendUnicast: {"type":0,"indexOrDestination":62239,"apsFrame":{"profileId":260,"sequence":50,"clusterId":64512,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"messageTag":51,"message":{"type":"Buffer","data":[5,139,18,203,0,0]}}
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:ezsp: ==> {"_cls_":"sendUnicast","_id_":52,"_isRequest_":true,"type":0,"indexOrDestination":62239,"apsFrame":{"profileId":260,"sequence":50,"clusterId":64512,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"messageTag":51,"message":{"type":"Buffer","data":[5,139,18,203,0,0]}}
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: --> DATA (5,2,0): 5d00013400001ff3040100fc010100010000323306058b12cb0000
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: --> [521f21a9602a15adaa904b25565493499d4e2799dec86200ef0d63897d3a737e]
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: -?- waiting (6)
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: <-- [261fa1a9602a15aff1c97e]
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: <-- DATA (2,6,0): 261fa1a9602a15aff1c97e
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: --> ACK  (3)
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: --> [83401b7e]
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: <-- ACK (6): 261fa1a9602a15aff1c97e
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:ezsp: <== Frame: 5d80013400001d
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:ezsp: <== 0x34: {"_cls_":"sendUnicast","_id_":52,"_isRequest_":false,"status":0,"sequence":29}
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: -+- waiting (6) success
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: <-- [361fb1a96b2a15adaa904b2556549349]
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: <-- [9d4e27b6dece67888b7e]
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: <-- DATA (3,6,0): 361fb1a96b2a15adaa904b25565493499d4e27b6dece67888b7e
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: --> ACK  (4)
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: --> [8430fc7e]
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: <-- ACK (6): 361fb1a96b2a15adaa904b25565493499d4e27b6dece67888b7e
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:ezsp: <== Frame: 5d90013f00001ff3040100fc0101000100001d330000
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:ezsp: <== 0x3f: {"_cls_":"messageSentHandler","_id_":63,"_isRequest_":false,"type":0,"indexOrDestination":62239,"apsFrame":{"profileId":260,"sequence":29,"clusterId":64512,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"messageTag":51,"status":0,"message":{"type":"Buffer","data":[]}}
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: <-- [461fb1a97d312a15b65894b624ab1593]
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: <-- [499cc7836ef23d9874facae89b37753f]
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: <-- [a7e9ee397e]
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: <-- DATA (4,6,0): 461fb1a9112a15b65894b624ab1593499cc7836ef23d9874facae89b37753fa7e9ee397e
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: --> ACK  (5)
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: --> [8520dd7e]
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: <-- ACK (6): 461fb1a9112a15b65894b624ab1593499cc7836ef23d9874facae89b37753fa7e9ee397e
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:ezsp: <== Frame: 5d9001450000040100fc01014001000089a4c51ff3ffff070c8b12cb0b000002
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:ezsp: <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"sequence":137,"clusterId":64512,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":320},"lastHopLqi":164,"lastHopRssi":-59,"sender":62239,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[12,139,18,203,11,0,0]}}
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp: processMessage: {"messageType":0,"apsFrame":{"profileId":260,"sequence":137,"clusterId":64512,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":320},"lqi":164,"rssi":-59,"sender":62239,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[12,139,18,203,11,0,0]}}
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:controller: Received payload: clusterID=64512, address=62239, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=164, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":4747,"transactionSequenceNumber":203,"commandIdentifier":11},"payload":{"cmdId":0,"statusCode":0},"command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}}
[2025-04-28 15:01:53] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'ha/Module FP sdb', payload '{"energy":20.63,"last_seen":"2025-04-28T15:01:53+02:00","linkquality":164,"pilot_wire_mode":"eco","power":0,"power_on_behavior":null,"state":"ON","update":{"installed_version":66816,"latest_version":66816,"state":"idle"}}'
[2025-04-28 15:01:53] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'ha/Module FP sdb', payload '{"energy":20.63,"last_seen":"2025-04-28T15:01:53+02:00","linkquality":164,"pilot_wire_mode":"off","power":0,"power_on_behavior":null,"state":"ON","update":{"installed_version":66816,"latest_version":66816,"state":"idle"}}'
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: <-- [561fb1a97d312a15b65894b624ab1593499cc4876ff23d9874f4cae89b9b743fa7cbcddc6f6e7e]
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: <-- DATA (5,6,0): 561fb1a9112a15b65894b624ab1593499cc4876ff23d9874f4cae89b9b743fa7cbcddc6f6e7e
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: --> ACK  (6)
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: --> [8610be7e]
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:uart: <-- ACK (6): 561fb1a9112a15b65894b624ab1593499cc4876ff23d9874f4cae89b9b743fa7cbcddc6f6e7e
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:ezsp: <== Frame: 5d9001450000040100fc0101400100008aa0c41ff3ffff090c8b12670a0000200002
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp:ezsp: <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"sequence":138,"clusterId":64512,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":320},"lastHopLqi":160,"lastHopRssi":-60,"sender":62239,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[12,139,18,103,10,0,0,32,0]}}
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:ezsp: processMessage: {"messageType":0,"apsFrame":{"profileId":260,"sequence":138,"clusterId":64512,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":320},"lqi":160,"rssi":-60,"sender":62239,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[12,139,18,103,10,0,0,32,0]}}
[2025-04-28 15:01:53] [34mdebug[39m: 	zh:controller: Received payload: clusterID=64512, address=62239, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=160, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultRes [... cut ...]
  port: /dev/ttyAMA1

### What did you expect to happen?

I would like to understand the reason for the behavior I’m experiencing and find a solution. This is because it forces me to regularly monitor my radiators (especially) to ensure the reliability of the reception or transmission of the commands sent.

### How to reproduce it (minimal and precise)

I don't know

### Zigbee2MQTT version

2.2.1-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

### Setup

Home Assistant Yellow

ondrejmo, 2025-04-30:
I think this is related to https://github.com/Koenkk/zigbee2mqtt/issues/24041

Koenkk, 2025-04-30:
You are using the old `ezsp` driver, it's recommended to use the new `ember` drivers ([info](https://github.com/Koenkk/zigbee2mqtt/discussions/21462))

closed 2025-07-07 by github-actions[bot]