Koenkk/zigbee2mqtt #27260: Aeotec Pico Shutter option for enabling shutter mode missing

c0ldtech, 2025-04-29:
### What happened?

For Aeotec Pico Shutter models, shutter mode needs to be enabled to be able to use shutter tilt. However, it is disabled by default.

### What did you expect to happen?

Have a control to enable/disable shutter mode.

According to the manual, 

https://aeotec.freshdesk.com/support/solutions/articles/6000266743-aeotec-pico-shutter-user-guide

there is a configuration cluster 0xFD03. An ID 0x0001 needs to be changed from the default 0x00 (Roll shade mode) to 0x01 (Shutter mode).

Any advice how to do this any other way would also be appreciated.

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

7.4.4

### Adapter

Sonoff EmberZNet

### Setup

Docker

rodlima78, 2025-05-05:
I've got about 10 of these here at home. What I've found out is that unfortunately the cluster 0xFD03 doesn't expose these attributes.

The only way I could set them to shutter mode was **engaging the device's tilt calibration with the input switch (S2 IIRC) as documented**. To go back to roll blinds mod, a factory reset would do it.

I've opened a ticket at Aeotec customer support and here's the engineer's reply:

> Thank you, it does seem that this option isn't available in the other clusters - but i've let our team know while we apply for a works with home assistant ZHA. It does seem in order for manufacturer/configuration cluster to be reached, this does need to be added directly to the database as a ZHA-quirk. 

This answer doesn't make much sense to me, because the attributes can't be accessed via dev console, even using a external converter that exposes the cluster to z2m. So for me it's a firmware issue, not z2m (or ZHA for that matter), but hey, they definitely know their firmware more than I do. I've raised this question to them and I'm waiting for their reply. I'd expect an OTA update of some sorts.

More people opening tickets might also help bumping the priority in solving his issue. Here's their support cernter: https://aeotec.freshdesk.com/support/home

c0ldtech, 2025-05-05:
That's strange, I thought I tried that. I was also put off by their "preparation" point 2: "Please set the operation mode to 1 (shutter mode).". 

But you're saying that this does not actually need to be the case and just pressing S2 six times will automatically enable shutter mode? Too bad I sent mine back, I might just order one again and try again. 

Does the tilt work reliably for you? Is it possible to send both tilt and position commands at the same time? I.e. will sending "position": 80, "tilt": 50 result in the shutter to go to 80% height and then tilt to 50%?

I'm having the Nodon shutter controls here right now, and with them that's not possible (and tilt is not too reliable, either).

c0ldtech, 2025-05-06:
> The only way I could set them to shutter mode was **engaging the device's tilt calibration with the input switch (S2 IIRC) as documented**. To go back to roll blinds mod, a factory reset would do it.
> 
> I've opened a ticket at Aeotec customer support and here's the engineer's reply:

Are you positive that you can indeed control tilt? I did engage tilt calibration mode (having the shutter all the way down and then pressing S2 six times), and it does report on tilt when moving. But setting a tilt doesn't actually change the actual tilt, it only changes the tilt position in the software. 

Is there something that I'm missing?

closed 2025-07-13 by github-actions[bot]