Koenkk/zigbee2mqtt #27262: [New device support]: Aqara Climate Sensor W100 X0028HPT89

on-off-on, 2025-04-29:
### Link

https://www.aqara.com/eu/product/climate-sensor-w100/

### Database entry

{"id":52,"type":"EndDevice","ieeeAddr":"0x54ef441001265aab","nwkAddr":14243,"manufId":4447,"manufName":"Aqara","powerSource":"Battery","modelId":"lumi.sensor_ht.agl001","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[18,1029,1026,1,3,0,64704],"outClusterList":[25],"clusters":{"msRelativeHumidity":{"attributes":{"measuredValue":4000}},"genBasic":{"attributes":{"61441":2,"modelId":"lumi.sensor_ht.agl001"}},"manuSpecificLumi":{"attributes":{"22":"3DC59BEA8B24719A4E92164951599C245245A9A2E505C68A9417CA0BEB6E413F","32":4,"223":{"type":"Buffer","data":[40,35,64,0,0,0,3,35,90,0,0,0,4,33,2,0,5,33,0,0,41,35,0,0,0,0,42,35,0,0,0,0,43,33,126,52,44,41,163,255]},"238":3620,"247":{"type":"Buffer","data":[3,40,24,4,33,0,0,5,33,3,0,10,33,126,52,12,32,10,13,35,36,14,0,0,19,32,0,100,41,142,9,101,33,236,15,102,32,100,103,32,0]},"252":0,"354":1000,"355":30000,"356":50,"357":1,"358":-2000,"359":6000,"360":0,"362":30000,"363":300,"364":1,"365":0,"366":9999,"367":0,"368":2,"370":0,"371":0}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2600}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":0}}},"binds":[{"cluster":1029,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b82e49","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b82e49","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b82e49","endpointID":1}],"configuredReportings":[{"cluster":1029,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100},{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10}],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[18],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[18],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":36,"stackVersion":27,"hwVersion":12,"dateCode":"20250417","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1745962646753}

### Zigbee2MQTT version

2.2.1

### Comments

Checkes actual version of zigbee2mqtt, checked supported devices on website

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['lumi.sensor_ht.agl001'],
    model: 'lumi.sensor_ht.agl001',
    vendor: 'Aqara',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"2":2,"3":3}}), m.humidity(), m.temperature(), m.battery()],
    meta: {"multiEndpoint":true},
};
```

### What does/doesn't work with the external definition?

3 switches with press, double, long are not supportet, external inputs not supported

on-off-on, 2025-04-29:
further details:
https://csa-iot.org/csa_product/climate-sensor-w100/

TechArea, 2025-05-06:
> However actions don't report on HA and I can't get the external sensor working from HA either.

My switches work. I can switch things via automation (see example). But what doesn't work is that the display shows values from the outdoor sensor. This interface doesn't seem to have been implemented. But i'd rather have no display than have aqara on my home assistant. Still... I miss that feature too!

`
alias: Aqara W100 - Master Schalter Aktionen
description: ""
triggers:
  - entity_id: sensor.eg_master_schalter_action
    trigger: state
conditions: []
actions:
  - choose:
      - conditions:
          - condition: state
            entity_id: sensor.eg_master_schalter_action
            state: single_plus
        sequence:
          - target:
              area_id: wohnzimmer
            action: light.turn_on
            data: {}
      - conditions:
          - condition: state
            entity_id: sensor.eg_master_schalter_action
            state: double_plus
        sequence:
          - data:
              brightness_pct: 1
            target:
              area_id: wohnzimmer
            action: light.turn_on
      - conditions:
          - condition: state
            entity_id: sensor.eg_master_schalter_action
            state: hold_plus
        sequence:
          - data:
              brightness_pct: 100
            target:
              area_id: wohnzimmer
            action: light.turn_on
      - conditions:
          - condition: state
            entity_id: sensor.eg_master_schalter_action
            state: single_minus
        sequence:
          - target:
              area_id: wohnzimmer
            action: light.turn_off
            data: {}
      - conditions:
          - condition: state
            entity_id: sensor.eg_master_schalter_action
            state: double_minus
        sequence:
          - target:
              area_id: erdgeschoss
            action: light.turn_off
            data: {}
      - conditions:
          - condition: state
            entity_id: sensor.eg_master_schalter_action
            state: hold_minus
        sequence:
          - target:
              area_id: haus
            action: light.turn_off
            data: {}
      - conditions:
          - condition: state
            entity_id: sensor.eg_master_schalter_action
            state: single_center
        sequence:
          - data:
              brightness_pct: 100
            target:
              area_id:
                - vorgarten
                - eingang
            action: light.turn_on
      - conditions:
          - condition: state
            entity_id: sensor.eg_master_schalter_action
            state: double_center
        sequence:
          - target:
              area_id: wintergarten
            action: light.turn_on
            data: {}
      - conditions:
          - condition: state
            entity_id: sensor.eg_master_schalter_action
            state: hold_center
        sequence:
          - target:
              area_id:
                - vorgarten
                - eingang
                - wintergarten
            action: light.turn_off
            data: {}
    default: []
mode: single
`

TechArea, 2025-05-06:
And here's a blueprint:

```
blueprint:
  name: Aqara W100 – 3 Tasten, 3 Funktionen
  description: Reagiere auf alle 9 möglichen Aktionen des Aqara W100 (TH-S04D)
  domain: automation
  input:
    schalter_sensor:
      name: Action-Sensor
      description: Sensor-Entität mit action-Attribut (z. B. sensor.eg_master_schalter_action)
      selector:
        entity:
          domain: sensor
    action_single_plus:
      name: + einfach
      default: []
      selector:
        action: {}
    action_double_plus:
      name: + doppelt
      default: []
      selector:
        action: {}
    action_hold_plus:
      name: + lang
      default: []
      selector:
        action: {}
    action_single_center:
      name: Mitte einfach
      default: []
      selector:
        action: {}
    action_double_center:
      name: Mitte doppelt
      default: []
      selector:
        action: {}
    action_hold_center:
      name: Mitte lang
      default: []
      selector:
        action: {}
    action_single_minus:
      name: – einfach
      default: []
      selector:
        action: {}
    action_double_minus:
      name: – doppelt
      default: []
      selector:
        action: {}
    action_hold_minus:
      name: – lang
      default: []
      selector:
        action: {}

trigger:
  - platform: state
    entity_id: !input schalter_sensor
    attribute: action

condition: []

action:
  - choose:
      - conditions:
          - condition: template
            value_template: "{{ trigger.to_state.attributes.action == 'single_plus' }}"
        sequence: !input action_single_plus

      - conditions:
          - condition: template
            value_template: "{{ trigger.to_state.attributes.action == 'double_plus' }}"
        sequence: !input action_double_plus

      - conditions:
          - condition: template
            value_template: "{{ trigger.to_state.attributes.action == 'hold_plus' }}"
        sequence: !input action_hold_plus

      - conditions:
          - condition: template
            value_template: "{{ trigger.to_state.attributes.action == 'single_center' }}"
        sequence: !input action_single_center

      - conditions:
          - condition: template
            value_template: "{{ trigger.to_state.attributes.action == 'double_center' }}"
        sequence: !input action_double_center

      - conditions:
          - condition: template
            value_template: "{{ trigger.to_state.attributes.action == 'hold_center' }}"
        sequence: !input action_hold_center

      - conditions:
          - condition: template
            value_template: "{{ trigger.to_state.attributes.action == 'single_minus' }}"
        sequence: !input action_single_minus

      - conditions:
          - condition: template
            value_template: "{{ trigger.to_state.attributes.action == 'double_minus' }}"
        sequence: !input action_double_minus

      - conditions:
          - condition: template
            value_template: "{{ trigger.to_state.attributes.action == 'hold_minus' }}"
        sequence: !input action_hold_minus
```

TechArea, 2025-05-07:
Funny... I added a second device and got different results.
Something isn't working properly.
Maybe an Update caused the issue?

[image]

TechArea, 2025-05-07:
Ok, I've found the difference.
One Z2M config had some modifications.
Now it works on both Home Assistant Services.

Z2M config YAML:
  experimental_event_entities: true
  legacy_action_sensor: true

jamesonuk, 2025-05-21:
Before I either pull what is left of my hair out or spend hours delving into this, how can you put this into Zigbee mode?
From the above it almost sounds like you have to the Aqara hub so that you can pair it via Thread and then download a Zigbee firmware?   What if you don't have an Aqara hub?

Pel1can111, 2025-05-21:
Beat me to it! yeah, it worked surprisingly well to switch it over using their app. Biggest issue with aqara stuff is the lack of knowing when new firmware gets released. I haven't see a way to check, short of pairing it back to their app/hub to check.

jamesonuk, 2025-05-21:
> > Before I either pull what is left of my hair out or spend hours delving into this, how can you put this into Zigbee mode? From the above it almost sounds like you have to the Aqara hub so that you can pair it via Thread and then download a Zigbee firmware? What if you don't have an Aqara hub?
> 
> You need to download Aqara Home, the device will appear and you'll have to click on switch to Zigbee mode. It will flash the firmware and once it's done it will be able to join Z2M.
> 
> No need for any Aqara hardware

I installed the Aqara Home app (on Pixel 9)( but no device appears.  (how can it communicate with the device?   I know the Pixel 9 has a Thread radio but does that allow it to communicate without a TBR?)

Pel1can111, 2025-05-21:
pretty sure this one has Bluetooth

jamesonuk, 2025-05-21:
Ummm have tried everything I can think of but I can't get the device to appear in Aqara Home only for the Matter Add Device popup to appear but if I do this via Google Home or Aqara Home it just goes though the Thread radio and asks which TBR I want to use....

jamesonuk, 2025-05-21:
It is amazing what happens when you read it properly...   Also had to bypass Google trying to add the device as Matter.  The Zigbee bit didn't stand out to me.    I will update the docs

[image]

jamesonuk, 2025-05-23:
OK docs updated but wow this is chatty.....

[image]
that is an update every six seconds which seems a tad excessive for a temp sensor...

Anyway I can look at that but after pairing I was getting lots of errors because values were out of range. (despite me never setting them) and these two just wont seem to stick.    I think some of the settings in the converter aren't right.

Period for example has min 0.5 and max 600 seconds but a scale of 1,000 which seems wrong.   If you attempt to set this to anything above 32.5 is fails because it is int16 so only allowable values are >= -32,768 and <= 32,767 and with scale 1000, 33 is 33,000.

What is more weird is that changing the period is also changing the low humidity threshold (ah no, that is a typo by the looks of it

https://github.com/Koenkk/zigbee-herdsman-converters/blob/d1e2d75a042b9ed7f6e6d516292686f54e569a69/src/devices/lumi.ts#L4658

https://github.com/Koenkk/zigbee-herdsman-converters/blob/d1e2d75a042b9ed7f6e6d516292686f54e569a69/src/devices/lumi.ts#L4638

both have been set to the same attribute

But not sure why the humidity threshold isn't working, you get no error but it just won't stick and is stuck at 30 which is out of range

TechArea, 2025-06-02:
@jamesonuk Have you managed to display external sensors, for example? E.g. the temperature of a sensor in the garden?

jamesonuk, 2025-06-03:
> [@jamesonuk](https://github.com/jamesonuk) Have you managed to display external sensors, for example? E.g. the temperature of a sensor in the garden?

Yes but I think the name is misleading.    AIUI the external temperature / humidity values are ones you have to set manually.   So if you have a sensor you want to show on this you need to create an automation to send an MQTT message to set these values when they change on your sensor.   I had to change the sensor to external I think for this to work.

This then just appears on the device as "secondary"

jamesonuk, 2025-06-03:
Before I submit a PR could someone just try and update the scale on temp/humidity threshold attributes to be 100 and confirm this works?  (Need to update the actual converter)

There is an oddity that the range (which I guess someone got from somewhere) is 0.2 to 3.0 for temperature but I cannot seem to set this to less than 0.5 but the device seems a bit temperamental as to when it will store changes and when it decides to just reset things.   There are definitely still some things wrong with the converter but this is the one that is causing errors.

J108560, 2025-06-03:
I can confirm that changing the SCALE factor to 100 for both values ​​clears the error message.

J108560, 2025-06-06:
@jamesonuk For humi_threshold you have change valueMax key,  not key scale! Is this so right?

greenspeedracer, 2025-08-05:
I can enable/disable "HVAC Mode," enable the middle display line, send values to the middle line, enable the middle line to be changed from the W100, and decode the payload packet sent back from the W100 with the changes from the middle display line all via writing with mqtt.publish in Home Assistant and looking at the responses sent back (decoding the payload packet). I am not going to lie, I am very stupid and have only used Zigbee devices for about a month total. If there is anyone who wants to help integrate this into proper support for this device, I will gladly hand over everything I know and help finish with anymore info I can provide. I've got a LOT of notes, but I think I nailed down all the relevant bits and pieces. 

Hopefully what I have was the unknown and someone who isn't as dumb as me can integrate it. If no one is interested, I can try but it might take a long while due to my lack of knowledge, lol.

nsvrana, 2025-08-06:
If you want to share your notes or list the commands you're using and their results, one of us can write-up the definition. I've done another Zigbee converter in the past and I think I should be able to extend this one.

greenspeedracer, 2025-08-06:
Alrighty, I'll start with the "brief" overview. And for the record, I don't expect anyone to do this if they don't want to. I merely wanted to share what I discovered. Maybe putting this together will be super easy for talented folks like you or maybe it's impossibly annoying and I will fumble through it later on like the bumbling idiot, lol. Also, I can provide a lot more data or command structure and context about any parts of this necessary if anyone is interested. I am trying my best to keep it somewhat "concise" here.

I got the python scripts uploaded to github. [W100 Scripts](https://github.com/greenspeedracer/W100-Scripts/tree/main) All these commands are written to endpoint 1, cluster FCC0 and attribute FFF2. I started by looking at the firmware for the Aqara Hub M3, then sniffing the zigbee traffic after I figured out what I might be looking for. I'm no good at coding, so ChatGPT helped with the python scripts that parse data and generate commands. Consequently, they look like garbage and I am sure are written terribly. They all work, though. 

1. **Send "HVAC Mode On" Command.** GenerateHVACOn_TD.py This enables the the "main control" icon on the display (#2 from the manual located at [W100 Manual](https://cdn.shopify.com/s/files/1/0710/9220/7830/files/Climate_Sensor_W100_User_Manual_EN.pdf). The middle row does _not_ display yet.

2. **Send "PMTSD Data" Command.** GeneratePMTSD_TD.py This actually enables the middle row display. It sends Power, Mode, Temp (Set Point), Speed (Fan Speed) and what I _think_ is Display. Display might be related to the option to automatically hide the middle line after a preset amount of time. I haven't seen the display value actually change at all in observation, even when changing the option directly in the Aqara app. However, changing the option in the Aqara app to not blank out the middle row after a bit didn't work either, so YMMV. This same command also updates the display with whatever values are encoded and sent along at any point. Values for PMTSD data are listed in the python script and in the readme on my github. This is how to update the W100 if a thermostat is changed in Home Assistant or whatever.

3. **Change Settings via the W100.** DecodePMTSDReq_Fd.py and DecodePMTSD_FD.py This one was a bit trickier to figure out. When in HVAC mode, the buttons don't send events, but control the settings in the middle row, which are then packaged into an encoded string and sent back to the coordinator. When you press a button, the entire middle row will start blinking. This is the W100 sending an attribute report asking for the coordinator to send a PMTSD command. It's a lumiHeader, some spacing, the IEEE address of the coodinator (or whatever IEEE was used in the HVACModeOn command) and ending in "08:00:08:44." When this is received, the coordinator needs to respond with the current PMTSD state by sending a PMTSD. I believe this is the W100's failsafe to stay in sync. Once the W100 receives a PMTSD write back, it enables editing the middle row. You edit around to your liking following the instructions in the manual and shortly after ceasing to press buttons, it sends an attribute report back containing the payload. The payload can be decrypted by using DecodePMTSD_FD.py. It doesn't always send full PMTSD data. If you modify only P, it only sends P data. Modifying M or T always sends both M and T. Modifying S always sends S and P. Combining an M and S, for example, will send M, T, S and P. DecodePMTSD_FD.py takes this into account. No idea about D, since I can't trigger a D change via the device. 

5. **Send "HVAC Mode Off" Command.** GenerateHVACOff_TD.py This disables the the "main control" icon on the display (#2 from the manual located at [W100 Manual](https://cdn.shopify.com/s/files/1/0710/9220/7830/files/Climate_Sensor_W100_User_Manual_EN.pdf) and consequently the middle row. It also reverts the buttons back to sending events.

I have just been sending commands using mqtt.publish and in the format below. Obviously sub the value for whatever you are trying to do. The IEEE addresses I have been using are listed on my github too. I am sure there are parts in the scripts where they aren't labeled hub/coordinator and device. The hub ends in 71:1a and the W100 in d6:31. I haven't seen it make any difference in using the actual coordinator's IEEE vs the one I have been using from the Hub M3, so I just kept using the Hub M3's.

{
  "write": {
    "cluster": "manuSpecificLumi",
    "payload": {
      "65522": {
        "type": 65,
        "value": [170,113,31,68,112,115,5,65,28,0,0,84,239,68,128,113,26,8,0,8,68,15,80,48,95,77,49,95,84,50,53,95,83,50,95,68,48]
      }
    }
  }
}

greenspeedracer, 2025-08-14:
I have a fully working external converter (I think that is the right term) and the automation and scripts to go along with it to allow me to control my Home Assistant Generic Thermostat from the W100 as well as enable and disable "thermostat mode." I am pretty sure I nailed down the sync of states between the W100 and HA and vice versa. I also cleaned up the "display off" toggle in the exposes section to properly notate what it does and fixed the value mapping to be correct.

Since I use a generic thermostat, I have a couple things hard coded to certain values (such as fan speed) and use the heat/cool/auto mode to indicate if the unit is actively cooling (snowflake) vs idle (sun). I would have done the auto icon for idle, but auto disables the setpoint display, for example. 

All this is pretty well structured (at least to me) with 3 automations that call a combination of 4 scripts. Someone should be able to relatively easily adapt things like fan speed and mode to their liking if they take a look at the scripts. If anyone is interested in taking a look at the automations, scripts, or converter, please let me know. Before making it available to everyone, I would prefer if someone who was interested took a look first to see if they saw any glaring errors or omissions. Eventually I think the goal would be to get the external converter integrated into z2m and wrap the automations and scripts into a HA blueprint? Like I said before, this is all very new to me.

If you are interested (and hopefully smarter than me), please let me know!

nsvrana, 2025-08-14:
Hey, funny timing, I was going to add to this PR with a converter today. It sounds like you are further along than I am. If you want to add it to your other repo, I'll happily take a look over the next 1-2 days and provide feedback.

kkossev, 2025-08-15:
@greenspeedracer, thank you for sharing your findings and code with the community! Using your information, I successfully integrated the W100 HVAC mode into another home automation platform. 

Thanks again for your work!

greenspeedracer, 2025-08-24:
> Hi, well done and thank you for this great job [@greenspeedracer](https://github.com/greenspeedracer) . I took over your work to code it in an esp32-c6 in order to be able to manage this climate sensor in thermostat mode without having a home automation box or rather in case the home automation box crashes. Everything works well but I'm stuck on the temperature to be able to put tenths of degrees Celsius. Even when we program 20.5 on the w100 with the buttons, the return of this one on T is 20. Have you found a solution?

I'm glad it was useful, @Noyax-37 ! To be honest, I didn't even necessarily consider that since I operate in F, so the W100 always gets set to a funky setpoint (it reports values in C regardless of what scale the display is set to show.) I never saw the W100 report anything other than whole number values, so I am not sure why the W100 allows input in half degrees. I'm sure there is a reason, I just haven't been clever (or lucky) enough to figure it out yet.

I have kinda been in a holding pattern to see if anyone here smarter than me had any input on all this before I tried to figure out if it was appropriate to try to merge into z2m somehow, lol. 

Later this week I can take a look again and see if I can figure anything out. Figuring out what "D" is referring to has been bothering me quite a bit too. I'm sure there is more to unlock...

greenspeedracer, 2025-08-24:
> Thanks for the reply. This gave me the idea to test switching from °C -> °F and then vice versa. When 22°C is displayed on the w100 and you switch to °F, it displays 71.6°F... And if we display 72°F on the W100 then when switching to °C it displays 22.2°C... 
> 
> Do you have the possibility to check if other clusters are present on the w100 in addition to those defined in z2m?

If it reports something on that cluster I suppose I could see it while sniffing. What are you thinking? If you clue me in to what you are thinking it might help since I am not the greatest at this lol

Noyax-37, 2025-08-25:
> If it reports something on that cluster I suppose I could see it while sniffing. What are you thinking? If you clue me in to what you are thinking it might help since I am not the greatest at this lol

I'm not thinking of anything in particular, but I'm thinking that if it can display it, it must be possible somewhere. As you found out, when you press a button, the hub sends the frame back to the w100 with the same data as the ones on it. At home, since I don't know what the hub should send to the w100, I send PMTSD back, and even if I have a display at 22.5 and I send T22.5, the display returns to 22.0. At home, with a real hub, if you set the w100 to 72.5°F, for example, with the buttons, you wait 1 minute, or whatever, but until you press one of the buttons, the w100 requests the PTMSD frame from your hub. What does the hub send back? Does your display return to 72.0 or does it stay at 72.5?

Otherwise, what I also imagine is that something else is being sent between the two devices, either an additional attribute on the FFC0 cluster or on another cluster.

Sorry if this isn't clear, I'll use Google Translate (I'm French).

Noyax-37, 2025-08-25:
> There are a lot of different attributes on the FCC0 cluster, but none are read or written during a change in thermostat settings (or on any other cluster, for that matter). If you are interested, I have a list of all the raw FCC0 attributes somewhere.

Yes, I would like the list if it is easy for you to send it to me. Thank you.

Noyax-37, 2025-08-26:
[image]
as it is written here, apparently for the moment only the E1 thermostat would be compatible thanks to the M3 hub with the w100. The E1 must send the THERMOSTAT_OCCUPIED_HEATING_SETPOINT info to the M3 and the M3 sends this info back to the W100 but how??? You would have to find someone who has the 3

greenspeedracer, 2025-08-26:
> <img alt="Image" width="1010" height="420" src="https://private-user-images.githubusercontent.com/5978058/482127059-daa81ea2-3667-444a-abae-447bbe2df4e7.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTYyMjkxMTYsIm5iZiI6MTc1NjIyODgxNiwicGF0aCI6Ii81OTc4MDU4LzQ4MjEyNzA1OS1kYWE4MWVhMi0zNjY3LTQ0NGEtYWJhZS00NDdiYmUyZGY0ZTcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDgyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA4MjZUMTcyMDE2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzkxNWM4NmUzNzU1ZGRkODlmODhkNjFmN2YzM2NkMTRkMWY1MWE5OGZhMWE0NTZhNjVkYTgwNGE3MDhmZWRjNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.MTuC4JF29ZPZvPZxB6D-esQSfbgvebDk_wzgphWM_cM"> as it is written here, apparently for the moment only the E1 thermostat would be compatible thanks to the M3 hub with the w100. The E1 must send the THERMOSTAT_OCCUPIED_HEATING_SETPOINT info to the M3 and the M3 sends this info back to the W100 but how??? You would have to find someone who has the 3

It pairs "with an Aqara thermostat or an IR air conditioner" via "the hub M3." I would be willing to bet it sends the information in the exact same way as I observed with an "IR air conditioner" and "the Hub M3." I cannot prove that, but the flow to pair it with an E1, W400, or IR AC is the same via the Hub M3.

Also, you keep missing the point that I cannot find the W100 storing or sending (or reference to it doing either) the setpoint other than in the string I referred to, which is sent as a whole number.

Noyax-37, 2025-08-27:
> From what I can tell, the M3 (at some point) only supported 1* C increments with the IR blaster, but later images of the app and videos of folks using the W100 definitely shows support for .5* C increments. I wonder if we're seeing dumps from an older firmware?

I think you're right, the ability to put a number after the decimal point will certainly come in a future firmware update.

> The W100 pairs with the thermostats via the Hub, so I don't know why it would be any different either way. But I can't say that for sure. If you want to get an M3 and one of the thermostats, like the E1 or W400, I can tell you exactly how I was able to sniff the traffic. But I only ever see it send, receive, or update anything on FFF2.

I'll see if I can get all of this for not too much money.

greenspeedracer, 2025-08-28:
> I think you're right, the ability to put a number after the decimal point will certainly come in a future firmware update.
>
>
>
>
> I'll see if I can get all of this for not too much money.

In the US, the hub is about $130. The thermostats are not available at all, unfortunately. Also I use one of my smlight zigbee coordinators to sniff the traffic but I'm sure there are many other options for that.

If I had the firmware for the W100 it might help test some of the theories, but I can't find the firmware for the W100 anywhere.

tchlyah, 2025-09-01:
> Sorry, I keep getting sidetracked trying to take your code and adjust it to what I'd like it to do in Home Assistant instead of giving feedback. Since there is so much going on to control this, my intention is to break it out to a full Home Assistant Add-on which, instead of relying on zigbee2mqtt exposing specific capabilities to Home Assistant, just interacts with the device directly via MQTT, and supports multiple controllers/hvacs on the same add-on. There's a bit of work to get that going fully, so I've just been slogging away.

Looks amazing! @nsvrana, where are you with this Add-On ?

Noyax-37, 2025-09-08:
Hello, I modified @greenspeedracer's super converter so that everything can be modified directly from the zigbee2mqtt interface. Now you can turn the thermostat function ON or OFF and send any PMTSD values ​​you want. On the screenshot below there is a decimal point for the PMTSD T but this is only in case a firmware upgrade allows it. For the moment the W100 only retains the numbers before the decimal point

The converter:
[w100.js](https://github.com/user-attachments/files/22202072/w100.js)

z2m screenshot:
[image]

Noyax-37, 2025-09-08:
> D is "wind mode." See below from a couple weeks back, if you would like yours to display the name as Aqara uses it :)

oh yes, As it's not easy to really know what it's for, I preferred to put this but I can change it ;)

Thank's a lot for your job

greenspeedracer, 2025-09-08:
> > D is "wind mode." See below from a couple weeks back, if you would like yours to display the name as Aqara uses it :)
> 
> oh yes, As it's not easy to really know what it's for, I preferred to put this but I can change it ;)
> 
> Thank's a lot for your job

You're welcome! I hope one day someone can experiment with something like the E1 Radiator Thermostat in conjunction with the M3 Hub and we can see if it does anything different :)

greenspeedracer, 2025-09-09:
> > Do you have a link to any of the videos showing the W100 setting the temp in a half degree and it sticking that way? I tried to find one, but couldn't. I'd like to see if I can spot any different behavior in the video that might be a clue.
> 
> @greenspeedracer In case it helps, my W100 displays half degrees: https://youtube.com/shorts/QTmwzwDpHik (I use this converter: https://github.com/Koenkk/zigbee2mqtt/issues/27262#issuecomment-3264730170 and unfortunately decoded PMTSD_T values are always truncated to integers, i.e. 24.0 instead of 24.5)

Unfortunately, they all show half degrees in Celsius, but do not support sending that information along in any way that I can figure out. On a real Aqara Hub M3, it will return to the whole integer. My automations also reflect this to provide appropriate feedback to the user. You can check out my GitHub for more info if you would like. I also have all the Home Assistant code and the converter there as well :)

Noyax-37, 2025-09-21:
> I just found this thread, it seems that my issue from the thread below has been discussed here but without a solution to date :
> 
> [#28521](https://github.com/Koenkk/zigbee2mqtt/issues/28521)

I noticed the same thing, it's not a problem with the z2m but with the w100 itself

aldoushx, 2025-09-21:
> > I just found this thread, it seems that my issue from the thread below has been discussed here but without a solution to date :
> > [#28521](https://github.com/Koenkk/zigbee2mqtt/issues/28521)
> 
> I noticed the same thing, it's not a problem with the z2m but with the w100 itself

I don't own an aqara zigbee bridge, so I cannot tell if the threshold reduction works within the aqara app. Did you try ?

hdejongh, 2025-10-03:
> Hello, I modified [@greenspeedracer](https://github.com/greenspeedracer)'s super converter so that everything can be modified directly from the zigbee2mqtt interface. Now you can turn the thermostat function ON or OFF and send any PMTSD values ​​you want. On the screenshot below there is a decimal point for the PMTSD T but this is only in case a firmware upgrade allows it. For the moment the W100 only retains the numbers before the decimal point
> 
> The converter:
> 
> [w100.js](https://github.com/user-attachments/files/22203605/w100.js)
> 
> z2m screenshot: <img alt="Image" width="1142" height="821" src="https://private-user-images.githubusercontent.com/5978058/486621291-48bb4aa2-8151-423a-90c5-c931ca1e724e.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTk0ODI0NzAsIm5iZiI6MTc1OTQ4MjE3MCwicGF0aCI6Ii81OTc4MDU4LzQ4NjYyMTI5MS00OGJiNGFhMi04MTUxLTQyM2EtOTBjNS1jOTMxY2ExZTcyNGUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MTAwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTEwMDNUMDkwMjUwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OWI0YTZjN2EwNWMwNjY2NjNiNTM3Njk1NGI0Mzg5ZjJlMDljNGIzYzkxYzdmZWU0MmE5OGExY2ZjZmIwNTg2NCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.0cJzz5jSoZVr-ChjARzThDsARzx2VSZZbRJikad4SGc">

I have downloaded your w100.js file, putted it into the folder \zigbee2mqtt\external_extensions and restarted z2m.
But the file gets renamed to .invalid, am i doing something wrong?

Running z2m 2.6.1

greenspeedracer, 2025-10-07:
> Any progress on this? [@Noyax-37](https://github.com/Noyax-37) solution is pretty much it, maybe some labelling and a PR would be ready?

I think I submitted a PR to merge my changes in? As I stated before, I am really unfamiliar with this. Hopefully I did it right. https://github.com/Koenkk/zigbee-herdsman-converters/pull/10238

jrodrigoferreira, 2025-10-18:
This is quite nice, I'm able to get most functionality with this external converter. But can you guys get the Action on the Homeassistant end? I see it in the exposed entities in z2m and I see I being pushed over MQTT, but I seem to have no entity in the Homeassistant device which maps it...

jrodrigoferreira, 2025-10-18:
@greenspeedracer I mean the Action that is exposed in zigbee2mqtt (single_minus, single_plus, double_center, etc). I can see it in the z2m interface and I can see it being sent in the JSON over MQTT, but then the device on the homeassistant side doesn't have any entity for it.

[image]

[image]

jrodrigoferreira, 2025-10-18:
Ah, understood! I thought it would be exposed the same way :) In that case, excellent work, thanks!

greenspeedracer, 2025-10-22:
> So what is the current status of this issue? Any progress for non experimental user? Want to use the set temp line and actions

The status of what issue?

biller-aivy, 2025-10-23:
Is it possible now to use the middle Status ui line to Set temp?

greenspeedracer, 2025-10-24:
> Is it possible now to use the middle Status ui line to Set temp?

Yes, there’s been significant discussion in this thread about it.

jrodrigoferreira, 2025-10-26:
I'm having an issue where the device keeps reporting data (temperature and humidity, button presses, etc) but stops responding to commands. I have a software defined thermostat on the HA side and I'm using the External Temperature on the W100 to show the current setpoint. I send the external sensor order followed by the external temperature setting, which works fine, but after a while the device stops reacting. The only way to get functionality back is by resetting and repairing it with the network.

Anyone else suffering from something similar?

douglascrc-git, 2025-10-31:
> > Is it possible now to use the middle Status ui line to Set temp?
> 
> Yes, if you use an extension, I'd recommend using mine linked above

¿Can you explain what extension are you talking about? I don't see any link ¿How do you use said extension? 

I remember doing somenthing similar in ZHA with external quirks when a device was no supported at all. But never did that in Z2MQTT.

biller-aivy, 2025-11-02:
> > > Is it possible now to use the middle Status ui line to Set temp?
> > 
> > 
> > Yes, if you use an extension, I'd recommend using mine linked above
> 
> Hi ! Where should I install this extension ? I'm using HAOS on proxmox.
> 
> There was a recent zigbee2mqtt update that says that the thermostat mode is enabled, but I still can't see the middle display.
> 
> Thanks !

A official update?

biller-aivy, 2025-11-05:
> I'm having an issue where the device keeps reporting data (temperature and humidity, button presses, etc) but stops responding to commands. I have a software defined thermostat on the HA side and [...]

Did you tried the new Release with Thermostate Mode?

biller-aivy, 2025-11-08:
@kipk I’m using a Shelly 1 Gen4 at my heating manifold, and now I can see the “Set Temperature” in Home Assistant.
What’s the next step if I want the thermostat to control the Shelly?
Do I need to create a virtual thermostat in Home Assistant?
I can’t find the W100 listed as a sensor there — am I missing something?

ABoldOne, 2025-11-08:
@KipK does your Converter support one decimal place? I've noticed that you can set 19.5 degres on the W100 itself but when setting it in HA or Z2M Dashboard it won't work.

KipK, 2025-11-12:
@douglascrc-git That’s totally normal 🙂
It’s actually a remote, not a real thermostat — so there’s no hvac_action since it’s not doing any heating or cooling itself. The device simply reports the HVAC mode.

KipK, 2025-12-12:
I need you guys to up this thread on aqara forum:

https://forum.aqara.com/t/w100-zigbee-firmware-inconsistencies/167176

thomasvnl, 2025-12-12:
> I need you guys to up this thread on aqara forum:
> 
> https://forum.aqara.com/t/w100-zigbee-firmware-inconsistencies/167176

Done, hope they pick it up as it is a bit weird to be able to set the decimal in thermostat mode without it being communicated over ZigBee. Thanks for addressing the issue at the aqara forums.

KipK, 2026-02-21:
I think the lumi manuspecific cluster is also publishing temperature. 

We should have a way to ignore it on the converter side. 
I'll give a look to this. I have not updated the fw yet.

KipK, 2026-02-23:
hugh. So we're pretty fu... 
Looks like a fw bug indeed. There's no way this has any sense by design. 

What does it says in Aqara forum ? Doesn't users report weird behavior too ? 

Thanks for the log

nstrelow, 2026-10-21:
Another bump since this is a cool device :)

vrsf-nl, 2026-10-21:
> 31\. Another bump since this is a cool device :)

I would like for it to be fixed as well but seeing how Aqara replies on their forum they have no intent to actually make this device really functional outside of their Aqara hub. Unless someone could build custom firmware for it and wants to spend all the time and effort involved I think its just not going to happen.

Will88fr, 2026-10-21:
Is it possible to download the current firmware maybe we can ask to Claude to fix the issue?

[1 more reply]