Koenkk/zigbee2mqtt #27265: S4SW-001X16EU not working with "on_time"

Deejayfool, 2025-04-30:
### What happened?

Nothing happen when I send this payload : 
`{"state": "ON", "on_time": 2}`
Or this : 
`{"state": "ON", "on_time": 2, "off_wait_time": 1}`
But it works when I send this : 
`{"state": "ON"}`

### What did you expect to happen?

In this [documentation](https://www.zigbee2mqtt.io/devices/S4SW-001X16EU.html), I see that I can send "on_time" with a duration in **seconds** that allow the Shelly to automatically switch off.
That's exactly what I need for my garage door. I need to send "ON" during some seconds (2 for example), and then switch the Shelly to OFF.

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

0x26680700

### Adapter

ConBee2/RaspBee2

### Setup

Linux Zigbee2Mqtt 6.1.0-34-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.135-1 (2025-04-25) x86_64 GNU/Linux
Debian GNU/Linux 12

### Debug log

[2025-04-30 20:01:22] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqtt/interrupteur_porte_garage/set' with data '{"state": "ON", "on_time": 2, "off_wait_time": 1}'
[2025-04-30 20:01:22] debug:    z2m: Publishing 'set' 'state' to 'interrupteur_porte_garage'
[2025-04-30 20:01:22] debug:    zh:controller:endpoint: ZCL command 0xa085e3fffebd8128/1 genOnOff.onWithTimedOff({"ctrlbits":0,"ontime":20,"offwaittime":10}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-04-30 20:01:22] debug:    zh:deconz:driver: DATA_REQUEST - destAddr: 0x95dc EP:1 SeqNr. 113 request id: 149
[2025-04-30 20:01:22] debug:    zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 149 status: 0
[2025-04-30 20:01:22] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-04-30 20:01:22] debug:    zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 149 status: 0
[2025-04-30 20:01:22] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-04-30 20:01:22] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 10100110
[2025-04-30 20:01:22] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-04-30 20:01:22] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 10100110
[2025-04-30 20:01:22] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-04-30 20:01:22] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 10101110
[2025-04-30 20:01:22] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2025-04-30 20:01:22] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 10101110
[2025-04-30 20:01:22] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2025-04-30 20:01:22] debug:    zh:deconz:driver: query aps data confirm
[2025-04-30 20:01:23] debug:    zh:deconz:driver: DATA_CONFIRM - sending data state request - SeqNr. 114
[2025-04-30 20:01:23] debug:    zh:deconz:driver: query aps data indication
[2025-04-30 20:01:23] debug:    zh:deconz:frameparser: DATA_CONFIRM RESPONSE - destAddr: 0x95dc request id: 149 confirm status: 0
[2025-04-30 20:01:23] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2025-04-30 20:01:23] debug:    zh:deconz: sendZclFrameToEndpoint - message send with transSeq Nr.: 95
[2025-04-30 20:01:23] debug:    zh:deconz: false, false, false, 10000
[2025-04-30 20:01:23] debug:    zh:deconz: resolve request (95)
[2025-04-30 20:01:23] debug:    zh:deconz:driver: query aps data indication
[2025-04-30 20:01:23] debug:    zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 115
[2025-04-30 20:01:23] debug:    zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 115 srcAddr: 0x95dc destAddr: 0x0 profile id: 0x104 cluster id: 0x6 lqi: 255
[2025-04-30 20:01:23] debug:    zh:deconz:frameparser: response payload: 185f0b4285
[2025-04-30 20:01:23] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-04-30 20:01:23] debug:    zh:controller: Received payload: clusterID=6, address=38364, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=255, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":95,"commandIdentifier":11},"payload":{"cmdId":66,"statusCode":133},"command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}}
[2025-04-30 20:01:23] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/interrupteur_porte_garage', payload '{"last_seen":"2025-04-30T18:01:23.104Z","linkquality":255,"state":"OFF"}'
[2025-04-30 20:01:23] debug:    zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 115 srcAddr: 0x95dc destAddr: 0x0 profile id: 0x104 cluster id: 0x6 lqi: 255
[2025-04-30 20:01:23] debug:    zh:deconz:frameparser: response payload: 185f0b4285
[2025-04-30 20:01:23] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-04-30 20:01:23] debug:    zh:controller: Received payload: clusterID=6, address=38364, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=255, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":95,"commandIdentifier":11},"payload":{"cmdId":66,"statusCode":133},"command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}}
[2025-04-30 20:01:23] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/interrupteur_porte_garage', payload '{"last_seen":"2025-04-30T18:01:23.105Z","linkquality":255,"state":"OFF"}'
[2025-04-30 20:01:23] debug:    zh:deconz: response received (95)
[2025-04-30 20:01:23] debug:    zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 116
[2025-04-30 20:01:23] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 10101010
[2025-04-30 20:01:23] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2025-04-30 20:01:23] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 10101010
[2025-04-30 20:01:23] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2025-04-30 20:01:23] debug:    zh:deconz:driver: query aps data indication
[2025-04-30 20:01:23] debug:    zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 117
[2025-04-30 20:01:23] debug:    zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 117 srcAddr: 0xb3d destAddr: 0x0 profile id: 0x104 cluster id: 0x702 lqi: 255
[2025-04-30 20:01:23] debug:    zh:deconz:frameparser: response payload: 18520a00042a020f00
[2025-04-30 20:01:23] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-04-30 20:01:23] debug:    zh:controller: Received payload: clusterID=1794, address=2877, groupID=0, endpoint=6, destinationEndpoint=1, wasBroadcast=false, linkQuality=255, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":82,"commandIdentifier":10},"payload":[{"attrId":1024,"dataType":42,"attrData":3842}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2025-04-30 20:01:23] debug:    zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 117 srcAddr: 0xb3d destAddr: 0x0 profile id: 0x104 cluster id: 0x702 lqi: 255
[2025-04-30 20:01:23] debug:    zh:deconz:frameparser: response payload: 18520a00042a020f00
[2025-04-30 20:01:23] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-04-30 20:01:23] debug:    zh:controller: Received payload: clusterID=1794, address=2877, groupID=0, endpoint=6, destinationEndpoint=1, wasBroadcast=false, linkQuality=255, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":82,"commandIdentifier":10},"payload":[{"attrId":1024,"dataType":42,"attrData":3842}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}

quatsch, 2025-07-06:
J'ai le même probleme ici...

horus389, 2025-07-26:
Hey!
Exactly the same issue here, doesn't work when I put more than the state in the payload.
Any news on this?
Thanks!

Deejayfool, 2025-07-26:
If it can help, I fixed it by directly set up "Auto off" in Shelly App.
So I only send the command "{"state": "ON"}" and so it stop after 1 second thanks to Shelly App.

faquin, 2025-09-05:
Eyh, I just spent last 2 hours trying to debut MQTT with on_time, and found the issue is not on my side...
Anyway, just commenting to get updated in this topic.