Koenkk/zigbee2mqtt #27268: ZY-M100-24GV2 presense sensor parameters not sticking

Mrsash2020, 2025-05-01:
### What happened?

The detection sensitivity and range don't seem to stick. Goes back to original. Been like this for a while but its annoying now.

### What did you expect to happen?

I am able to change those parameters but they don't seem to stick permanently if I don't touch them

### How to reproduce it (minimal and precise)

Changes back when left alone

### Zigbee2MQTT version

Current version: 2.2.1-1

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

Win 11 Pro 64 Machine/Vmware Player

### Debug log

Think didn't find anything in the Logs window but will try and get logs soon. Just need to understand how I do it.

kip3435423, 2025-05-02:
same for me i even tried automations as a work around but its constant!

joshuas79, 2025-06-11:
Same issue here