Koenkk/zigbee2mqtt #27270: Firmware Upgrade Issue with Ubisys S1-R

rob1981forest, 2025-05-01:
### What happened?

I have the Ubisys S1-R working well within Z2M but there is an issue i have been dealing with their support team on that is fixed by the latest firmware. An auto update check doesn't show any available firmware so i downloaded the firmware file and added it locally to my 'my_index.json' file, the device now see firmware available but won't take it as Z2M reports "**update of 'Car Charger Switch' failed (Hardware version mismatch)**"

this is the device in z2m - https://www.zigbee2mqtt.io/devices/S1-R.html#ubisys-s1-r

ubisys have two variants of this S1-R - the original and a series 2 which includes bluetooth (this is my version) - so i'm wondering if something is going on here inside the z2m logic ?

here a link to the ubisys firmware pages if that helps - https://www.ubisys.de/en/support/firmware/

this is the firmware i need for my device - https://www.ubisys.de/wp-content/uploads/10F2-7B45-0100-0100-02610463-ubisys-s1r-qpg6105.ota.zigbee

### What did you expect to happen?

firmware to upgrade ok

### How to reproduce it (minimal and precise)

add firmware file and attempt upgrade

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

20250321

### Adapter

SLZB-06

### Setup

Home Assistant Add-on running in Synology VM

rob1981forest, 2025-05-15:
Is anyone available to help here please ?

rob1981forest, 2025-05-19:
when i do a firmware check now i just get the error:

z2m: Failed to check if update available for 'Car Charger Switch' (Device didn't respond to OTA request)

I've been adding to the Ubisys app, taking it off and re-adding it a lot over the past few days to try and do the upgrade via that method. Not sure why it now tells me the device wont respond to firmware check - everything else about the device seems to work ok.

daNick73, 2025-06-07:
Update:

For my Ubisys S1-R Series 2, an OTA Firmware can now be installed. 
version is 0.2.61-0.4.63, which matches data on https://www.ubisys.de/en/support/firmware/

so, seems like @Koenkk fixed something again ... like all the times before :)

closed 2025-08-15 by github-actions[bot]