Koenkk/zigbee2mqtt #27273: Unexpected Valve Activation Behavior in GIEX GX03 Water Controller

Popoboxxo, 2025-05-01:
### What happened?

The GIEX GX03 water controller exhibits inconsistent behavior when managing multiple valves via Zigbee commands.

Valve 1 functions as expected: Turning it on initiates water flow, and turning it off stops it reliably. A brief delay between operations is observed but does not affect functionality.

Valve 2, however, behaves unexpectedly. When turning Valve 2 on, Valve 1 appears to activate as well, according to the UI and Zigbee command feedback.

Reproducibility: The issue occurs consistently across multiple attempts.

Actual Behavior:

While the UI and Zigbee commands indicate both Valve 1 and Valve 2 are active when Valve 2 is turned on, only Valve 2 actually allows water flow in reality.

To enable water flow through Valve 1, it must first be switched off and then turned back on again.

This discrepancy between the UI feedback, Zigbee commands, and the actual water flow suggests an inconsistency in command execution or state representation.

### What did you expect to happen?

Each valve should function independently without triggering or affecting the other unless explicitly commanded.

### How to reproduce it (minimal and precise)

Steps to Reproduce:

Turn on Valve 1 → Water flows correctly.

Turn off Valve 1 → Water stops.

Turn on Valve 2 → UI/Zigbee indicate Valve 1 is also active.

Observe that only Valve 2 is actually allowing water flow.

Turn off Valve 1 and turn it on again → Only now does it properly allow water flow.

### Zigbee2MQTT version

 2.2.1

### Adapter firmware version

20210708

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

HomeAssistent Addon, ProxMox HomeAssistent Installation

### Debug log

will be added

Popoboxxo, 2025-05-01:
Behavior Video - UI z2m and HomeAssistent
https://github.com/user-attachments/assets/3b02a5b9-3557-4d01-8ad4-e5a4f48c9640

chris-1243, 2025-05-01:
It should be fixed in may release.

At least some work has been done

Check

https://github.com/Koenkk/zigbee-herdsman-converters/pull/9266

Popoboxxo, 2025-05-01:
Awesome :) Thx @chris-1243

Popoboxxo, 2025-05-10:
Problem is Fixed! Tested several times without andy issues!

closed 2025-05-10 by Popoboxxo