Koenkk/zigbee2mqtt #27279: 2.3.0 does not start after update

Fitzbert-Fitz, 2025-05-01:
### What happened?

Zigbee2mqtt was working fine. after upgrading it does not start anymore. 
Using a Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2 with ember firmware on a PI5 with haos.

[21:49:57] INFO: Preparing to start...
[21:49:58] INFO: Socat not enabled
[21:49:58] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-05-01 21:49:59] info: 	z2m: Logging to console, file (filename: log.log)
[2025-05-01 21:49:59] info: 	z2m: Starting Zigbee2MQTT version 2.3.0 (commit #unknown)
[2025-05-01 21:49:59] info: 	z2m: Starting zigbee-herdsman (4.0.0)
[2025-05-01 21:49:59] info: 	zh:ember: Using default stack config.
[2025-05-01 21:49:59] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-05-01 21:49:59] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-05-01 21:49:59] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-05-01 21:49:59] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-05-01 21:49:59] info: 	zh:ember:uart:ash: Serial port opened
[2025-05-01 21:49:59] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-05-01 21:50:01] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-05-01 21:50:01] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-05-01 21:50:04] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-05-01 21:50:04] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-05-01 21:50:06] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-05-01 21:50:06] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-05-01 21:50:09] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-05-01 21:50:09] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-05-01 21:50:11] error: 	z2m: Error while starting zigbee-herdsman
[2025-05-01 21:50:11] error: 	z2m: Failed to start zigbee-herdsman
[2025-05-01 21:50:11] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-05-01 21:50:11] error: 	z2m: Exiting...
[2025-05-01 21:50:11] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:666:19)
    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1548:24)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/controller/controller.ts:133:29)
    at Zigbee.start (/app/lib/zigbee.ts:70:27)
    at Controller.start (/app/lib/controller.ts:104:13)
    at start (/app/index.js:149:5)

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

no coordinator revision shown under settings about revision. i think silabsv7.4.5.0

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2 

### Setup

Add on on haos on a pi5

twantythree, 2025-05-01:
I have the same issue with the same hardware; downgrading to the old backup 2.2.1-1 was useless.

Fitzbert-Fitz, 2025-05-02:
@twantythree thx for the hint. After changing to `adapter: zstack` I got a different error message and z2m still not starting. Changing back to `adapter: ember`. It starts working. 🤔 strange behaviour

chris-1243, 2025-05-02:
@Fitzbert-Fitz 

You have a Dongle-E according the information you posted:
`Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2`
It is based on a Silabs chip and if you have a firmware 7.4.x or higher, you have to use `ember`.

If you use the Dongle-P which uses a TI chip, then you must use `zstack`.