Koenkk/zigbee2mqtt #27282: [New device support]: Zemismart (existing device but missing Covers cluster)

Majestic7979, 2025-05-01:
### Link

https://www.zigbee2mqtt.io/devices/ZM85EL-2Z.html#zemismart-zm85el-2z

### Database entry

Already known

### Zigbee2MQTT version

 2.2.1 commit: a14feef8c13f5ce49b0ac358d98d3234d21d7ff4

### Comments

THis device is already supported but it does not expose a critical cluster for binding to remotes/buttons. Being a curtain, it's imperative that it can work with buttons. I have IKEA buttons specific for IKEA blinds and they work fine, I can bind it and the Covers cluster shows up for the IKEA blinds but not for this device. So I can only control it via Z2M dashboard or home assistant. Not everyone has access to those at home so buttons are essential. Could this please be fixed so we can bind the curtains to our buttons, as the only cluster we see when trying to bind button to this zemismart motor is "Groups" and if we bind the button to that, it does nothing at all when we press it. Weirdly, it works fine on ZHA, so it could be a definition issue.

### External definition

```shell
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_nw1r9hp6',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

No "Covers" cluster exposed to bind to buttons for physical control

Koenkk, 2025-05-03:
Could you try to:

- Stop Z2M
- Open `data/database.db` with a text editor and change all occurrences of `_TZE200_nw1r9hp6` to `_TZE200_cf1sl3tj`
- Start Z2M and see if the device now works

Majestic7979, 2025-05-04:
> Could you try to:
> 
> * Stop Z2M
> * Open `data/database.db` with a text editor and change all occurrences of `_TZE200_nw1r9hp6` to `_TZE200_cf1sl3tj`
> * Start Z2M and see if the device now works

Unfortunately it made no difference

[image]

Koenkk, 2025-05-04:
what shows up under the about tab? (of the device)

Majestic7979, 2025-05-09:
> what shows up under the about tab? (of the device)

[image]
[image]
[image]
[image]

Majestic7979, 2025-06-16:
Wondering if you had a chance to review this and are able to provide a comment about it? The motors only come with a single RF remote but these ikea zigbee buttons are cheap and widely available, as well as being more convenient and discreet to position around the house so it's really needed in my case. Would be extremely grateful if this could be implemented! THank you in advance.

Koenkk, 2025-06-17:
Could you try to stop z2m, add `258` to the `inClusterList` of the cover `data/database.db` entry , start z2m and see if that works?

closed 2025-08-24 by github-actions[bot]