Koenkk/zigbee2mqtt #27285: Deconz ConbeeII - Not working in V2.3.0

Imaginous, 2025-05-02:
### What happened?

After upgrading V2.2.1 to V2.3.0 I get an error on the Deconz ConbeeII

### What did you expect to happen?

For it to work normally

### How to reproduce it (minimal and precise)

Use a Deconz ConbeeII adapter.

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

Latest

### Adapter

ConbeeII

### Setup

Standalone on a raspberry pi 3B+

### Debug log

`May 02 11:14:27 domoticapi1 npm[2016]: [2025-05-02 11:14:27] error:         z2m: Failed to start zigbee-herdsman
May 02 11:14:27 domoticapi1 npm[2016]: [2025-05-02 11:14:27] error:         z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
May 02 11:14:27 domoticapi1 npm[2016]: [2025-05-02 11:14:27] error:         z2m: Exiting...
May 02 11:14:27 domoticapi1 npm[2016]: [2025-05-02 11:14:27] error:         z2m: Error: Get coordinator version Error: TypeError [ERR_INVALID_ARG_TYPE]: The first argument must be of type string or an instance of Buffer, ArrayBuffer, or Array or an Array-like Object. Received type number (170)
May 02 11:14:27 domoticapi1 npm[2016]:     at DeconzAdapter.getCoordinatorVersion (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:289:19)
May 02 11:14:27 domoticapi1 npm[2016]:     at Controller.getCoordinatorVersion (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/controller/controller.ts:415:16)
May 02 11:14:27 domoticapi1 npm[2016]:     at Zigbee.getCoordinatorVersion (/opt/zigbee2mqtt/lib/zigbee.ts:203:16)
May 02 11:14:27 domoticapi1 npm[2016]:     at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:133:65)
May 02 11:14:27 domoticapi1 npm[2016]:     at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:104:13)
May 02 11:14:27 domoticapi1 npm[2016]:     at start (/opt/zigbee2mqtt/index.js:149:5)
`

Koenkk, 2025-05-03:
Could you provide the full debug log of this? (from the point where you start z2m)

Koenkk, 2025-05-05:
> Now it started, but many devices are offline and not responding.

Great that it started, I don't see any devices failing to respond in this log (log is very short)

mglaabde, 2025-05-05:
I have massive issues, too. Quite similar setup.
In particular, Philips Hue devices are experiencing connectivity issues.

Imaginous, 2025-05-06:
Here is a screenshot of the web interface and a longer log.

[debug.txt](https://github.com/user-attachments/files/20067906/debug.txt)
[image]

Imaginous, 2025-05-07:
As soon as I start the previous version it works flawless again.

First I thought it were only battery operated devices, but I don't see any shared "property" that may cause this.

I have ordered a ZigStar UTP, arriving next week, for testing.

No corresponding pan ID or channel here.

Imaginous, 2025-05-17:
@Koenkk I have transfered the config of my Deconz ConbeeII to the UZG-01 with Zigpy.
Then tested everything, upgraded to V2.3.0, everything is working now with the UZG-01.

Only the new adapter (connected via UTP) seems to be slower than the ConbeeII on the USB port.
Is this a possible cause of the 115200 baudrate of the zstack firmware?

Koenkk, 2025-05-18:
>  Is this a possible cause of the 115200 baudrate of the zstack firmware?

Very likely not, what do you mean with "slower"?

Imaginous, 2025-05-19:
I will explain the setup.
I have an Ikea button. When I press it my own software detects the click via MQTT. Zigbee2mqtt and Mosquito are running on a Raspberry Pi 3B+. 

Then it sends 2 individual packages (I know I can create groups) containing a state and brightness to 2 lamps.

With the Deconz Conbee II the lamps turn on simultaneously, the delay is not noticeable with the naked eye.

If I do the same with the UZG-01 via wired lan or usb the first lamp turns on almost directly, but there is about half a second delay before turning on the second lamp.

This is a pretty constant delay. This happens with the V2.2.x and V2.3.0 firmware.

A long, long time ago (maybe 3 years ago) I had this issues with the Deconz too with a certain version... Back then you made some alterations and the Deconz was fast again.

The behavior of the UZG-01 seems the same.
It looks like the commands are delayed somehow.
I tried adapter_delay=0 and concurrent=16, but that didn't change anything.

Sensor input is almost instantaneous and the first command send out also, but consecutive commands send directly after one and other have a (guessed) delay of about 0.5 seconds.

Koenkk, 2025-05-19:
Could you provide the debug log of when the delay occurs?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Imaginous, 2025-05-19:
Will make a log tomorrow. Keep you posted.

Imaginous, 2025-05-21:
This is the debug log file. Note that it's one controller with 2 light endpoints (L1, L2).

File: [debug_20250521.txt](https://github.com/user-attachments/files/20363200/debug_20250521.txt)

Code block:

`May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Slaapkamer1_Verlichting/l1/set' with data '{
May 21 10:20:25 domoticapi1 npm[19619]:   "state": "on",
May 21 10:20:25 domoticapi1 npm[19619]:   "brightness": 254
May 21 10:20:25 domoticapi1 npm[19619]: }'
May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         z2m: Publishing 'set' 'brightness' to 'Slaapkamer1_Verlichting'
May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         zh:controller:endpoint: ZCL command 0xbc33acfffe6563a7/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false})
May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         zh:zstack: sendZclFrameToEndpointInternal 0xbc33acfffe6563a7:46932/1 (0,0,1)
May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":46932,"destendpoint":1,"srcendpoint":1,"clusterid":8,"transid":127,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,251,4,254,10,0]}}
May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         zh:zstack:unpi:writer: --> frame [254,16,36,1,84,183,1,1,8,0,127,0,30,6,1,251,4,254,10,0,179]
May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Slaapkamer1_Verlichting/l2/set' with data '{
May 21 10:20:25 domoticapi1 npm[19619]:   "state": "on",
May 21 10:20:25 domoticapi1 npm[19619]:   "brightness": 254
May 21 10:20:25 domoticapi1 npm[19619]: }'
May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         z2m: Publishing 'set' 'brightness' to 'Slaapkamer1_Verlichting'
May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         zh:controller:endpoint: ZCL command 0xbc33acfffe6563a7/2 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false})
May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         zh:zstack:unpi:parser: --- parseNext []
May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,127,185]
May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,127,185]
May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,127] - 185
May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":127}
May 21 10:20:25 domoticapi1 npm[19619]: [2025-05-21 10:20:25] debug:         zh:zstack:unpi:parser: --- parseNext []
May 21 10:20:26 domoticapi1 npm[19619]: [2025-05-21 10:20:26] debug:         zh:zstack:unpi:parser: <-- [254,5,69,196,84,183,1,121,92,67]
May 21 10:20:26 domoticapi1 npm[19619]: [2025-05-21 10:20:26] debug:         zh:zstack:unpi:parser: --- parseNext [254,5,69,196,84,183,1,121,92,67]
May 21 10:20:26 domoticapi1 npm[19619]: [2025-05-21 10:20:26] debug:         zh:zstack:unpi:parser: --> parsed 5 - 2 - 5 - 196 - [84,183,1,121,92] - 67
May 21 10:20:26 domoticapi1 npm[19619]: [2025-05-21 10:20:26] debug:         zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd
May 21 10:20:26 domoticapi1 npm[19619]: [2025-05-21 10:20:26] debug:         zh:zstack:unpi:parser: --- parseNext []
May 21 10:20:26 domoticapi1 npm[19619]: [2025-05-21 10:20:26] debug:         zh:zstack:unpi:parser: <-- [254,25,68,129,0,0,8,0,84,183,1,1,0,178,0,135,249,38,0,0,5,8,251,11,4,0,121,92,28,29]
May 21 10:20:26 domoticapi1 npm[19619]: [2025-05-21 10:20:26] debug:         zh:zstack:unpi:parser: --- parseNext [254,25,68,129,0,0,8,0,84,183,1,1,0,178,0,135,249,38,0,0,5,8,251,11,4,0,121,92,28,29]
May 21 10:20:26 domoticapi1 npm[19619]: [2025-05-21 10:20:26] debug:         zh:zstack:unpi:parser: --> parsed 25 - 2 - 4 - 129 - [0,0,8,0,84,183,1,1,0,178,0,135,249,38,0,0,5,8,251,11,4,0,121,92,28] - 29
May 21 10:20:26 domoticapi1 npm[19619]: [2025-05-21 10:20:26] debug:         zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":8,"srcaddr":46932,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":178,"securityuse":0,"timestamp":2554247,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[8,251,11,4,0]}}
[... cut ...] ,"value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}}
May 21 10:20:28 domoticapi1 npm[19619]: [2025-05-21 10:20:28] debug:         zh:zstack:unpi:parser: --- parseNext []
May 21 10:20:28 domoticapi1 npm[19619]: [2025-05-21 10:20:28] debug:         z2m: Received Zigbee message from 'Blitzwolf_Stroommeting', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":19,"rmsCurrent":141,"rmsVoltage":232}' from endpoint 1 with groupID 0
`

Koenkk, 2025-05-22:
This is an interesting case indeed, both ember and zstack only send 1 command to a device at a time (to not overload it). Which means that for the command to the second endpoint to be send, the first one must first be acknowledged. This is what causes the delay, if it were two separate device you would not have a delay.

What I would advise here is to put both endpoints in the same group, this will make sure they always turn on at exactly the same time.

Imaginous, 2025-05-22:
Do you know if Deconz handles it differently?

With Deconz it was fast. Will test the XIAO M24 tomorrow.
Just out of interest.

Also noticed that the UZG-01 reboots between 11 hours upto 2 days.  I'm not the only one with this issue. 

Keep up the good work.

Koenkk, 2025-05-25:
Yes, deconz has no queueing

raoulbhatia, 2025-07-25:
I think I was facing a similar issue and had to stay in 2.2.0 for a long time.

I recently upgraded to Zigbee2MQTT version
    [2.5.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.5.1) commit: 5063c7d3ad88f9ed651dc36fd9e583ff452e18da

Everything seems to work fine now for my small use case (mostly sensors).

closed 2025-10-02 by github-actions[bot]