Koenkk/zigbee2mqtt #27289: Theme installation breaks Configuration.yaml with !include_dir_merge_named

levexis, 2025-05-02:
### What happened?

After installing a theme I found zigbee2mqtt would not start. It would seem that homeassistant is updating files called configuration.yaml. I'm not sure where to log this but it would seem the way to prevent this is to not to use the same name. The line, which blows up is:

homeassistant:
  packages: !include_dir_merge_named themes/

Log error:

Starting Zigbee2MQTT without watchdog.
/app/node_modules/.pnpm/js-yaml@4.1.0/node_modules/js-yaml/lib/loader.js:183
return new YAMLException(message, mark);
^
YAMLException: unknown tag !<!include_dir_merge_named> (35:42)

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

20221226

### Adapter

Zonoff USB 3 stick

### Setup

Add-on

### Debug log

Starting Zigbee2MQTT without watchdog.
/app/node_modules/.pnpm/js-yaml@4.1.0/node_modules/js-yaml/lib/loader.js:183
return new YAMLException(message, mark);
^
YAMLException: unknown tag !<!include_dir_merge_named> (35:42)

Nerivec, 2025-05-02:
HA configs should not be present in the `configuration.yaml` for Z2M. You seem to have mixed the two files together.

levexis, 2025-05-02:
I didn't touch anything, these were auto-inserted by HACS I believe. First I knew was when Z2M stopped working and was unable to restart.