Koenkk/zigbee2mqtt #27292: Invalid option for select 'Celsius' (valid options: ['celsius', 'fahrenheit'])

vlad36N, 2025-05-02:
### What happened?

It's auto discovery.
On Zigbee2MQTT devices I can see json values per device:
{
"battery": 100,
"humidity": 40.6,
"linkquality": 142,
"temperature": 26.5,
"temperature_units": "Celsius",
"update": {
"installed_version": 8960,
"latest_version": 8960,
"state": "idle"
}
}
It's capital in Celsius. 

Errors:
Logger: homeassistant.components.mqtt.select
Source: components/mqtt/select.py:125
integration: MQTT (documentation, issues)
First occurred: 2:39:47 AM (201 occurrences)
Last logged: 12:00:03 PM

Invalid option for select.living_temp_h_temperature_units: 'Celsius' (valid options: ['celsius', 'fahrenheit'])
Invalid option for select.bedroom_temp_h_temperature_units: 'Celsius' (valid options: ['celsius', 'fahrenheit'])
Invalid option for select.small_bathroom_temp_h_temperature_units: 'Celsius' (valid options: ['celsius', 'fahrenheit'])
Invalid option for select.office_temp_h_temperature_units: 'Celsius' (valid options: ['celsius', 'fahrenheit'])
Invalid option for select.full_bathroom_temp_h_temperature_units: 'Celsius' (valid options: ['celsius', 'fahrenheit'])

### Zigbee2MQTT version

2.3.0-1

### Adapter firmware version

20240914

### Adapter

UZG-01

### Setup

Add-on on HA OS running on PI 4.
Core 2025.4.4
Supervisor 2025.04.1
Operating System 15.2
Frontend 20250411.0

### Debug log

Logger: homeassistant.components.mqtt.select
Source: components/mqtt/select.py:125
integration: MQTT (documentation, issues)
First occurred: 2:39:47 AM (202 occurrences)
Last logged: 12:07:22 PM

Invalid option for select.living_temp_h_temperature_units: 'Celsius' (valid options: ['celsius', 'fahrenheit'])
Invalid option for select.bedroom_temp_h_temperature_units: 'Celsius' (valid options: ['celsius', 'fahrenheit'])
Invalid option for select.small_bathroom_temp_h_temperature_units: 'Celsius' (valid options: ['celsius', 'fahrenheit'])
Invalid option for select.office_temp_h_temperature_units: 'Celsius' (valid options: ['celsius', 'fahrenheit'])
Invalid option for select.full_bathroom_temp_h_temperature_units: 'Celsius' (valid options: ['celsius', 'fahrenheit'])

chris-1243, 2025-05-02:
It is due to those two PRs

https://github.com/Koenkk/zigbee-herdsman-converters/pull/9177

https://github.com/Koenkk/zigbee-herdsman-converters/pull/9180

There was a syntax error which had to be corrected.

Open the expose tab of your SNZB-02D, wake up your device and select `celsius` or `fahrenheit` according to your use and you will be good to go

vlad36N, 2025-05-02:
Changed. Thank you.

chpego, 2025-05-02:
The same goes for me, and I made the changes in the Expose section of each device, fortunately I don't have many of them.

it doesn't seem to be mentioned in the latest release note this modification

thanks @chris-1243 for finding the PR

chris-1243, 2025-05-02:
@chpego 

> thanks @chris-1243 for finding the PR

Well, I did the correction as I added the SNZB-02LD and I have been advised by **Koenkk** to modify this part of the code for this specific device and I did apply this small fix for the SNZB-02D as well.

It is unfortunatelly not shown in the release note of 2.3.0

sdotter, 2025-05-18:
yep same here

chris-1243, 2025-05-18:
@sdotter 

The issue have been adressed previously in this thread as well as how to solve it. I kindly invite you to read the post above.

xathon, 2025-06-01:
Just wanted to note that for me with the Bosch Radiator thermostat II, the setting was not in the Exposes tab, but in Settings (specific).

Loic691, 2025-06-10:
Hi
I don't find where we can change the unit in the expose tab of device in Z2M.
Can you help ?

ptr727, 2025-08-06:
I also noticed this warning in my logs:

```
Logger: homeassistant.components.mqtt.select
Source: components/mqtt/select.py:125
integration: MQTT (documentation, issues)
First occurred: 7:02:56 AM (15 occurrences)
Last logged: 8:30:20 AM

Invalid option for select.sonoff_temp_555f6b_temperature_units: 'Celsius' (valid options: ['celsius', 'fahrenheit'])
```

Seems that rather than having to manually change the units via config to correct teh case,\ that Z2M should report the units in a normalized form, i.e. if acceptable units need to be lowercase, then report in `celcius` vs. `Celcius`.