Koenkk/zigbee2mqtt #27297: Ingress not working if HTTPS / auth enabled

ZTHawk, 2025-05-02:
### What happened?

When having either HTTPS and/or Auth_token enabled then the Zigbee2MQTT instance is not reachable via ingress.
* My 'configuration.yaml' (located in HA/zigbee2mqtt) looks like this:
```
frontend:
  port: 8099
  ssl_cert: /ssl/zigbee2mqtt.crt
  ssl_key: /ssl/zigbee2mqtt.key
  auth_token: '!secret.yaml z2m_secret_pw'
```

Site cannot be displayed. I get "502: Bad Gateway" and a popup:
```
Zigbee2MQTT

The add-on seems to not be ready, it might still be starting. Do you want to try again?
```

Related to #23883 and #23648 

### What did you expect to happen?

I can access the frontend even if HTTPS and/or auth_token is enabled.

### How to reproduce it (minimal and precise)

1. Enable SSL and/or 'Auth token'.
2. Naviagate to Zigbee2MQTT frontend via Home Assistant

### Zigbee2MQTT version

2.3.0-1

### Adapter firmware version

EmberZNet 7.4.5 [GA]

### Adapter

SONOFF Dongle Plus V2

### Setup

HA Addon and HA in proxmox

PacmanForever, 2025-05-02:
Following the issue states...

ZTHawk, 2025-05-03:
> Following the issue states...

Sry, I do not understand your comment

PacmanForever, 2025-05-03:
> > Following the issue states...
> 
> Sry, I do not understand your comment

I was just saying that so that I can get new comments. To keep track of the comments.

ZTHawk, 2025-07-02:
v2.5.1 is now showing the UI, but everything empty.
Eg: [image]

SSL is working. Having "auth_token" defined leads to the shown error.
Using the standalone URL "<HA IP>:8099" is showing everything just fine.

ZTHawk, 2025-09-02:
Still applicable (v2.6.1)

avbor, 2026-02-23:
Up ^

ZTHawk, 2026-09-25:
Still applicable in v2.14.1