Koenkk/zigbee2mqtt #27305: [New device support]: ZMS-206US-4 (3 gangs version is already supported)

K-Benaissa, 2025-05-03:
### Link

https://www.zigbee2mqtt.io/devices/ZMS-206US-3.html

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xa4c138f0f55671b5","nwkAddr":19134,"manufId":4417,"manufName":"_TZE204_wwaeqnrf","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0016  /\u0013\u0017  /e\u0018  /\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1746259462523}

### Zigbee2MQTT version

2.3.1

### Comments

Please the 3 switch version is already supported, is it possible to add 4 switch version (US model)
Best regards,

### External definition

```shell
No loaded
```

### What does/doesn't work with the external definition?

No supported

Koenkk, 2025-05-04:
Added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/9279

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-05-04 by Koenkk

K-Benaissa, 2025-05-04:
> Added in [Koenkk/zigbee-herdsman-converters#9279](https://github.com/Koenkk/zigbee-herdsman-converters/pull/9279)
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

@Koenkk 

Great, I'll test it and post my feedback.
Thank you very much.

Koenkk, 2025-05-05:
Only reinstall edeg

K-Benaissa, 2025-05-16:
But there is not the name for the fourth switch (only 3 Names).
Thank you for your time.

Koenkk, 2025-05-16:
What do you mean with "there is not the name" here?

K-Benaissa, 2025-05-17:
@Koenkk

ZMS-206US-4  worked fine with 4 gangs,
It has 4 gangs (DISPLAY NAMES) you have integrated only the abality to change 3 of the 4 names (like 3 gangs version)
the is no field to indicate the forth name in Z2MQTT. It doesn't exposes : "name_l4"

mpekin1, 2025-05-18:
> [@Koenkk](https://github.com/Koenkk)
> 
> ZMS-206US-4 worked fine with 4 gangs, It has 4 gangs (DISPLAY NAMES) you have integrated only the abality to change 3 of the 4 names (like 3 gangs version) the is no field to indicate the forth name in Z2MQTT. It doesn't exposes : "name_l4"

I also have the same issue. As per the photo name_I4 is not settable

[image]

ntk9, 2025-05-22:
I have problem with changing the names to any other language that is not english, this is the list of all the language supported

chinese, english, french, garman, russian, portuguese, korean, japanese, vietnamese, arabic, hebrew, spanish

Koenkk, 2025-06-03:
@tlemmons added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Lucianosouza99, 2025-09-05:
Hello, @Koenkk  you please help me? I bought the 3-channel switch but Z2M shows it as not supported.

[image]

Koenkk, 2025-09-06:
@Lucianosouza99 added!

Lucianosouza99, 2025-09-06:
@Koenkk thanks