Koenkk/zigbee2mqtt #27311: z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)

nblount, 2025-05-04:
### What happened?

Seems to be a connectivity issue between zigbee2mqtt and co-ordinator since upgrading to 2.3.0.
If selecting permit join it says it is enabled but does not show the count-down and devices don't appear to join, but later join does work, it is very inconsistant.
Large number of devices show as offline (+- 1/3 (25 devices)) but has been has high as 90% requiring a restart of zigbee2mqtt and sometimes the co-ordinator.

### What did you expect to happen?

Zigbee network to be a bit more stable.

### How to reproduce it (minimal and precise)

It appears if the slzb-06 is restarted then zigbee2mqtt connect but functions don't work as expected.

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

20240710

### Adapter

slzb-06

### Setup

Add-on to Home Assistant running under HA as a VM.

### Debug log

[08:52:23] INFO: Preparing to start...
[08:52:23] INFO: Socat not enabled
[08:52:23] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-05-04 08:52:24] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-05-04.08-49-03'
[2025-05-04 08:52:24] info: 	z2m: Logging to console, file (filename: log.log)
[2025-05-04 08:52:24] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-05-04 08:52:24] info: 	z2m: Starting Zigbee2MQTT version 2.3.0 (commit #unknown)
[2025-05-04 08:52:24] info: 	z2m: Starting zigbee-herdsman (4.0.0)
[2025-05-04 08:52:24] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":28910,\"extendedPanID\":[132,7,124,169,39,126,237,9],\"channelList\":[25],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"baudRate\":115200,\"path\":\"tcp://slzb-06.local:6638\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false,\"transmitPower\":20}}"'
[2025-05-04 08:52:24] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":28910,"extendedPanID":[132,7,124,169,39,126,237,9],"channelList":[25]},"serialPort":{"baudRate":115200,"path":"tcp://slzb-06.local:6638","adapter":"zstack"},"adapter":{"disableLED":false,"transmitPower":20},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-05-04 08:52:24] info: 	zh:zstack:znp: Opening TCP socket with slzb-06.local:6638
[2025-05-04 08:52:24] info: 	zh:zstack:znp: Socket connected
[2025-05-04 08:52:24] info: 	zh:zstack:znp: Socket ready
[2025-05-04 08:52:24] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-05-04 08:52:24] debug: 	zh:zstack:unpi:writer: --> buffer [239]
[2025-05-04 08:52:25] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-05-04 08:52:25] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-05-04 08:52:25] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-05-04 08:52:31] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-05-04 08:52:31] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-05-04 08:52:37] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-05-04 08:52:37] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-05-04 08:52:43] error: 	z2m: Error while starting zigbee-herdsman
[2025-05-04 08:52:43] error: 	z2m: Failed to start zigbee-herdsman
[2025-05-04 08:52:43] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-05-04 08:52:43] error: 	z2m: Exiting...
[2025-05-04 08:52:43] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:114:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/controller/controller.ts:133:29)
    at Zigbee.start (/app/lib/zigbee.ts:70:27)
    at Controller.start (/app/lib/controller.ts:104:13)
    at start (/app/index.js:149:5)

Some other logs from Zigbee2mqtt log tab:
{"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
error 2025-05-04 09:40:13zh:controller:device: Read response to 0xa4c138894f792ec0 failed (ZCL command 0xa4c138894f792ec0/1 genTime.readRsp({"localTime":799666729}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":147,"writeUndiv":false}) failed (Data request failed with error: 'Timeout' (9999)))
error 2025-05-04 09:40:33zh:controller:device: Read response to 0xa4c138894f792ec0 failed (ZCL command 0xa4c138894f792ec0/1 genTime.readRsp({"localTime":799666730}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":147,"writeUndiv":false}) failed (Data request failed with error: 'Timeout' (9999)))
error 2025-05-04 09:40:43zh:controller:device: Read response to 0xa4c138894f792ec0 failed (ZCL command 0xa4c138894f792ec0/1 genTime.readRsp({"localTime":799666736}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":147,"writeUndiv":false}) failed (Data request failed with error: 'Timeout' (9999)))
error 2025-05-04 09:57:29z2m: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: '--> 'SREQ: ZDO - PERMIT_JOINING_REQUEST - 0ffcfffe01' failed with status '(0x11: BUFFER_FULL)''

Meganitrospeed, 2025-05-05:
Same issue here

nblount, 2025-05-05:
Can't see how that is going to help but I will try it. The amount of time Zigbee remains stable seems very variable, it has been connected for 16 hours but most devices are showing as offline.

nblount, 2025-05-05:
Just changing the multi-threaded setting did not allow zigbee2mqtt to connect I have to power cycle the co-ordinator.

nblount, 2025-05-05:
Keep seeing errors like this:
warning 2025-05-05 08:31:00z2m: Failed to ping 'Garage Socket 5' (attempt 1/1, ZCL command 0x00158d00025752ab/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":11083,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":190,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,32,0,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')))

If I try and enable join mode I also see BUFFER_FULL

nblount, 2025-05-05:
I have regenerated the pan_id etc and zigbee2mqtt is up and running but I will have to reconnect all the devices, will see how we go. Will take a while to get all the 80+ devices linked again and some are not playing nice repairing.

Maximus0808, 2025-05-06:
Hello,
I have the same problem.
I've tried the solutions listed in the instructions. Unfortunately, no success. I've uninstalled/installed, deleted folders...

DELL Mini-PC
ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus 
Firmware CC1352P2_CC2652P_launchpad_coordinator_20240710

Error: Bad Gateway 502

```
Welcome to the Home Assistant command line.

System information
  IPv4 addresses for eno1:  192.168.178.180/24
  IPv6 addresses for eno1:  2001:9e8:e0:6900:e665:b7c5:f3c6:1cbd/64, fe80::49e3:f3b3:aa71:d7f2/64

  OS Version:               Home Assistant OS 15.2
  Home Assistant Core:      2025.4.4

  Home Assistant URL:       http://homeassistant.local:8123
  Observer URL:             http://homeassistant.local:4357
➜  ~ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx    1 root     root            13 May  6 07:25 usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8e88adc0ea4eef1181fc46b3174bec31-if00-port0 -> ../../ttyUSB0
➜  ~ 
➜  ~ 
➜  ~ test -w /dev/ttyUSB0 && echo success || echo failure
success
➜  ~ 
```

Konfig:

```
version: 4
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto:1883
  user: addons
  password: yahw4fi0eey5Ootheesoo1lug7faipheeb1Iuweboovej5eeJau7avohSh9waivo

serial:
  port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8e88adc0ea4eef1181fc46b3174bec31-if00-port0
  adapter: zstack

advanced:
  log_level: info
  channel: 11
  network_key:
    - 65
    - 192
    - 169
    - 63
    - 75
    - 100
    - 17
    - 115
    - 49
    - 161
    - 248
    - 130
    - 63
    - 202
    - 85
    - 39
  pan_id: 49781
  ext_pan_id:
    - 117
    - 84
    - 198
    - 165
    - 116
    - 188
    - 119
    - 176

frontend:
  enabled: true
  port: 8099

homeassistant:
  enabled: true
```

2mqtt Protokoll
```
[07:41:16] INFO: Preparing to start...
[07:41:16] INFO: Socat not enabled
[07:41:16] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-05-06 07:41:17] info: 	z2m: Logging to console, file (filename: log.log)
[2025-05-06 07:41:17] info: 	z2m: Starting Zigbee2MQTT version 2.3.0 (commit #unknown)
[2025-05-06 07:41:17] info: 	z2m: Starting zigbee-herdsman (4.0.0)
[2025-05-06 07:41:17] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8e88adc0ea4eef1181fc46b3174bec31-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-05-06 07:41:17] info: 	zh:zstack:znp: Serialport opened
[2025-05-06 07:41:17] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-05-06 07:41:19] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-05-06 07:41:37] error: 	z2m: Error while starting zigbee-herdsman
[2025-05-06 07:41:37] error: 	z2m: Failed to start zigbee-herdsman
[2025-05-06 07:41:37] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-05-06 07:41:37] error: 	z2m: Exiting...
[2025-05-06 07:41:37] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:114:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/controller/controller.ts:133:29)
    at Zigbee.start (/app/lib/zigbee.ts:70:27)
    at Controller.start (/app/lib/controller.ts:104:13)
    at start (/app/index.js:149:5)
```

nblount, 2025-05-07:
This is what I have done:
1. Stop zigbee2mqtt
2. Upgrade Zigbee Co-ordinator to firmware 20250321 (https://github.com/Koenkk/Z-Stack-firmware/discussions/545) how to upgrade and where to get the firmware will depend on your device mine is a SLZB-06 and you just browse to it and request a list of firmware.
3. Restart Co-ordinator (power cycle is best)
4. Edit the zigbee2mqtt configuration.yaml and change these 3 entries:  
  network_key: GENERATE
  pan_id: GENERATE
  ext_pan_id: GENERATE
5. Restart zigbee2mqtt
6. Hopefully it all starts but you will likely have no devices listed. Click on Allow Join in zigbee2mqtt and go arround repairing the devices, starting with the ones nearest to the co-ordinator seems to work best. Devices will return with the same details they had before.

So far the network has been stable (over a day so far, nothing has go offline), I still have a few devices mainly power monitoring plugs that refuse to re-connect which I need to try and figure out.

nblount, 2025-05-14:
Happened again after a couple of days started seeing BUFFER_FULL errors in the logs and everything showing as offline. Hard reboot of the co-ordinator and most things came back online. Maybe need to look at the co-ordinator firmware but this only started after the latest zigbee2mqtt upgrade.

Maximus0808, 2025-05-14:
I used the firmware 20250321 and then set up Z2M as described above.
However, I only have one Zigbee device in operation so far. I'm still a beginner and am still building the network.
I think it's likely that at some point there will be too many devices for one stick. There may be differences. Some Zigbee hubs can manage more devices, others fewer.

closed 2025-07-21 by github-actions[bot]