Koenkk/zigbee2mqtt #27312: Ynoa 8718801528334 - direct binding not working

mrmaximas, 2025-05-04:
### What happened?

I bind this device https://www.zigbee2mqtt.io/devices/8718801528334.html#ynoa-8718801528334 to the Aqara bulb (trying Ikea too), but direct binding not working, the bulb not controlable.
The manufacturer say (literally) ‘direct connection should be possible, at least with the Ynoa bulb’.
How do I fix it?

[image]
Thanks!

### Zigbee2MQTT version

2.3.0-dev commit: ff823dba

### Adapter firmware version

20240710

### Adapter

CC2252P7

### Setup

Mac OS M1

mrmaximas, 2025-05-09:
Hello @Koenkk  can you help me?

Koenkk, 2025-05-10:
Any chance you could sniff the Zigbee traffic when the remote is bound to the bulb and it should be sending commands to it?

mrmaximas, 2025-05-10:
@Koenkk thanks for the answer. I don't have Ynoa bulb but i have the Ikea and Aqara bulbs + sniffer dongle. It's ok?

Koenkk, 2025-05-11:
yes!

mrmaximas, 2025-05-12:
Here is two sniffs. 
Binding Ynoa button 0x043D https://www.zigbee2mqtt.io/devices/8718801528334.html#ynoa-8718801528334
to 
Aqara bulb 0xEA40 https://www.zigbee2mqtt.io/devices/ZNLDP13LM.html#aqara-znldp13lm 
And button press with on/off, then two button holds with change brightness.
It looks nice, but it doesn't work.
[Ynoa.zip](https://github.com/user-attachments/files/20165362/Ynoa.zip)

mrmaximas, 2025-05-12:
@Koenkk yes! it's works! Thanks! 
I have these 6 new buttons, they all seem to send messages to a group with the same number. Does this mean I can only use one button?

mrmaximas, 2025-05-12:
@Koenkk trying add to group https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-L_1.html#lonsonho-qs-zigbee-d02-triac-l_1 - only on/off working, without brightness.

Koenkk, 2025-05-13:
You can try binding them to a different group, not sure if that works or if the device uses a fixed one.

closed 2025-05-20 by mrmaximas