Koenkk/zigbee2mqtt #27313: Ubisys C4 - "Failed to find an endpoint which supports the OTA cluster" when checking for firmware update

gongsearch, 2025-05-04:
### What happened?

Clicking "Check for firmware update" on a newly paired C4 gives:

"Failed to check if update available for 'C4 Test' ([0x001fee000000b1e0 | C4 (5504)] Failed to find an endpoint which supports the OTA cluster)"

There is actually no cluster named genOta for this device.

### What did you expect to happen?

As the currently installed firmware is 2.4 and 2.5 is available I would expect to be questioned whether I want to upgrade the firmware. 

### How to reproduce it (minimal and precise)

Pair Ubisys C4, click "Check for firmware update"

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

8.0.2 [GA]

### Adapter

slzb-07mg24 / EmberZNet 

### Setup

x86_64, official docker image

chris-1243, 2025-05-14:
`genOta` is on endpoint 232. I have a C4 (5504) and it is finding latest firmware (2.6.0).

Maybe remove the device, restart Z2M and pair it again.

gongsearch, 2025-05-14:
Thanks for your suggestion, just tried it. Sadly with the same result. Endpoint 232 looks like this for me (before and after re-pairing):

[image]

chris-1243, 2025-05-15:
Would you mind to share the `database.db` entry of your device ? I checked both on `latest-dev` and `latest-stable` release (2.3.0) and everything is fine on my side.

I tried with the exact same setup as yours (SLZB07MG24 - 8.0.2) and on Docker. The only difference I might find is the firmware on the Ubisys C4. Mine is on 2.5.0. It used to be on lower firmware revision and I did not encounter any problem to update it even if it is not needed to update a working device as it may break its compatibility with Z2M.

gongsearch, 2025-05-15:
The technical specification (https://www.ubisys.de/downloads/ubisys-c4-technical-reference.pdf page 25) does not mention any change and lists the ota-cluster under endpoint 232.

I tried adding cluster 25 (0x0019) in endpoint 232 in datbase.db - now I do no longer get the error-message about the missing ota-cluster but the firmware-upgrade-check just fails silently. Was worth a try.

Last resort now is trying another zigbee-stick. Will try that this weekend.

chris-1243, 2025-05-15:
I would also contact Ubisys and ask why there is no `genOta` on your device.

chris-1243, 2025-05-16:
This thread might be interesting for you

https://github.com/Koenkk/zigbee-herdsman-converters/issues/9356

cmaschke, 2025-05-19:
> The technical specification (https://www.ubisys.de/downloads/ubisys-c4-technical-reference.pdf page 25) does not mention any change and lists the ota-cluster under endpoint 232.
>
> I tried adding [...]

This is/was indeed a problem and has been fixed with the latest firmware update - which unfortunately does not reach your device.
Depending on the OTA server used, it may work without the cluster (as with our Gateway G1), but this is not a guarantee.
Another option is to use Zigbee Direct (Bluetooth) to update the device - please get in touch with our support department and we will sort things out.

closed 2025-07-27 by github-actions[bot]