Koenkk/zigbee2mqtt #27321: [New device support]: Aubess TMZ02  wrongly discovered as_TZ3000_lmlsduws

franiu222, 2025-05-05:
### Link

https://pl.aliexpress.com/item/1005008059596235.html?spm=a2g0o.productlist.main.11.60c7ZRu9ZRu9kX&algo_pvid=dc2a8d7f-f160-4f8b-8f90-6e591d824c8f&algo_exp_id=dc2a8d7f-f160-4f8b-8f90-6e591d824c8f-5&pdp_ext_f=%7B%22order%22%3A%2218%22%2C%22eval%22%3A%221%22%7D&pdp_npi=4%40dis%21PLN%2120.79%2120.79%21%21%2138.66%2138.66%21%4021038e6617464292033224354eebc4%2112000043490965700%21sea%21PL%21843549711%21X&curPageLogUid=bhmGcXytCPx5&utparam-url=scene%3Asearch%7Cquery_from%3A

### Database entry

{"id":96,"type":"Router","ieeeAddr":"0xa4c138ac6b8fbcd7","nwkAddr":44201,"manufId":4417,"manufName":"_TZ3000_lmlsduws","powerSource":"Mains (single phase)","modelId":"TS0002","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAAAgAA"}},"genBasic":{"attributes":{"65503":"*\u0001\u0000\u0000\u0012","65506":56,"65508":0,"65534":0,"modelId":"TS0002","manufacturerName":"_TZ3000_lmlsduws","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"moesStartUpOnOff":2,"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x6c5cb1fffe29f124","endpointID":1}],"configuredReportings":[],"meta":{"scenes":{"0_2":{"name":"Scene 0","state":{}}}}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x6c5cb1fffe29f124","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1746427648903}

### Zigbee2MQTT version

2.2.1

### Comments

Issue similat to https://github.com/Koenkk/zigbee2mqtt/issues/21926  WHD02https://www.zigbee2mqtt.io/devices/WHD02.html wrongly reported as 
TS0002_basic https://www.zigbee2mqtt.io/devices/TS0002_basic.html#tuya-ts0002_basic

### External definition

```shell
Not neded
```

### What does/doesn't work with the external definition?

.

Koenkk, 2025-05-05:
The WHD02 is a 1 gang switch, if I understand correctly your device has 2 gangs?

franiu222, 2025-05-07:
Sorry, my mistake. My device has 2 gangs.

Koenkk, 2025-05-07:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) tomorrow and in the next release which is every 1st of the month.

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `TMZ02.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

franiu222, 2025-07-02:
Sory but i have problem with adding photo. I'm not used to use github (I know its hard to belive). Can you add photo that i prepared and edded below?
[image]

closed 2025-09-09 by github-actions[bot]

andrei-lazarov, 2025-09-19:
I submitted a PR with the correct image