Koenkk/zigbee2mqtt #27322: Not responding Zigbee devices after upgrade to 2.3.0

petronovak, 2025-05-05:
### What happened?

On version 2.3.0 suddenly, lost connection with most devices in my Zigbee network. They shown online, but don't work any set or read operations.
Devices ([Tuya TS0207_repeater](https://www.zigbee2mqtt.io/devices/TS0207_repeater.html#tuya-ts0207-repeater), [Moes BRT-100-TRV](https://www.zigbee2mqtt.io/devices/BRT-100-TRV.html#moes-brt-100-trv), [Nous A1Z](https://www.zigbee2mqtt.io/devices/A1Z.html#nous-a1z)).
End devices not responding (directly connected to coordinator, routers, like [Tuya TS0207_repeater](https://www.zigbee2mqtt.io/devices/TS0207_repeater.html#tuya-ts0207-repeater), [Nous A1Z](https://www.zigbee2mqtt.io/devices/A1Z.html#nous-a1z))

### What did you expect to happen?

Work as usual

### How to reproduce it (minimal and precise)

I tried to remove my repeater TS0207 (https://www.zigbee2mqtt.io/devices/TS0207_repeater.html). But, now, it can't join it to my network, and have error log message.

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

Add-on on Home Assistant (RaspberryPI 4B)

### Debug log

[2025-05-05 12:51:43] error: 	zh:controller:device: Read response to 0xa4c1385f8512df74 failed (ZCL command 0xa4c1385f8512df74/1 genTime.readRsp({"localTime":799764561}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":152,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))

Koenkk, 2025-05-05:
> MAC channel access failure' (225)))

This indicates wireless interference, see https://www.zigbee2mqtt.io/guide/faq/#common-error-codes

grasmuis1987, 2025-05-05:
I'm having the same issue after the latest update, most of my devices now disconnected and unable to pair them again.

warning 2025-05-05 20:33:08z2m: Failed to ping '0xb40ecfd02eac0000' (attempt 1/1, ZCL command 0xb40ecfd02eac0000/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))
warning 2025-05-05 20:33:11z2m: Failed to ping '0xb40ecfd03a910000' (attempt 1/1, ZCL command 0xb40ecfd03a910000/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))

Zigbee2MQTT version
2.3.0

Adapter
SF-Zig-3-Dongle-Plus

Setup
Add-on on Home Assistant (RaspberryPI 4B)

petronovak, 2025-05-06:
@Koenkk 
> > MAC channel access failure' (225)))
> 
> This indicates wireless interference, see https://www.zigbee2mqtt.io/guide/faq/#common-error-codes

Very strange. Because it worked fine for the last few months. And a few hours after the last update I noticed the problem. There have been no changes in the Zigbee or Wifi network for a very long time, there are no neighbors since I live in a private house.

grasmuis1987, 2025-05-06:
> [@Koenkk](https://github.com/Koenkk)
> 
> > > MAC channel access failure' (225)))
> > 
> > 
> > This indicates wireless interference, see https://www.zigbee2mqtt.io/guide/faq/#common-error-codes
> 
> Very strange. Because it worked fine for the last few months. And a few hours after the last update I noticed the problem. There have been no changes in the Zigbee or Wifi network for a very long time, there are no neighbors since I live in a private house. When I removed one of the devices from the Zigbee network, I can't add it back. The distance to my coordinator is less than 1 meter. Disconnecting a half of devices didn't help also.

I have also been running it for about 3 years without any issues and changes except the update. After the update, none of my devices are working, and they all stopped reporting back at the same time. So don't think it's related to signal or interference. I will revert to a backup before the update and see what happens.

grasmuis1987, 2025-05-06:
I have reverted back to version 2.2.1 and all seems to be back up and running again.

Koenkk, 2025-05-06:
Could you also try the [20250403](https://github.com/Koenkk/Z-Stack-firmware/discussions/546) firmware?

D3SL, 2025-05-08:
I had the same issue (#26654) back at 2.1.3. Thankfully after the harsh lesson of 2.0 I always make a backup of HA in proxmox before attempting any new Z2M version.

rcarmo, 2025-05-11:
Came here to say that after moving from 2.2.1 to 2.3.0 I have Sonoff ZBMINIL2 devices dropping off the network. I upgraded my coordinator to 20240710 (was a bit overdue), but 2.2.1 made a couple of them come back almost immediately. The rest I'm trying to re-pair now.

petronovak, 2025-05-15:
I disconnected power from my Coordinator (SLZB-06), it's connected with Home Assistant (RaspberryPI 4B) via Ethernet. Removed antenna from SLZB-06. Connected power to SLZB-06. Restarted Zigbee2MQTT addon. But it's didn't help. Then I disconnected power from SLZB-06 again. Connected back antenna to SLZB-06 and connected power to it. Restarted Zigbee2MQTT addon again and my Zigbee network started back to life (one Zigbee router I re-paired). Looks like, now it works fine. Thanks all for help. If no objections, we can close this issue. Thanks @Koenkk for your work