Koenkk/zigbee2mqtt #27328: Hive SLT6 Stuck In Searching After Pairing To Z2M

b177y, 2025-05-05:
### What happened?

After pairing the Hive SLR2c and Hive SLT6 to Zigbee2Mqtt following the instructions in the SLT6 docs page, the thermostat stays stuck in searching until it times out. When pairing the thermostat with the receiver in standalone mode, it pairs successfully. When pairing with Z2M using the instructions, both devices pair to Z2M successfully, but not to each other, meaning the thermostat stays stuck in 'searching', and the receiver stays blinking orange waiting for a thermostat to connect. On most attempts, the interview usually fails on the first time for both devices, with the error 'can not get active endpoints', but succeeds on the next attempt.

I have had this setup working before and had to re-pair it once before. Both the initial and repair were painful and took many attempts, however I've been trying repeatedly for the last day and still no luck. I have also replaced the batteries in the receiver with new batteries.

I'm wondering if a possible workaround would be to use the bind functionality in Z2M to manually 'pair' the thermostat to the receiver. There might be more to it than that, and I've had a play around with binding but no luck so far. If anyone else has this set up and can see the bindings that are set up, any help is appreciated!

### What did you expect to happen?

I expected that after following the pairing instructions, the devices would pair both to Z2M and to each other.

### How to reproduce it (minimal and precise)

Zigbee2MQTT version
    [2.3.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.3.0) commit: [aec59cb774aa09a793ae41cd60e6a5cef15b293c ](https://github.com/Koenkk/zigbee2mqtt/commit/aec59cb774aa09a793ae41cd60e6a5cef15b293c)

Coordinator type
    EmberZNet

Coordinator revision
    7.4.4 [GA]

Coordinator IEEE Address
    0xe0798dfffe7190c3

Frontend version
    0.9.10

zigbee-herdsman-converters version
    23.36.0

zigbee-herdsman version
    4.0.0

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

7.4.4 [GA]

### Adapter

ember

### Setup

Z2M is running on docker with docker-compose, on Pi4 (Ubuntu Server)

Linux ubuntu 5.15.0-1076-raspi #79-Ubuntu SMP PREEMPT Wed Mar 26 06:11:54 UTC 2025 aarch64 aarch64 aarch64 GNU/Linux
Ubuntu 22.04.3 LTS

### Debug log

[z2m-debug.log](https://github.com/user-attachments/files/20041199/z2m-debug.log)

closed 2025-07-12 by github-actions[bot]

b177y, 2025-07-19:
Confirming I am still getting this issue with [2.5.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.5.1) commit: 5063c7d3ad88f9ed651dc36fd9e583ff452e18da

b177y, 2025-07-19:
Sometimes getting:
```
z2m: Failed to configure '0x001e5e09027d7641', attempt 2 (Error: Bind 0x001e5e09027d7641/5 genBasic from '0xe0798dfffe7190c3/1' failed (Delivery failed for '34795'.) at EmberOneWaitress.deliveryFailedFor (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:100:31) at EmberAdapter.onMessageSent (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:458:42) at Ezsp.emit (node:events:518:28) at Ezsp.ezspMessageSentHandler (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5076:14) at Ezsp.callbackDispatch (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:734:26) at Ezsp.onAshFrame (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:387:22) at UartAsh.emit (node:events:518:28) at Immediate.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1192:45) at processImmediate (node:internal/timers:485:21))
```
When ettempting to pair hive receiver

b177y, 2025-07-19:
I have now downgraded to 1.36.0 and I can now pair the Hive receiver and thermostat, with them both pairing together

fspv, 2025-08-03:
update: I've deleted the receiver from zigbee2mqtt and re-added receiver and thermostat, and everything works now

colinmeinke, 2026-10-24:
**Possible cause and fix: dimmers echoing announces (SLR1c + SLT6 linked once they were powered off)**

Setup: Zigbee2MQTT 2.14.2, `ember` adapter, Sonoff Dongle-E (EmberZNet 7.4.4), Hive SLR1c + SLT6, no Hive hub. Same symptom as this issue: both joined and interviewed fine, but the thermostat stayed on "Searching" and the receiver kept flashing amber.

What the debug log showed: while searching, the SLT6 announces every ~30 s. Every time, within 1–2 s, 4–6 Enkin ZDM150 dimmers (routers) broadcast 6–8 device announces of their own. Z2M then reads each dimmer's state "after reconnect", and many of those reads time out. Nothing else announced all day; the burst only came in response to another device's announce. The log of a failed attempt a year earlier (1.42.0) shows the same thing: 99% of the dimmer announces came within 3 s of a thermostat announce.

Fix: with the thermostat still searching, I switched off the lighting circuits (the dimmers' power). Its next two announces went out with no echo. Within about a minute the receiver reported the thermostat's temperature, the thermostat showed "Pairing successful!" and the receiver went solid green. No rejoin was needed, and it stayed linked when the power came back.

It's one success, so treat it as a lead rather than proof. It matches a [Home Assistant forum report](https://community.home-assistant.io/t/hive-thermostat-zigbee2mqtt/1028490) where removing two Candeo dimmers fixed the same symptom.

Tip: count `announced itself` lines for every device, not just the Hive ones. The echo is easy to miss if you only filter on the new device's address.