Koenkk/zigbee2mqtt #27330: [Wrong device]:

s3frank, 2025-05-05:
### Link

docs/devices/TS130F.md

### Model

SC500ZB V4 is recognized as TS130F

### Description

Documentation needs update on Calibration

### Vendor

LoraTap

### Picture (link)

https://www.zigbee2mqtt.io/devices/SC500ZB.html

### Database entry

I don't know how to get this.

### Notes

To use this you need to set calibration twice, once for the up state and once for the down state. I was not able to do this in the UI of Z2M for the device but only by using Dev Console. The calibration procedure in the documentation is not complete / incorrect and will only configure one of the two calibration values.

Here is the zigbee commands I send to make it work:
{"write":
	{"cluster":"closuresWindowCovering",
	"payload":
		{
		"moesCalibrationTime": 320
		}
	}
}

I needed to do that to two different end points per device.
Endpoint 1 and 2. Not sure which is for up or down calibration time but for me setting them the same was fine.
So publishing via mqtt to: 

basetopic/loratap_shutter/1/set
basetopic/loratap_shutter/2/set

Once you do that the devices are super reliable to use.

If the UI would recognize this and enable the manual setting for both up and down calibration time it would be great.

The 320 used in my example translates into 32 seconds.

juagicre, 2025-05-22:
With already 9 SC500ZB calibrated, I cannot calibrate anymore any new one... and the calibrated ones cannot be recalibrated. Not sure if I'm doing something different or z2m changed... also i've noticed that the calibration documentation is missing nowadays... it smells to me something changed the way z2m is dealing with those devices... hopefully there is a way to get the calibration time set in some way... any ideas?

marcpuig, 2026-08-27:
Same module here (LoraTap, detected as TS130F). It's very buggy — don't buy it. I'll be replacing them.

Open / Close / Stop work if I set the calibration time to the real full travel time: the reported position then tracks correctly.

Setting a position does not. Slider, numeric field and the 25% step buttons all fail. Those go through goToLiftPercentage (0–100). Open/close/stop are different Zigbee commands (upOpen / downClose / stop), which this firmware actually implements.

If I set calibration time to 1/4 of the full travel, 25% steps look like they work: 100% → 75% → 50% → 25% → 0%. But a jump from 100% to 0% only moves physically by one step (~75%), then the device reports 0%. Same if I press Close from 100%: it moves ~25% down and still shows 0%.

So the stored calibration time is treated as the duration of a single 25% step, not the full travel. You can have either stepwise 25% moves or a proper full open/close, not both.

Z2M also publishes the requested % immediately, so Home Assistant shows 0% while the blind is still around 75%.

I also had to write moesCalibrationTime on both endpoints 1 and 2 (value in tenths of a second), as in this issue. The UI calibration procedure only sets one of them.