Koenkk/zigbee2mqtt #27342: Smartwings Device does not show Battery for WM25L-Z

bud1979, 2025-05-06:
### What happened?

After I added in the device via discover, it has a entry in MQTT for Battery, but after 1 week, there is still no information from the shade on the battery level

I just get logs like:

```
2025-05-04.17-08-30/log.log:[2025-05-05 21:45:23] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen - Table', payload '{"battery":null,"linkquality":98,"position":25,"state":"OPEN"}'
2025-05-04.17-08-30/log.log:[2025-05-06 16:14:05] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen - Table', payload '{"battery":null,"linkquality":105,"position":0,"state":"OPEN"}'
2025-05-04.17-08-30/log.log:[2025-05-06 16:14:13] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen - Table', payload '{"battery":null,"linkquality":105,"position":0,"state":"CLOSE"}'
2025-05-04.17-08-30/log.log:[2025-05-06 16:15:51] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen - Table', payload '{"battery":null,"linkquality":105,"position":25,"state":"CLOSE"}'
2025-05-04.17-08-30/log.log:[2025-05-06 16:15:59] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen - Table', payload '{"battery":null,"linkquality":102,"position":25,"state":"OPEN"}'
2025-05-04.17-08-30/log.log:[2025-05-06 16:42:32] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen - Table', payload '{"battery":null,"linkquality":105,"position":0,"state":"OPEN"}'
2025-05-04.17-08-30/log.log:[2025-05-06 16:42:41] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen - Table', payload '{"battery":null,"linkquality":102,"position":0,"state":"CLOSE"}'
```

### What did you expect to happen?

To see the battery level available

### How to reproduce it (minimal and precise)

Just add SmartWings blind to the MQTT

### Zigbee2MQTT version

 2.3.0 commit: unknown

### Adapter firmware version

20240710

### Adapter

usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Run as module of Home Assistant in a VM

chris-1243, 2025-05-07:
How does the reporting of this device look like ? Any values ?

bud1979, 2025-05-08:
Hey Chris, not sure what you are looking for exactly so will try to add what I can:

In the state tab i get:
`{
    "linkquality": 102,
    "position": 0,
    "state": "CLOSE"
}`
In the exposes, i get this:

[image]

Let me know if I can provide anything else, and appreciate the help.

chris-1243, 2025-05-08:
Could you set the reporting as set in the picture

[image]

Before clicking on `Apply` wake up your device

And keep us informed if you see any change

bud1979, 2025-05-08:
I did try that, and got these errors about 30 seconds after clicking apply.  Before i did click apply i moved the blind up and down to wake it up and make sure it was responding:

[image]

And the two logs:

`error 2025-05-08 09:06:47z2m: Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ZCL command 0x94ec32fffeb15b55/1 genPowerCfg.configReport([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 61880 - 1 - 144 - 1 - 7 after 10000ms)'

info 2025-05-08 09:06:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure_reporting', payload '{"data":{},"error":"ZCL command 0x94ec32fffeb15b55/1 genPowerCfg.configReport([{\"attribute\":\"batteryPercentageRemaining\",\"minimumReportInterval\":3600,\"maximumReportInterval\":65000,\"reportableChange\":0}], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"reservedBits\":0,\"writeUndiv\":false}) failed (Timeout - 61880 - 1 - 144 - 1 - 7 after 10000ms)","status":"error","transaction":"rxg16-6"}'`

bud1979, 2025-05-08:
Using your thoughts there, i did go to Dev Console, and picked the following and it did pull the battery percentage:

[image]

chris-1243, 2025-05-08:
Is your battery fully charged?

bud1979, 2025-05-08:
Yes, it is fully charged, or very close if not.   I did move the blind up and down, would that work to wake it up?

Thanks for the help too.

chris-1243, 2025-05-08:
It should be enough. I did some search on Github and found several issue with this device and the device defintion has changed many times...

Could you try this external converter ?

In app/data folder, create a sub-folder `external_converters` and in this folder, add a new file WM25L-Z.mjs and add the code below

```typescript
import * as fz from "zigbee-herdsman-converters/converters/fromZigbee";
import * as tz from "zigbee-herdsman-converters/converters/toZigbee";
import * as exposes from "zigbee-herdsman-converters/lib/exposes";
import * as reporting from "zigbee-herdsman-converters/lib/reporting";

const e = exposes.presets;

export default {
    zigbeeModel: ["WM25/L-Z"],
    model: "WM25L-Z",
    vendor: "Smartwings",
    description: "Roller shade",
    fromZigbee: [fz.cover_position_tilt, fz.battery],
    toZigbee: [tz.cover_state, tz.cover_position_tilt],
    exposes: [e.cover_position(), e.battery()],
    meta: {battery: {dontDividePercentage: true}, coverInverted: true},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ["genPowerCfg", "closuresWindowCovering"]);
        await reporting.batteryPercentageRemaining(endpoint);
        device.powerSource = "Mains (single phase)";
        device.save();
    },
};
```

While done, restart Z2M and check the log to ensure it has been loaded correctly

bud1979, 2025-05-08:
Done, it was loaded.

[2025-05-08 13:16:02] info: 	z2m: Loaded external converter 'WM25L-Z.mjs'.

I tried the reporting idea again, but still same issue. but the battery is showing up, not sure if that was from my previous query, or the new config.

I will keep an eye on it.

chris-1243, 2025-05-08:
Check your log time to time to see if you have any errors from this device.

Please report them here and also if everything is fine. We will then advise to have a long term approach from your feedback

closed 2025-07-15 by github-actions[bot]