Koenkk/zigbee2mqtt #27345: [SONOFF SNZB-02D] Device print error since 2.3.0

mlepannetier, 2025-05-07:
### What happened?

Since 2.3.0 SONOFF SNZB-02D print error in domoticz logs

### What did you expect to happen?

No error printing

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

20220219

### Adapter

ZStack3x0

### Setup

Domoticz in docker mode
Zigbee2MQTT in docker mode

### Debug log

In domoticz : 
2025-05-07 15:34:27.285 Error: Zigbee2MQTT: Select device "0x3c2ef5fffebc6a92 (Temperature units)" doesn't have the option for received STATE "Celsius")

chris-1243, 2025-05-07:
Please check

https://github.com/Koenkk/zigbee2mqtt/issues/27292

closed 2025-05-07 by Koenkk

mlepannetier, 2025-05-08:
Thank's @chris-1243 ;)