Koenkk/zigbee2mqtt #27346: [Tuya TS0601_soil_3] Exposed values are not correctly interpreted

tdm98-ai, 2025-05-07:
### What happened?

Hi,

For the Tuya TS601_soil_3 sensor the values are mixed up between "humidity" and "battery level":

The 2 AAA batteries are new, just bought Duracell batteries.

`{
"device":{
"friendlyName":"Humidite Sol Exterieur"
},
"payload":{
"battery":100,
"humidity":34,
"linkquality":105,
"temperature":12.40
},
"timestamp":"2025-05-07T17:35:35.991Z"
}`

From the Expose view I can see:

Temperature
12.4°C
Soil moisture
100%
Battery
Remaining battery in %, can take up to 24 hours before reported
34%
Battery state
medium
Linkquality
105lqi

### What did you expect to happen?

Soil moisture
34%
Battery
Remaining battery in %, can take up to 24 hours before reported
100%

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

20230507

### Adapter

ZStack3x0

### Setup

Add-on from Home Assistant

Koenkk, 2025-05-07:
Could you provide the data/database.db entry of this device?

tdm98-ai, 2025-05-08:
> Could you provide the data/database.db entry of this device?

How can I get this information ?

tdm98-ai, 2025-05-08:
`{"id":43,"type":"EndDevice","ieeeAddr":"0xa4c13877bcba51ba","nwkAddr":55236,"manufId":4417,"manufName":"_TZE284_aao3yzhs","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"#R�/\u0012","65506":56,"65508":0,"65534":0,"modelId":"TS0601","appVersion":77,"manufacturerName":"_TZE284_aao3yzhs","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":1946894323},"lastSeen":1746713716236}`

tillman4, 2025-05-08:
> FWIW on my end I'm pretty confident the values are not swapped, the moisture measurement correlates pretty clearly with when plants are watered.

Yes I can confirm that the values are not swapped. It's easy to see by taking the sensor out of the soil and pressing the internal button, which triggers a new measurement. The problem is that these sensors are unfortunately bad in two ways: They report 100% soil moisture at much lower actual soil moisture, and the battery voltage is interpreted wrongly in the sensors firmware. Both issues occur also when connecting the sensors to a tuya hub.

heldchen, 2025-06-01:
I'm getting this error for TS0601_soil_3 - it seems instead of the `battery` (int) value the `battery_state` (string) is passed to Home Assistant:
```
2025-06-02 01:21:12.804 ERROR (MainThread) [homeassistant.helpers.event] Error while dispatching event for sensor.sensor_soil_chili_battery_state to <Job track state_changed event ['sensor.sensor_soil_chili_soil_moisture', 'sensor.sensor_soil_chili_battery_state', 'sensor.sensor_soil_chili_temperature'] HassJobType.Callback <bound method Plant._state_changed_event of <entity plant.chili=problem>>>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 355, in _async_dispatch_entity_id_event
    hass.async_run_hass_job(job, event)
    ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 938, in async_run_hass_job
    hassjob.target(*args)
    ~~~~~~~~~~~~~~^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/plant/__init__.py", line 195, in _state_changed_event
    self.state_changed(event.data["entity_id"], event.data["new_state"])
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/plant/__init__.py", line 215, in state_changed
    value = int(float(value))
                ~~~~~^^^^^^^
ValueError: could not convert string to float: 'medium'
```
current state in Z2M: 
```
{
    "battery": 48,
    "battery_state": "medium",
    "last_seen": "2025-06-02T01:28:41+02:00",
    "linkquality": 168,
    "soil_moisture": 100,
    "temperature": 22.1,
    "temperature_unit": "celsius"
}
```

incidentially, the reported battery state for these devices seems whacky at best - fresh tested batteries are reported with value `48` aka "medium" right after being inserted :-)

avilao, 2025-06-08:
I'm also having this issue with 3 of these devices.

matteobrusa, 2025-06-15:
Same here, 100% moisture is definitely way off.

1f5, 2025-06-23:
I've got the same problem with both battery level and moisture.

Koenkk, 2025-06-24:
Would this be fixed if we swap soil moisture and battery?

Koenkk, 2025-06-24:
The value is not scaled, so the value send by the device is not modified before sending it to MQTT. Can you provide the debug log when first making the sensor dry, then putting it in water and drying it again?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

matteobrusa, 2025-06-24:
I've added to my message above the parser logs - that might be of interest for you.

Koenkk, 2025-06-25:
@matteobrusa  so in your case the values look OK?

```bash
# Dry -> soil moisture = 10
[2025-06-24 22:24:16] ESC[34mdebugESC[39m: z2m: Received Zigbee message from 'tuya_test4', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,10],"type":"Buffer"},"datatype":2,"dp":3},{"data":{"data":[0,0,1,12],"type":"Buffer"},"datatype":2,"dp":5},{"data":{"data":[0,0,0,3],"type":"Buffer"},"datatype":2,"dp":15},{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":14},{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":9}],"seq":51713}' from endpoint 1 with groupID 0
[2025-06-24 22:24:16] ESC[32minfoESC[39m: z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/tuya_test4', payload '{"battery":8,"battery_state":"low","last_seen":"2025-06-24T22:24:16+02:00","linkquality":87,"soil_moisture":10,"temperature":26.8,"temperature_unit":"fahrenheit"}'

# Submerged in water (glass) -> soil moisture = 100%
[2025-06-24 22:29:34] ESC[34mdebugESC[39m: z2m: Received Zigbee message from 'tuya_test4', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":3},{"data":{"data":[0,0,1,12],"type":"Buffer"},"datatype":2,"dp":5},{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":9}],"seq":51969}' from endpoint 1 with groupID 0
[2025-06-24 22:29:34] ESC[32minfoESC[39m: z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/tuya_test4', payload '{"battery":8,"battery_state":"low","last_seen":"2025-06-24T22:29:34+02:00","linkquality":87,"soil_moisture":100,"temperature":26.8,"temperature_unit":"fahrenheit"}'

# Dry -> soil moisture = 10
[2025-06-24 22:31:43] ESC[34mdebugESC[39m: z2m: Received Zigbee message from 'tuya_test4', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,10],"type":"Buffer"},"datatype":2,"dp":3},{"data":{"data":[0,0,1,14],"type":"Buffer"},"datatype":2,"dp":5},{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":9}],"seq":52225}' from endpoint 1 with groupID 0
[2025-06-24 22:31:43] ESC[32minfoESC[39m: z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/tuya_test4', payload '{"battery":8,"battery_state":"low","last_seen":"2025-06-24T22:31:43+02:00","linkquality":80,"soil_moisture":10,"temperature":27,"temperature_unit":"fahrenheit"}'
```

@heldchen your logging also looks OK:

```bash
# Submerged -> soil moisture = 100
[2025-06-24 22:28:43] debug: 	z2m: Received Zigbee message from 'sensor-soil-basil-77347f', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":3},{"data":{"data":[0,0,0,240],"type":"Buffer"},"datatype":2,"dp":5},{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":9}],"seq":13585}' from endpoint 1 with groupID 0
[2025-06-24 22:28:43] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor-soil-basil-77347f', payload '{"battery":6,"battery_state":"low","last_seen":"2025-06-24T22:28:43+02:00","linkquality":144,"soil_moisture":100,"temperature":24,"temperature_unit":"celsius"}'

# Dry -> soil moisture = 14 and 12
[2025-06-24 22:29:46] debug: 	z2m: Received Zigbee message from 'sensor-soil-basil-77347f', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,14],"type":"Buffer"},"datatype":2,"dp":3},{"data":{"data":[0,0,0,247],"type":"Buffer"},"datatype":2,"dp":5},{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":9}],"seq":13841}' from endpoint 1 with groupID 0
[2025-06-24 22:29:46] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor-soil-basil-77347f', payload '{"battery":6,"battery_state":"low","last_seen":"2025-06-24T22:29:46+02:00","linkquality":112,"soil_moisture":14,"temperature":24.7,"temperature_unit":"celsius"}'
[2025-06-24 22:30:48] debug: 	z2m: Received Zigbee message from 'sensor-soil-basil-77347f', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,12],"type":"Buffer"},"datatype":2,"dp":3},{"data":{"data":[0,0,0,252],"type":"Buffer"},"datatype":2,"dp":5},{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":9}],"seq":14097}' from endpoint 1 with groupID 0
[2025-06-24 22:30:48] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor-soil-basil-77347f', payload '{"battery":6,"battery_state":"low","last_seen":"2025-06-24T22:30:48+02:00","linkquality":112,"soil_moisture":12,"temperature":25.2,"temperature_unit":"celsius"}'

# Submerged -> soil moisture = 100
[2025-06-24 22:32:53] debug: 	z2m: Received Zigbee message from 'sensor-soil-basil-77347f', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":3},{"data":{"data":[0,0,0,253],"type":"Buffer"},"datatype":2,"dp":5},{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":9}],"seq":14353}' from endpoint 1 with groupID 0
[2025-06-24 22:32:53] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor-soil-basil-77347f', payload '{"battery":6,"battery_state":"low","last_seen":"2025-06-24T22:32:53+02:00","linkquality":112,"soil_moisture":100,"temperature":25.3,"temperature_unit":"celsius"}'
```

Am I missing something?

matteobrusa, 2025-06-25:
The problem is that as soon as you stick the sensor in the soil, the reported moisture goes up to 100%. 
I've bought a [COOLO CS-201Z](https://www.zigbee2mqtt.io/devices/CS-201Z.html#coolo-cs-201z) and placed it next to the Tuya. The first one reports 52%, the second 100%.
I'm good to donate 10€ for you to buy one of them from ali, maybe it's easier for you to troubleshoot the issue first-hand. Let me know if you like the idea.

matteobrusa, 2025-06-28:
@Koenkk I wonder myself, and find it unlikely.
I have some [Tuya TS0601_soil](https://www.zigbee2mqtt.io/devices/TS0601_soil.html#tuya-ts0601-soil) reliably caring for my outdoor plants since long time, this seems to be an iteration of it.

I don't have a Tuya hub to test it, if anyone reading this thread can support, we'll have a chance to fix this issue.

gycone, 2025-07-19:
I have a few of these sensors ( _TZE284_aao3yzhs ) and I'm experiencing the same issues discussed here. I'm curious if there's a solution to get the readings as close to the truth as possible or if they're compromised from the factory.

Nisbo, 2025-07-29:
I would wait till someone provides information how to update the firmware.