Koenkk/zigbee2mqtt #27352: [New device support]: Tuya ZigBee 24G MmWave Radar Smart Human Presence Detector  [_TZE204_no6qtgtl]

milobt, 2025-05-08:
### Link

https://www.aliexpress.com/item/1005008493603271.html?spm=a2g0o.order_list.order_list_main.5.6ab51802iimRSt

### Database entry

{"id":113,"type":"Router","ieeeAddr":"0xa4c13809ffc2789d","nwkAddr":63569,"manufId":4417,"manufName":"_TZE204_no6qtgtl","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0005\u0000\u0000\u0000e\u0006\u0000\u0000\u0000\u0012:?O/\u0013:?O/e;?O/\u0012\u0005\u0000\u0000\u0000e\u0006\u0000\u0000\u0000\u0012","65506":56,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_no6qtgtl","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1746696610798}

### Zigbee2MQTT version

2.3.0

### Comments

Is a variant of TS0601 presence sensor range.
Pairs with a Tuya Hub and a provides the following:
Sensors: Occupancy
Controls: Clostest target distance
Configuration: Far detection, Near detection

Have also opened up one unit to take a look inside.
The Aliexpress listing is misleading and does not state power source, which is in fact USB-C.
It also shows an internal image - take from the Xiaomi Mijia XMOSB01XS bluetooth presence sensor - which shows PIR sensor as well as MMWave, but this is not the case with this Tuya Presence Sensor.
I also do not see a photodiode for illuminance sensing.

[image]
[image]
[image]

### External definition

```shell
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_no6qtgtl',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

Only link quality is populalated when pairing with Z2M.

kvazis, 2025-05-14:
Hi, working converter - 
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE204_no6qtgtl'}],
    model: 'TS0601_no6qtgtl',
    vendor: 'Tuya',
    description: 'Tuya 24GHz human presence sensor (custom external converter)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.enum('presence_state', ea.STATE, ['none', 'motion', 'stationary'])
            .withDescription('Presence state: none, motion, or stationary'),
        e.numeric('target_distance_closest', ea.STATE)
            .withUnit('m')
            .withValueMin(0)
            .withValueMax(10)
            .withDescription('Closest detected target distance'),
        e.numeric('near_detection', ea.STATE_SET)
            .withUnit('m')
            .withValueMin(0.1)
            .withValueMax(6)
            .withValueStep(0.01)
            .withDescription('Minimum detection range'),
        e.numeric('far_detection', ea.STATE_SET)
            .withUnit('m')
            .withValueMin(0.1)
            .withValueMax(6)
            .withValueStep(0.01)
            .withDescription('Maximum detection range'),
        e.numeric('static_sensitivity', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(10)
            .withDescription('Static presence sensitivity'),
        e.numeric('motion_sensitivity', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(10)
            .withDescription('Motion sensitivity'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'presence_state', tuya.valueConverterBasic.lookup({
                none: 0,
                motion: 1,
                stationary: 2,
            })],
            [3, 'near_detection', tuya.valueConverter.divideBy100],
            [4, 'far_detection', tuya.valueConverter.divideBy100],
            [9, 'target_distance_closest', tuya.valueConverter.divideBy100],
            [101, 'static_sensitivity', tuya.valueConverter.raw],
            [102, 'motion_sensitivity', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```

Koenkk, 2025-05-14:
@kvazis Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

milobt, 2025-05-14:
@kvazis have you tried this and have it working with the same device?
I’ve just put it in a file and added as an external converter; it successfully interviewed but didn’t configure properly. 
Still says not supported and only get link quality.

kvazis, 2025-05-16:
> [@kvazis](https://github.com/kvazis) Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

Done - https://github.com/Koenkk/zigbee-herdsman-converters/pull/9355

Koenkk, 2025-05-16:
Merged, thanks @kvazis !

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-05-16 by Koenkk