Koenkk/zigbee2mqtt #27357: Invalid value for number (sensitivity)

markg85, 2025-05-08:
### What happened?

The error that shows in home assistant logging:
```
2025-05-08 15:11:28.183 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.breath_sensor_presence_sensitivity: 0 (range 1.0 - 10.0)
2025-05-08 15:11:28.183 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.breath_sensor_move_sensitivity: 0 (range 1.0 - 10.0)
```

It's this sensor / device:
```
z2m: breath_sensor (0xe498bb1945e563e4): ZY-M100-24GV3 - Tuya 24G MmWave radar human presence motion sensor（added distance switch） (Router)
```

This is the faulty payload:
```
May 08 17:32:37 supermars node[13000]: [2025-05-08 17:32:37] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/breath_sensor', payload '{"detection_distance_max":5.25,"detection_distance_min":1.5,"distance":1.2,"find_switch":"ON","illuminance":49,"linkquality":123,"move_sensitivity":0,"presence":true,"presence_sensitivity":0,"presence_timeout":1,"state":"move"}'
```

Given the HA error above, the `move_sensitivity` probably has to be 1.0.
Same for `presence_sensitivity`.

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

Irrelevant

### Adapter

Irrelevant

### Setup

Irrelevant

KingAdeCode, 2025-05-29:
Having the same issue

[image]

KingAdeCode, 2025-05-29:
[image]

KingAdeCode, 2025-05-29:
The device: https://www.zigbee2mqtt.io/devices/ZY-M100-24GV3.html#tuya-zy-m100-24gv3

KingAdeCode, 2025-05-29:
I temporarily fixed this by setting all configurations above "0.". Check my configuration here:

[image]

sanketgroup, 2025-07-16:
But even if we set value to above zero on these sliders, the values go back to 0 after few seconds/minutes.

markg85, 2025-09-27:
This issue still persists.

It's not a thing that breaks which is ironically unfortunate else you would have noticed and fixed it.
Just start home assistant and monitor the logs and you'll see this.

closed 2025-12-05 by github-actions[bot]