Koenkk/zigbee2mqtt #27360: [Wrong device]: Yookee D10110_1 suddenly recognized as Tuya TS130F

rohankapoorcom, 2025-05-09:
### Link

https://www.yoolax.com/collections/motorized-shades

### Model

D10110_1

### Description

Smart blind

### Vendor

Yookee

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/D10110_1.png

### Database entry

N/A

### Notes

Support for this Yookee Blind Motor was added #21924.

I just updated to Z2M 2.3.0-dev commit: [1d6ba5393a1dc5626b6ca539e3ec245cc3d61a56](https://github.com/Koenkk/zigbee2mqtt/commit/1d6ba5393a1dc5626b6ca539e3ec245cc3d61a56) and discovered that the motor is no longer identifying itself correctly.

I have a separate installation of Z2M still on version 2.1 and it is identifying the motor correctly.

Koenkk, 2025-05-10:
Do you have any external converter for this device? It should be detected as D10110_1 indeed: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts#L2792

rohankapoorcom, 2025-05-10:
Nope, no external converters installed. All I did was update the docker image and then on restart the identification was wrong.

chris-1243, 2025-05-11:
@Koenkk Sorry to ping you directly, should we not use a `fingerrpint` for the tuya device detection ?

```
fingerprint: tuya.fingerprint("TS0301", ["_TZE200_9caxna4s"]),
```
and the converter might look like

```typescript
{
    fingerprint: tuya.fingerprint("TS0301", ["_TZE200_9caxna4s"]),
    model: "TS0301",
    vendor: "Tuya",
    description: "Cover",
    extend: [m.battery(), m.windowCovering({controls: ["lift"]})],
    whiteLabel: [tuya.whitelabel("Yookee", "D10110_1", "Smart blind", ["_TZE200_9caxna4s"])],
},

I may create a PR if this seems fine for you

Koenkk, 2025-05-11:
@rohankapoorcom I copied your db entry into mine and it gets detected correctly on the latest dev:

[image]

[image]

@chris-1243 `fingerprint` is not needed here, this is the only `TS0301` being supported so `zigbeeModel` is sufficient.

rohankapoorcom, 2025-05-13:
@Koenkk I updated to the latest dev again and then it changed to yet another Tuya device 😂.

I then realized what the problem is. I am running a custom version of the Inovelli converter still (like we had discussed in Discord a few weeks ago) and after the changes were done to dynamically load converters, I had bind-mounted the `inovelli.js` file as well as the `models-index.json` file into the docker container. I guess that file has to be updated every time I pull an updated version of zigbee-herdsman-converters. Once I rebased my branch on the main and rebuilt the files, everything was fine. It is weird to me that all of my other devices (some Xiaomi, some Tuya, many Inovelli) were all fine though.

Looks like there's no bug, just operator error.

Thanks for helping me debug and sorry for the confusion!

closed 2025-05-13 by Koenkk