Koenkk/zigbee2mqtt #27364: [New device support]: New variation of TuYa TS110E_2gang_2: _TZ3210_mt5xjoy6, branded as Nedis

Andunes, 2025-05-09:
### Link

https://nedis.se/sv-se/product/smarta-hem/power-en-energy/strombrytare/550786231/smartlife-strombrytare-zigbee-30-2x100-w-terminalanslutning-app-tillganglig-for-android-ios

### Database entry

{"id":6,"type":"Router","ieeeAddr":"0xa4c138e5410dff6a","nwkAddr":39013,"manufId":4417,"manufName":"_TZ3210_mt5xjoy6","powerSource":"Mains (single phase)","modelId":"TS110E","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[5,4,6,8,57345,4096,3,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS110E","manufacturerName":"_TZ3210_mt5xjoy6","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"61440":0,"onOff":1,"moesStartUpOnOff":1}},"genLevelCtrl":{"attributes":{"61440":392,"64515":10,"64516":1000}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":257,"inClusterList":[5,4,6,8,57345,5,4],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"61440":0,"onOff":1}},"genLevelCtrl":{"attributes":{"61440":392,"64515":10,"64516":1000}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1746801129269}

### Zigbee2MQTT version

2.3.0

### Comments

Haven't figured out how to try anything yet on a Home Assistant Green but i assume the solution would be similar to this issue: https://github.com/Koenkk/zigbee2mqtt/issues/21292 and adding _TZ3210_mt5xjoy6

### External definition

```shell
-
```

### What does/doesn't work with the external definition?

-

chris-1243, 2025-05-09:
```typescript
import * as fz from "zigbee-herdsman-converters/converters/fromZigbee";
import * as tz from "zigbee-herdsman-converters/converters/toZigbee";
import * as exposes from "zigbee-herdsman-converters/lib/exposes";
import * as reporting from "zigbee-herdsman-converters/lib/reporting";
import * as tuya from "zigbee-herdsman-converters/lib/tuya";

const e = exposes.presets;
const ea = exposes.access;

export default {
    fingerprint: tuya.fingerprint("TS110E", ["_TZ3210_pagajpog", "_TZ3210_4ubylghk", "_TZ3210_vfwhhldz", "_TZ3210_3mpwqzuu"]),
    model: "TS110E_2gang_2",
    vendor: "Tuya",
    description: "2 channel dimmer",
    whiteLabel: [tuya.whitelabel("Nedis", "ZBWD20RD", "2 Gang dimmer 2x100W", ["_TZ3210_mt5xjoy6"])],
    fromZigbee: [fz.TS110E, fz.TS110E_light_type, tuya.fz.power_on_behavior_1, fz.on_off],
    toZigbee: [tz.TS110E_onoff_brightness, tz.TS110E_options, tuya.tz.power_on_behavior_1, tz.light_brightness_move],
    meta: {multiEndpoint: true},
    exposes: [
        e.light_brightness().withMinBrightness().withMaxBrightness().withEndpoint("l1"),
        e.light_brightness().withMinBrightness().withMaxBrightness().withEndpoint("l2"),
        e.power_on_behavior().withEndpoint("l1"),
        e.power_on_behavior().withEndpoint("l2"),
        tuya.exposes.lightType().withAccess(ea.ALL),
        tuya.exposes.switchType().withEndpoint("l1"),
        tuya.exposes.switchType().withEndpoint("l2"),
    ],
    configure: async (device, coordinatorEndpoint) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ["genOnOff", "genLevelCtrl"]);
        await reporting.onOff(endpoint);
    },
    endpoint: (device) => {
        return {l1: 1, l2: 2};
    },
};
```

Could you try this external converter and check if it works with your device. Add it as an `*.mjs` file. Check the link below how to add it.

https://www.zigbee2mqtt.io/advanced/more/external_converters.html

Andunes, 2025-05-09:
Thank you Chris, that pointed me in the right direction. I made two stupid mistakes while experimenting earlier that stopped the external converter from loading:

1. Place the file in _data/external_converters_ instead of _zigbee2mqtt/external_converters_
2. Ended the filename in _.js_ instead of _.mjs_

I took your code, added `"_TZ3210_mt5xjoy6"` after `fingerprint:` and now everything looks correct in the UI. I just have to double check that everything actually works tomorrow.

```
import * as fz from "zigbee-herdsman-converters/converters/fromZigbee";
import * as tz from "zigbee-herdsman-converters/converters/toZigbee";
import * as exposes from "zigbee-herdsman-converters/lib/exposes";
import * as reporting from "zigbee-herdsman-converters/lib/reporting";
import * as tuya from "zigbee-herdsman-converters/lib/tuya";

const e = exposes.presets;
const ea = exposes.access;

export default {
    fingerprint: tuya.fingerprint("TS110E", ["_TZ3210_pagajpog", "_TZ3210_4ubylghk", "_TZ3210_vfwhhldz", "_TZ3210_3mpwqzuu", "_TZ3210_mt5xjoy6]),
    model: "TS110E_2gang_2",
    vendor: "Tuya",
    description: "2 channel dimmer",
    whiteLabel: [tuya.whitelabel("Nedis", "ZBWD20RD", "2 Gang dimmer 2x100W", ["_TZ3210_mt5xjoy6"])],
    fromZigbee: [fz.TS110E, fz.TS110E_light_type, tuya.fz.power_on_behavior_1, fz.on_off],
    toZigbee: [tz.TS110E_onoff_brightness, tz.TS110E_options, tuya.tz.power_on_behavior_1, tz.light_brightness_move],
    meta: {multiEndpoint: true},
    exposes: [
        e.light_brightness().withMinBrightness().withMaxBrightness().withEndpoint("l1"),
        e.light_brightness().withMinBrightness().withMaxBrightness().withEndpoint("l2"),
        e.power_on_behavior().withEndpoint("l1"),
        e.power_on_behavior().withEndpoint("l2"),
        tuya.exposes.lightType().withAccess(ea.ALL),
        tuya.exposes.switchType().withEndpoint("l1"),
        tuya.exposes.switchType().withEndpoint("l2"),
    ],
    configure: async (device, coordinatorEndpoint) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ["genOnOff", "genLevelCtrl"]);
        await reporting.onOff(endpoint);
    },
    endpoint: (device) => {
        return {l1: 1, l2: 2};
    },
};
```

Andunes, 2025-05-14:
After some testing it seems to work without issues.

closed 2025-05-14 by Koenkk

pullasuti, 2025-05-27:
These switches have become available locally, so purchased one hoping to get it going with Z2M. Have not managed to get it to work 100% with the above converter, however. 

There seem to be issues with getting/setting the light type, power-on behavior and switch type for l1 and l2. power_on_behavior_l1 is the only value that can be set/get, rest report "UNSUPPORTED_ATTRIBUTE". 
    
    "light_type": null,
    "power_on_behavior_l1": "on",
    "power_on_behavior_l2": null,
    "switch_type_l1": null,
    "switch_type_l2": null,

`error 2025-05-27 11:09:05z2m: Publish 'set' 'switch_type' to '0xa4c1388a986dabe7' failed: 'Error: ZCL command 0xa4c1388a986dabe7/1 genLevelCtrl.write({"64514":{"value":1,"type":32}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
error 2025-05-27 11:09:32z2m: Publish 'set' 'switch_type' to '0xa4c1388a986dabe7' failed: 'Error: ZCL command 0xa4c1388a986dabe7/1 genLevelCtrl.write({"64514":{"value":2,"type":32}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
error 2025-05-27 11:09:36z2m: Publish 'set' 'power_on_behavior' to '0xa4c1388a986dabe7' failed: 'Error: ZCL command 0xa4c1388a986dabe7/2 genOnOff.write({"moesStartUpOnOff":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

Cimmerholm, 2025-07-23:
@pullasuti 
I have the same issue. 

> There seem to be issues with getting/setting the light type, power-on behavior and switch type for l1 and l2. power_on_behavior_l1 is the only value that can be set/get, rest report "UNSUPPORTED_ATTRIBUTE".

rwestnl, 2025-11-14:
I also have the Nedis ZBWD20RD dual triac dimmer.
when trying to set the switch_type_l1 or switch_type_l2 i got a UNSUPPORTED_ATTRIBUTE popup message.
is there a way to read/test or test to set this by the dev-console?

gitvisual, 2025-11-14:
> I also have the Nedis ZBWD20RD dual triac dimmer. when trying to set the switch_type_l1 or switch_type_l2 i got a UNSUPPORTED_ATTRIBUTE popup message. is there a way to read/test or test to set this by the dev-console?

I e-mailed NEDIS directly and asked about this a couple of weeks ago: their response was this:

**"No, it will not work with a momentary switch.
We do not have another model that will work that way"**

So I guess the only problem for this is that Z2M are showing those options, no way (according to Nedis) to actually get them to do anything since the product can't work like that