Koenkk/zigbee2mqtt #27370: Frient/Develco SMSZB-120 Test Alart no feedback in z2mqtt

Jogiarea, 2025-05-10:
### What happened?

Hello, I have two Frient/Develco SMSZB-120 smoke detectors in use, I have not yet been able to test smoke feedback, but this does not work with z2M, when I press the button / test button on the smoke detector (i.e. at the bottom of the smoke detector) there is no feedback in z2M and I cannot create an automation with it, does anyone know the problem?

### What did you expect to happen?

sdisconnect
reset
new connect

### Zigbee2MQTT version

 2.3.0-1

### Adapter firmware version

20240520 09:54

### Adapter

 EZSP v12 7.3.1.0 build 176 (sonoff)

### Setup

x64 Homeassistant OS

### Debug log

no debugs

eyesaider, 2025-05-10:
The documentation in this repository is excellent! Very clear and comprehensive.

Jogiarea, 2025-05-10:
> Die Dokumentation in diesem Repository ist ausgezeichnet! Sehr übersichtlich und umfassend.

Thanks for the feedback but I don't understand, it was meant to be ironic, yes, the z2mqtt instructions are great but I couldn't find anything on this. Doesn't help me at all

closed 2025-07-17 by github-actions[bot]