Koenkk/zigbee2mqtt #27374: Fast battery drain with Aqara Smoke Alarm JY-GZ-01AQ

peteh, 2025-05-10:
### What happened?

I have two smoke detectors from Aqara (JY-GZ-01AQ). Both lost more than half of the battery within 1-2 months. 

[image]

Note: I had the first firealarm connected to my normal setup using a Dongle-P, the battery drain issue was present there as well. Thus, it might be unrelated to the actual adapter. 

### What did you expect to happen?

They are supposed to last up to 10 years, thus they should not drain that fast. 

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

20250321

### Adapter

slzb06

### Setup

z2m in docker
homeassistant in docker
connected via mqtt

### Debug log

[log.log](https://github.com/user-attachments/files/20143995/log.log)

nickread4nick, 2025-05-11:
For what its worth, I have seen the same behaviour - but only on the 2nd and 3rd fire alarms. My first, which I got a few months earlier to test, is still at 100% battery after a year. The 2nd and 3rd units on the other hand are down to ~32% after 6 months. The first is a bit closer to the controller and has an LQI of ~109. The 2nd and 3rd have lqi of ~80 and ~40 respectively, so less but not terrible. All configured the same and with the same firmware version (20230310).

peteh, 2025-05-15:
That would mean that there is a hardware difference or quality issue with the batteries possibly. 

I'll try to reach out to aqara, not sure how good their customer service is though.

rhwilr, 2025-05-15:
I'd be interested to hear if you get a response. I see the same issue with my 3 units. All were bought and installed at the start of the year and all 3 are already below 40%:

[image]

Interestingly, the blue one is right next to the coordinator with a lqi of ~130, while the other two are around 50.

peteh, 2025-05-15:
I sent them a request via this weird page, lets see: https://static-resource.aqara.com/html/feedback-shopify.html

peteh, 2025-05-16:
Response from them

* z2m is not officially supported
* connect them to a Aqara hub to check for firmware updates - when asked what different firmwares are available and what changed there was no answer
* Buy a new battery

Unfortunately I'll go to vacation soon - so I cannot currently order an aqara hub. 

My current plan: 
* get an aqara hub and connect one of the two to there and see if it updates - even though my guess is that this is just a bs response and there is only one firmware version
* Leave the other one in my current setup
* add new batteries to both
* see how the battery drain is affected

@nickread4nick can you somehow check if your two new devices have a different fw than the older one? It looks like there is some fw reporting in the state: 

[image]

nickread4nick, 2025-05-17:
Strangely that firmware section doesn't show up in any of my three. This is
the info on the about page (again the same for all three)

peteh, 2025-05-17:
> Strangely that firmware section doesn't show up in any of my three. This is
> the info on the about page (again the same for all three)
> […](#)

Check the state in the state section of the device (see the screenshot from my previous reply).

nickread4nick, 2025-05-17:
Yes I realise 🙂

peteh, 2025-05-17:
@nickread4nick 
> Yes I realise 🙂

So what does it say for you in the state page?

nickread4nick, 2025-05-17:
I posted a screenshot in my previous post. It has all the other entries but
nothing about firmware.

peteh, 2025-05-19:
Aqara confirmed that 17 is the latest version. Thus, upgrading will not help. 

Nevertheless I'll continue to grab a hub from them and two new batteries to see the drain behavior.

peteh, 2025-06-01:
> Is the battery really draining? For me it drained very fast at the beginning (3 months to go from 100% down to 60%) and is now draining very slowly. It seems that the device only returns the battery voltage, then zigbee2mqtt computes the battery percentage as a linear function between 2850 and 3000 mV. I think a more advanced battery level computation needs to be implemented.

Good point. I'll double check this as soon as I get my aqara hub and check the reported soc there. It hopefully arrives tomorrow.

peteh, 2025-06-02:
I managed to link one of the smoke alarms to the Aqara E1 Hub. 

[image]

So 43% battery in Zigbee2mqtt is still considered "normal" in Aqara Home. 

Unfortunately Aqara Home does not provide a state of charge estimation. 

So it seems to be likely that z2m's soc estimate is very off as @aurel32 said.

rhwilr, 2025-06-03:
I also did a little bit more research:
[image]

z2m reports the voltage for the battery. My 3 detectors are all almost exactly at 2900mV.
The Aqara Smoke Detector uses a CR17450 Primary Lithium Battery, which is a Lithium manganese dioxide (LiMnO2) cell.
These cells have a very flat discarge curve from 3.0V to 2.8V : 

[image]

As with many lithium battery chemistries, you can't accurately calculate the battery soc from the voltage.
So unless our Smoke Detectors all start dying in 2 months, I'd agree that the percentage in z2m is probably wrong.

nickread4nick, 2025-06-09:
I agree the SOC number reported by Z2M is likely 'wrong' inasfar as the discharge is not a linear one by voltage. However given the nature of the curve, that would tend to keep the SOC% high longer than was real, not cause it to drop early. And even if the value is inexact, the downward trend is real. My graph for the voltage of my three units:

[image]

The blue one which is still at 100% is clearly higher and levelish compared to the dropping lines (yellow/orange) of the other two, which now report ~25%. 

My personal guess is that these batteries are expensive, and as I bought mine from Aliexpress, they probably cheaped out on the battery when I bought the second two (with lower readings). I've got replacement batteries standing by for when they go offline.

peteh, 2025-06-14:
'''
Hi Pete,

Thank you for your reply.

Currently, the data we can provide is: when connected to the Aqara platform, the device will report low battery if the voltage drops below 2.7V.
 
Sincerely, 
Isabel
Aqara Technical Support
'''

So we could add another binary sensor that says battery low instead for this threshold

aurel32, 2025-06-14:
> So we could add another binary sensor that says battery low instead for this threshold

Thanks, that confirms that the current battery percentage calculation is off. We might also change the range used to compute the battery percentage from 3000 -> 2850 mV to 3000 -> 2700 mV.

aurel32, 2025-06-15:
> The risk of that approach is that because the end of the curve drops off
> rapidly at the end, with that configuration you might go rapidly from 30%
> to zero and have the detector drop off the network (safety risk). Not a
> problem if you monitor it, but can catch you unawares if you're only
> checking every few months.

Even if the battery level drops rapidly at the end down to 0%, this 0% would still match the low level indicator in the original Aquara framework. And I guess they have taken some margin. Said otherwise zigbee2mqtt would still be more pessimistic than the original Aquara framework.

nickread4nick, 2025-06-15:
Yes slightly more pessimistic - but you have to balance inaccuracy from
doing a straight line calculation from voltage to SOC. In other words,
probably better to be slightly pessimistic on 'zero' and more accurate over
the lifetime as a whole?

weichenb, 2025-06-16:
What do you think, to test it out the voltage threshold with a laboratory power supply?
If I am not wrong, the device even begins to beep before it shuts off. Alternatively we could test what voltage level does still support proper communication.
I think my power supply supports voltage settings at least in the hundredth range.

peteh, 2025-06-16:
I was thinking about that too but I'm not sure how accurate my power supply would be and I don't have that much free time at the moment. 

We could also reach out to aqara customer support. 

After sending a link to this issue they told me the 2700mAh threshold.

weichenb, 2025-06-17:
This evening I set up a test case on my couch. While watching TV I tested some different voltages with Aqara smoke sensor.
[image]

Test case:
1. set a specific voltage on laboratory power supply.
2. checked rugularily voltage on a separate meter
3. pushed latest every 2 minutes the button on the sensor to force an update to zigbee controller
4. continued at least 35 minutes to document final result

My findings so far:

- the laboratory power supply assured a stable voltage +-0,0002V over complete eval period.
- between  laboratory power supply and meter there was only a difference of 0,1%, so quite precise. 
- the Aqara sensor requires about 35 minutes until it reports a stable voltage. Before it slowly converges final level.
- between 35 and 60 minutes voltage did not change anymore. Therefore I assume that final voltage level got represented.
- **During my test cases so far the Aqara device interestingly provided a ~5,2% too low voltage (~0,15V). This means that SoC will be ever too low!**  

[image]

I am going to complete series during next 5 days.

peteh, 2025-06-18:
@weichenb 

https://github.com/Koenkk/zigbee-herdsman-converters/blob/26bcefabba512a69495d86fb5f6493db73f5ff1b/src/devices/lumi.ts#L2223

You can load the repo in vscode and jump to the methods. It's just linear calculation. 

From you measurements it seems we should should 
* let a binary battery indicator say battery low at 2700 
* a 0% soc at around 2500 (maybe with a parameterized curve fit from your measurements)

weichenb, 2025-06-18:
@peteh :
I studied the code, especially also in utils.ts. From my point of view we should not overstress a device specific adaption. Maybe the min-max approach works as well. 
I adapted the min value to 2500 instead of 2850 on my local installation.
 `meta: {battery: {voltageToPercentage: {min: 2500, max: 3000}}}, `

I will check how it behaves with different voltages.

peteh, 2025-06-18:
> [@peteh](https://github.com/peteh) : Thanks for your guidance.
> 
> I studied the code, especially also in utils.ts. From my point of view we should not overstress a device specific adaption. Maybe the min-max approach works as well. I adapted the min value to 2500 instead of 2850 on my local installation. `meta: {battery: {voltageToPercentage: {min: 2500, max: 3000}}}, `
> 
> I will check how it behaves with different voltages.

I think the Low battery entity should also still be added. Other devices also have battery low but it usually maps directly to a bool flag in the state message, not to a specific voltage. At least I haven't found one which uses voltage mapping.

weichenb, 2025-06-18:
I created y a pull request on [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters), where the related code is maintained.

weichenb, 2025-06-23:
Just to let you know, modification for Aqara Smoke Alarm JY-GZ-01AQ got live since  zigbee-herdsman-converters v23.63.0.
Thanks to Koen for the quick implementation!

weichenb, 2025-07-04:
> 24% to 78% post Z2M upgrade - lets hope its true ;-)

With shown SOC your battery should have a cuurent voltage of about 2,880V, right?
If so, device's own  battery warnkng is still 0,380V upfront.

Please consider that voltage based SOC approximation will be never 100% reliable, since for that battery type voltage has a non-linear behavior (see above).

I would appreciate, that xou keep me informed about remaining battery life. Sensor is expected to start acustical warning at 2 500V.

If my test seties was correct, it will have safed a lot of unnecessary battery changes ;-)

nickread4nick, 2025-07-04:
> I think consumption is still very fast (compared to 10 years) but yes
> otherwise I would be replacing this year already.
>
> Apropos, how long runs this battery already?

About 7 months now. And I have another one the same time/same soc, but a
third running around 9 months and still at 100%/3v+

peteh, 2025-11-11:
> Interestingly, the battery voltage of the two units that were dropping too
> quickly have started to go back up...
> […](#)

Which makes sense as the curves have been adapted in z2m for voltage. You should see a spike up after one of the last z2m updates and then slowly go down again.

nickread4nick, 2025-11-11:
Sorry I forgot that responding via email doesn't include any attached images. You are right that of course the battery **percentage** jumped up after the 'fix' but it didn't keep dropping after that, in fact it levelled out and then started to rise again:

[image]

[image]

What I was talking about was that the actual **voltage** (which didn't get changed AFAIK) levelled off and started to rise:

[image]

GrouzyW, 2025-12-30:
I was searching about a similar issue with my recently installed Aqara Smoke Detector and Z2M.
I have 3 of them since beg December, installed at the same time.
One is at 96%, one at 91%, but the third one dropped almost 25% within the last 3 days !

[image]

closed 2026-03-09 by github-actions[bot]