Koenkk/zigbee2mqtt #27375: Z2M wants to write into `/usr` even `ZIGBEE2MQTT_DATA` is specified to `/var/lib/zigbee2mqtt` when installed system-wide

adam900710, 2025-05-10:
### What happened?

[Step to reproduce]
 
1. Install `zigbee2mqtt` from a distro's repo

   In my case it's Archlinux's `extra/zigbee2mqtt` package

3. Create external converters in `/var/lib/zigbee2mqtt/external_converters`

   The directory is specified by `ZIGBEE2MQTT_DATA` environment variable for the z2m service, and all runtime data should only exist inside that directory (with all proper ownership handled by the package manager)

3. Start z2m service and check the log

   The log shows:

```
[2025-05-09 10:11:32] error:    z2m: Failed to start 'ExternalConverters' (Error: ENOENT: no such file or directory, mkdir '/usr/share/webapps/zigbee2mqtt/dist/external_converters'
    at Object.mkdirSync (node:fs:1361:26)
    at ExternalConverters.getFilePath (/usr/share/webapps/zigbee2mqtt/lib/extension/externalJS.ts:68:16)
    at ExternalConverters.loadFiles (/usr/share/webapps/zigbee2mqtt/lib/extension/externalJS.ts:204:38)
    at ExternalConverters.start (/usr/share/webapps/zigbee2mqtt/lib/extension/externalJS.ts:54:20)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Controller.startExtension (/usr/share/webapps/zigbee2mqtt/lib/controller.ts:271:13)
    at Controller.start (/usr/share/webapps/zigbee2mqtt/lib/controller.ts:151:13)
    at start (/usr/share/webapps/zigbee2mqtt/index.js:149:5))
```

   And no external converters are loaded.

Remember any reasonable system-wide installed package should not write into `/usr`, and systemd even has options `ProtectSystem` to mount `/usr` read-only for a service.

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

Irrelevant

### Adapter

Irrelevant

### Setup

Bare-metal.

### Debug log

```
[2025-05-09 10:11:32] error:    z2m: Failed to start 'ExternalConverters' (Error: ENOENT: no such file or directory, mkdir '/usr/share/webapps/zigbee2mqtt/dist/external_converters'
    at Object.mkdirSync (node:fs:1361:26)
    at ExternalConverters.getFilePath (/usr/share/webapps/zigbee2mqtt/lib/extension/externalJS.ts:68:16)
    at ExternalConverters.loadFiles (/usr/share/webapps/zigbee2mqtt/lib/extension/externalJS.ts:204:38)
    at ExternalConverters.start (/usr/share/webapps/zigbee2mqtt/lib/extension/externalJS.ts:54:20)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Controller.startExtension (/usr/share/webapps/zigbee2mqtt/lib/controller.ts:271:13)
    at Controller.start (/usr/share/webapps/zigbee2mqtt/lib/controller.ts:151:13)
    at start (/usr/share/webapps/zigbee2mqtt/index.js:149:5))
```

Kranzes, 2025-05-11:
`Failed to start 'ExternalConverters' (Error: ENOENT: no such file or directory, mkdir '/nix/store/y4fx69wiqmpry0q8ii7hjvsp6yd05bhz-zigbee2mqtt-2.3.0/lib/node_modules/zigbee2mqtt/dist/external_converters'` on NixOS too.

Koenkk, 2025-05-19:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-05-19 by Koenkk