Koenkk/zigbee2mqtt #27381: Sonoff dongle-p: cannot start docker image (Error: SRSP - SYS - ping after 6000ms)

lcarlier, 2025-05-11:
### What happened?

Hello,

I'm trying to start zigbee2mqtt docker but I'm getting the following error `Error: SRSP - SYS - ping after 6000ms`

### What did you expect to happen?

zigbee2mqtt starting

### How to reproduce it (minimal and precise)

Start docker image and see the failure

### Zigbee2MQTT version

 2.3.0 (commit #aec59cb774aa09a793ae41cd60e6a5cef15b293c)

### Adapter firmware version

CC1352P2_CC2652P_launchpad_coordinator_20240710.hex

### Adapter

Sonoff dongle-p

### Setup

lcarlier@raspberrypi3:~/zigbee2mqtt $ uname -a && cat /etc/issue.net
Linux raspberrypi3 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
Debian GNU/Linux 12

### Debug log

[log.log](https://github.com/user-attachments/files/20150997/log.log)

chris-1243, 2025-05-11:
From your log, you have `"rtscts":true`.

It should be set to `false` unless you have manually change this on the Dongle-P.

Check your `configuration.yaml`

lcarlier, 2025-05-11:
Thanks for the quick feedback.

I tried with `"rtscts":false` but still the same.

Anythin else I can check?

[log.log](https://github.com/user-attachments/files/20151340/log.log)

chris-1243, 2025-05-11:
You will need to share your `docker-compose.yaml` or docker command and your `configuration.yaml`.

lcarlier, 2025-05-11:
I'm sharing here my `configuration.yaml` and `docker-compose.yml` (I had to change the extension to txt else github didn't allow me to attach)

For reference I also have the following symlink
```bash
lcarlier@raspberrypi3:~/zigbee2mqtt $ ls -l /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_de9fdcf0ac6bef119969a1adc169b110-if00-port0 
lrwxrwxrwx 1 root root 13 May 11 21:20 /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_de9fdcf0ac6bef119969a1adc169b110-if00-port0 -> ../../ttyUSB0
```

[docker-compose.txt](https://github.com/user-attachments/files/20151430/docker-compose.txt)
[configuration.txt](https://github.com/user-attachments/files/20151431/configuration.txt)

chris-1243, 2025-05-11:
Nothing wrong in your configuration. Do you use ZHA or any software which might access your dongle?

lcarlier, 2025-05-12:
I don't believe so.

The following command doesn't returns anything
```bash
sudo lsof | grep ttyUSB0
```

Also I restarted my RPI and I also stopped HA to make sure ZHA isn't running (even though I haven't installed it). Even with that still the same error

I stopped HA because I had indeed suspicion that it might access the dongle because I see that HA reports my dongle as new device

[image]
Is that expected with the zigbee2mqtt firmware?

chris-1243, 2025-05-12:
Do you run HA in docker as well ? If yes, have you shared your dongle in HA `docker-compose.yaml` ?

It might be firmware related as well but 20240710 should work well with your Dongle-P. I nerver had issues with this firmware revision

lcarlier, 2025-05-12:
I don't think I have modemamanger installed or running. Should I run it?

Regarding HA, I'm running it in docker and I was actually surprised that it detected the dongle because I'm not sharing /dev/ttyUSB0 with it. Still a mistery to me.

Here is what I'm sharing with HA docker
```
    volumes:
      - /home/lcarlier/homeassistant/config:/config
      - /etc/localtime:/etc/localtime:ro
      - /run/dbus:/run/dbus:ro
```

Here is the output of what you were asking
```
lcarlier@raspberrypi3:~/homeassistant $ sudo ls -l /proc/[0-9]*/fd/ |grep /dev/ttyUSB0
lcarlier@raspberrypi3:~/homeassistant $ 
```

chris-1243, 2025-05-12:
> I don't think I have modemamanger installed or running. Should I run it?

ModemManager is installed by default on modern Linux distro (at least Unbuntu and Debian). This software is known to cause issues with the serial port and Z2M. So, I would strongly advise to turn it off even to uninstall it if you do not plan to use it.

https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html#modemmanager-is-installed

> Regarding HA, I'm running it in docker and I was actually surprised that it detected the dongle because I'm not sharing /dev/ttyUSB0 with it. Still a mistery to me.

I am as well as I do run everything vai Docker (HA, Z2M, Mosquitto,..) and HA has never seen my dongle.

lcarlier, 2025-05-12:
I purged modemmanager to make sure but still no luck.

Also what I did is to run the `ls` command while zigbee2mqtt is starting and I can indeed see the device being accessed

```
lcarlier@raspberrypi3:~ $ sudo ls -l /proc/[0-9]*/fd/ |grep /dev/ttyUSB0
lrwx------ 1 root root 64 May 12 17:06 21 -> /dev/ttyUSB0
```

Once I shut off my zigbee2mqtt container, then the command doesn't return anything.

Is there anyway for me to check that I flashed the right firmware on my dongle?

lcarlier, 2025-05-12:
To rule out any conflict with any other software, I tried to setup the dongle on my desktop running Manjaro. I'm getting exactly the same error.

On my desktop, there is no HA or anything like that. So nothing can use my dongle there.

This is quite a beefy desktop so I also rule out the "Make sure your computer has enough resource" (https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html#make-sure-your-computer-has-enough-resources-available) as well as the power supply issue.

Are there any way to check the installed firmware on the dongle?

Can we also maybe connect manually to the dongle via /dev/ttyUSB0 and run some commands? I blindly tried with minicom with the baudrate 115200 but I get no response.

chris-1243, 2025-05-12:
You may try the tool from Smlight

https://smlight.tech/flasher/#other_cc

Or just flash your dongle again with the good firmware

lcarlier, 2025-05-12:
Oh wow! Finally.

I tried to flash the firmware with the TI flashing tool but apparently it didn't go well...

Now I did it with https://github.com/xyzroe/ZigStarGW-MT and I'm getting to the next step

```
May 12 22:29:30 raspberrypi3 docker-compose[120152]: zigbee2mqtt    | [2025-05-12 22:29:30] info:         z2m: Currently 0 devices are joined.
May 12 22:29:30 raspberrypi3 docker-compose[120152]: zigbee2mqtt    | [2025-05-12 22:29:30] info:         z2m: Connecting to MQTT server at mqtt://localhost:1883
May 12 22:29:30 raspberrypi3 docker-compose[120152]: zigbee2mqtt    | [2025-05-12 22:29:30] error:         z2m: MQTT failed to connect, exiting... (connect ECONNREFUSED 127.0.0.1:1883)
May 12 22:29:30 raspberrypi3 docker-compose[120152]: zigbee2mqtt    | [2025-05-12 22:29:30] info:         z2m: Stopping zigbee-herdsman...
```

i.e. zigbee to mqtt cannot connect to the broker. But that has nothing to do with the firmware or dongle then.

I had to setup the ip address of the host from the container i.e. mqtt://172.17.0.1:1883

```
May 12 22:35:26 raspberrypi3 docker-compose[121935]: zigbee2mqtt    | [2025-05-12 22:35:26] info:         z2m: Currently 0 devices are joined.
May 12 22:35:26 raspberrypi3 docker-compose[121935]: zigbee2mqtt    | [2025-05-12 22:35:26] info:         z2m: Connecting to MQTT server at mqtt://172.17.0.1:1883
May 12 22:35:26 raspberrypi3 docker-compose[121935]: zigbee2mqtt    | [2025-05-12 22:35:26] info:         z2m: Connected to MQTT server
May 12 22:35:26 raspberrypi3 docker-compose[121935]: zigbee2mqtt    | [2025-05-12 22:35:26] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
May 12 22:35:26 raspberrypi3 docker-compose[121935]: zigbee2mqtt    | [2025-05-12 22:35:26] info:         z2m: Started frontend on port 8080
May 12 22:35:26 raspberrypi3 docker-compose[121935]: zigbee2mqtt    | [2025-05-12 22:35:26] info:         z2m: Zigbee2MQTT started!
```

Now zigbee2mqtt is up and running.

Thanks a lot for the support! I'm closing this ticket as it is solved

closed 2025-05-12 by lcarlier

chris-1243, 2025-05-12:
As Docker may change the IP address when restarting, I would use mdns instead
```
mqtt:
  server: mqtt://mosquitto:1883
```

lcarlier, 2025-05-12:
Seems like the mdns entry isn't found in my case

```
May 12 22:46:05 raspberrypi3 docker-compose[124044]: zigbee2mqtt    | [2025-05-12 22:46:05] info:         z2m: Currently 0 devices are joined.
May 12 22:46:05 raspberrypi3 docker-compose[124044]: zigbee2mqtt    | [2025-05-12 22:46:05] info:         z2m: Connecting to MQTT server at mqtt://mosquitto:1883
May 12 22:46:05 raspberrypi3 docker-compose[124044]: zigbee2mqtt    | [2025-05-12 22:46:05] error:         z2m: MQTT failed to connect, exiting... (getaddrinfo ENOTFOUND mosquitto)
May 12 22:46:05 raspberrypi3 docker-compose[124044]: zigbee2mqtt    | [2025-05-12 22:46:05] info:         z2m: Stopping zigbee-herdsman...
```

chris-1243, 2025-05-12:
It depends on your mqtt broker container name and how you setup your docker network. I should have been more precise