Koenkk/zigbee2mqtt #27385: Ikea RODRET (E2201) not showing action in HA

p4r4ll4x, 2025-05-12:
### What happened?

I have a Rodret switch from Ikea.
I can pair it and it show action in z2m console, but in Home Assistant, i do not see the action entity.

[image]

I already deleted the device in HA and Z2M and repaired, but i never get the action entity.
Am i missing something or is there something wrong?

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

EmberZNet 8.0.2 [GA]

### Adapter

SLZB-06Mg24

### Setup

Hassio with Z2M addon

Koenkk, 2025-05-12:
This is one of the Zigbee2MQTT 2.0.0 breaking changes, from https://github.com/Koenkk/zigbee2mqtt/discussions/24198

> - All action sensors are now disabled by default (`sensor.*_action` entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead. In case you really need the action sensors, add the following to your `configuration.yaml`.
>   ```yaml
>   homeassistant:
>     legacy_action_sensor: true
>   ```

This is one of the Zigbee2MQTT 2.0.0 breaking changes, see https://github.com/Koenkk/zigbee2mqtt/discussions/24364 on how to fix this.

closed 2025-05-12 by Koenkk

Orgjvr, 2025-08-17:
I am having the exact same problem. This button does not have any entity in Home Assistant showing the state of the button. There is a "action" in MQTT which shows current state, but nothing going to HA. 

When following the instruction in [Via MQTT device trigger (recommended)](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) there is no trigger for "action". 
The only options for trigger is:
- battery_level
- voltage
- changed_states (update availability changed)
- turned_off (became up-to-date)
- turned_on (got an update available)
- pressed (Identify has been pressed)

@p4r4ll4x Did you get a fix for this button?

I really think that the Home Assistant auto discovery message is missing some info? Can someone guide me in where I can look into the auto discovery message for this button? 

@Koenkk I know you closed this issue, but I really do not think it is fixed. I do have a workaround by creating a manual sensor in Home Assistant configuration which subscribes to the MQTT topic, but I think we are trying to move away from these legacy ways of doing things.