Koenkk/zigbee2mqtt #27395: Error: ZCL command - Timeout from my devices

sherkan-666, 2025-05-13:
### What happened?

Hello,
I've had a problem for 2 days. I thought it was due to an update to Home Assistant or Z2M, but even after restoring my backups, I still have the problem.
My devices that send back information seem to be working correctly (not all of them), for example: my Aqara temperature sensors, my Aqara opening sensors, but I can no longer control my other equipment, for example: connected sockets of different brands
In the logs I see timeouts with the following message: 'Error: ZCL command
Looking around a bit, I see that some people have solved the problem by adding a USB extension lead to their adapter, but I already have a 2m USB extension lead
I haven't made any particular changes recently except for updating HA and Z2M
I'm still running on a NUC in Proxmox, the adapter is still connected to the same USB socket and I haven't added any new equipment to my Zigbee network.
Restarting Z2M doesn't change anything, I've also tried changing the USB port or shifting my adapter but it's no better.

[image]

### Zigbee2MQTT version

2.3.0-1

### Adapter firmware version

0x26660700

### Adapter

ConBee2/RaspBee2

### Setup

Z2m in module of Home Assistant 
Version of Home Assistant : 
Core 2025.5.1
Supervisor 2025.05.0
Operating System 15.2
Interface utilisateur 20250509.0

cojaski, 2025-05-14:
Same problems here using ZBDongle-E.  I've actually found a better response rate by moving my usb BACK to the NUC.  About 1/5 tries my actions will work even with the error, but try to switch them back, same error, no response.

sherkan-666, 2025-05-20:
I uninstalled, deleted the entire Zigbee2MQTT folder in the HA conf directory (otherwise it wouldn’t work) and reinstalled Z2M.
I’m in the process of joining my equipment but it seems to be working again.
I don’t know the root cause of the problem though.

closed 2025-05-20 by sherkan-666