Koenkk/zigbee2mqtt #27396: [Wrong device]: LED RGBCW controller Miboxer FUT039Z wrongly identified as FUT037Z+

opvos, 2025-05-13:
### Link

https://www.zigbee2mqtt.io/devices/FUT039Z.html

### Model

FUT039Z

### Description

RGB CCT Led Controller

### Vendor

MiBoxer

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/FUT039Z.png

### Database entry

tba

### Notes

Request to add modelID: _TZB210_rs0ufzwg to whitelabel entry in zigbee-herdsman-converters/src/devices/tuya.ts >> FUT039Z 

Correct fingerprint (FUT039Z): 
        zigbeeModel: ["TS0505B"],
        model: "TS0505B_1",
        vendor: "Tuya",
        description: "Zigbee RGB+CCT light",

-
tuya.whitelabel("MiBoxer", "FUT039Z", "RGB+CCT LED controller", ["_TZ3210_jicmoite"]),
+
tuya.whitelabel("MiBoxer", "FUT039Z", "RGB+CCT LED controller", ["_TZ3210_jicmoite", "_TZB210_rs0ufzwg"]),

chris-1243, 2025-05-14:
Added in

https://github.com/Koenkk/zigbee-herdsman-converters/pull/9344

Koenkk, 2025-05-14:
Thanks @chris-1243 

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-05-14 by Koenkk

opvos, 2025-05-15:
Excellent, thanks a lot!

djsomi, 2025-06-11:
Hi Guys, 

I am suffering with these devices, 

TS0505B
_TZB210_jogusbq1
_TZB210_rs0ufzwg

I have several of them, they are working fine in Homebridge-Homekit, but in Home Assistant the Color Temperature selector missing. Can't set white light.

Can you help me out how to solve?

Many thanks!