Koenkk/zigbee2mqtt #27410: Schneider Electric CCT5010-0001 power-on behaviour unsupported attribute

tomlut, 2025-05-14:
### What happened?

I have fifteen Clipsal Wiser CLP5010WDZ dimmer pucks that are discovered as Schneider Electric CCT5010-0001 devices. For the most part these operate as expected. Clipsal is the Australian Supplier of Schneider Electric devices. 

However the power-on behaviour control attribute for the CCT5010-0001 component is not functioning on any of the devices. It produces the following error in Z2M when the control value is changed:

[image]

It also occasionally produces an error in the HA log even though I do not attempt to change this attribute anywhere in my config:

```
33m2025-05-14 05:23:23.430 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'power_on_behavior' when rendering '{{ value_json.power_on_behavior }}'
```

### What did you expect to happen?

Either the control to change as per the set value or the control not to be exposed.

### How to reproduce it (minimal and precise)

Attempt to change the power-on behaviour control of a Schneider Electric CCT5010-0001 device from the device's "Exposes" page.

### Zigbee2MQTT version

2.3.0-1

### Adapter firmware version

20240710

### Adapter

Tubes' CC2652P2 Based Zigbee to Ethernet Serial Coordinator, ZStack3x0

### Setup

HA Add-on.

NUC with Generic x86-64 image, core-2025.5.1, OS 15.2

### Debug log

```
info 2025-05-15 09:24:47z2m: Successfully changed options
info 2025-05-15 09:24:48z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"r34i4-1"}'
info 2025-05-15 09:24:48z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Lounge Dehumidifier Monitor', payload '{"child_lock":"LOCK","current":0.57,"device":{"applicationVersion":112,"friendlyName":"Lounge Dehumidifier Monitor","hardwareVersion":0,"ieeeAddr":"0xa4c13820322b6c12","manufacturerID":4107,"manufacturerName":"_TZ3000_0zfrhq4i","model":"TS011F_plug_3","networkAddress":54743,"powerSource":"Mains (single phase)","softwareBuildID":"1.0.5\u0000","stackVersion":2,"type":"Router","zclVersion":3},"energy":120.73,"indicator_mode":"off/on","linkquality":87,"power":2,"power_outage_memory":"on","state":"ON","update":{"installed_version":268514823,"latest_version":268514823,"state":"idle"},"voltage":244}'
info 2025-05-15 09:25:02z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LED Table', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"device":{"applicationVersion":192,"dateCode":"","friendlyName":"LED Table","hardwareVersion":1,"ieeeAddr":"0xa4c138d62297623c","manufacturerID":4417,"manufacturerName":"_TZ3000_cehuw1lw","model":"TS011F_plug_1","networkAddress":56576,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":0.12,"indicator_mode":"off/on","linkquality":61,"power":0,"power_outage_memory":"off","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":244}'
error 2025-05-15 09:25:07z2m: Publish 'set' 'power_on_behavior' to 'Bathroom Light' failed: 'Error: Got `UNSUPPORTED_ATTRIBUTE` error, device does not support power on behaviour'
info 2025-05-15 09:25:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LED Table', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"device":{"applicationVersion":192,"dateCode":"","friendlyName":"LED Table","hardwareVersion":1,"ieeeAddr":"0xa4c138d62297623c","manufacturerID":4417,"manufacturerName":"_TZ3000_cehuw1lw","model":"TS011F_plug_1","networkAddress":56576,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":0.12,"indicator_mode":"off/on","linkquality":61,"power":0,"power_outage_memory":"off","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":244}'
info 2025-05-15 09:25:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LED Table', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"device":{"applicationVersion":192,"dateCode":"","friendlyName":"LED Table","hardwareVersion":1,"ieeeAddr":"0xa4c138d62297623c","manufacturerID":4417,"manufacturerName":"_TZ3000_cehuw1lw","model":"TS011F_plug_1","networkAddress":56576,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":0.12,"indicator_mode":"off/on","linkquality":61,"power":0,"power_outage_memory":"off","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":243}'
info 2025-05-15 09:25:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Dishwasher', payload '{"child_lock":"UNLOCK","current":0.04,"device":{"applicationVersion":112,"friendlyName":"Dishwasher","hardwareVersion":0,"ieeeAddr":"0xa4c1380b8ae54549","manufacturerID":4107,"manufacturerName":"_TZ3000_0zfrhq4i","model":"TS011F_plug_3","networkAddress":18394,"powerSource":"Mains (single phase)","softwareBuildID":"1.0.5\u0000","stackVersion":2,"type":"Router","zclVersion":3},"energy":19.01,"indicator_mode":"off/on","linkquality":105,"power":0,"power_outage_memory":"off","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":245}'
```

tomlut, 2025-07-15:
Still an issue.

tomlut, 2025-09-14:
Still an issue.

tomlut, 2025-11-14:
Still an issue.

tomlut, 2026-01-14:
Still an issue

tomlut, 2026-03-16:
Still an issue

tomlut, 2026-07-17:
Still an issue.

hole222, 2026-09-17:
Still an issue.

andrei-lazarov, 2026-09-18:
It's not exposed anymore
https://www.zigbee2mqtt.io/devices/CCT5010-0001.html