Koenkk/zigbee2mqtt #27417: z2m: Error: Error while opening socket on SLZB-06 after firmware update

JC38, 2025-05-16:
### What happened?

Hello,

After upgrading the Zigbee radio firmware to the latest, Z2M fails to start again with this error :
 Starting Zigbee2MQTT without watchdog.
[2025-05-16 08:36:01] error:    zh:zstack:znp: Socket error Error: read ECONNRESET
[2025-05-16 08:36:01] error:    z2m: Error while starting zigbee-herdsman
[2025-05-16 08:36:01] error:    z2m: Failed to start zigbee-herdsman
[2025-05-16 08:36:01] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-05-16 08:36:01] error:    z2m: Exiting...
[2025-05-16 08:36:01] error:    z2m: Error: Error while opening socket
    at Socket.<anonymous> (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:160:24)
    at Socket.emit (node:events:530:35)
    at emitErrorNT (node:internal/streams/destroy:170:8)
    at emitErrorCloseNT (node:internal/streams/destroy:129:3)
    at processTicksAndRejections (node:internal/process/task_queues:90:21)

I rolled back the radio firmware to its previous version, the ESP firmware and also restore my Z2M backup to yesterday but nothing change, I can't use Z2M anymore.

I also tried to reinstall a lot of different firmwares but without success. I also have activated the multi-socket option without success.
Actually, the SLZB-06 is running with 20240710 radio firmware and v2.8.5.dev12 ESP firmware.
Which is giving this log:
Starting Zigbee2MQTT without watchdog.
[2025-05-16 10:34:59] error:    z2m: Error while starting zigbee-herdsman
[2025-05-16 10:34:59] error:    z2m: Failed to start zigbee-herdsman
[2025-05-16 10:34:59] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-05-16 10:34:59] error:    z2m: Exiting...
[2025-05-16 10:34:59] error:    z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:114:27)
    at Controller.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/controller/controller.ts:133:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:70:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:104:13)
    at start (/opt/zigbee2mqtt/index.js:149:5)

And inside the console :
[08:16:14] setup | Starting firmware: v2.8.5.dev12
[08:16:14] setup | FS mounted
[08:16:15] ConfigHelper | load config
[08:16:15] ConfigHelper | config open: Ok
[08:16:15] setup | Reboot reason: SOFTWARE
[08:16:15] setup | starting ZB1 serial on: 115200 baud
[08:16:15] setup | Coordinator mode: LAN
[08:16:15] setup | Device type: SLZB-06
[08:16:15] setup | Radio mode: "ZB COORD" Radio FW version: 20240710 Radio FW CH: DEV
[08:16:15] Network | init
[08:16:15] [LEDs init] | status: 1
[08:16:16] [LEDs init] | done
[08:16:17] Network | EVENT: 1 - ETH_START
[08:16:17] Network | EVENT: 3 - ETH_CONNECTED
[08:16:17] Network | EVENT: 5 - ETH_GOT_IP
[08:16:17] Network | [mDNS] Started
[08:16:17] Network | ETH init: OK
[08:16:17] Network | ETH static IP config: OK
[08:16:17] Network | ETH MAC: A8:42:E3:46:A7:67 IPv4: 192.168.0.9 GW: 192.168.0.254 Speed: 100Mbps DNS1: 8.8.8.8 DNS2: 8.8.4.4
[08:16:17] Network | fireNetworkUp
[08:16:17] taskZB | Starting server on port: 6638
[08:16:17] setup | Filesystem size: 3456
[08:16:17] Web | Web services init done
[08:16:17] EventSender | [_handleTask] start task
[08:16:17] setup | Filesystem used: 192
[08:16:17] time | time sync start
[08:16:17] internet | connected
[08:16:17] Network | EVENT: 5 - ETH_GOT_IP
[08:16:17] setup | done
[10:16:17] Network | [mDNS] Started
[10:16:17] Network | ETH MAC: A8:42:E3:46:A7:67 IPv4: 192.168.0.9 GW: 192.168.0.254 Speed: 100Mbps DNS1: 8.8.8.8 DNS2: 8.8.4.4
[10:16:17] time | timezone: CET-1CEST,M3.5.0,M10.5.0/3
[10:16:17] time | Friday, May 16 2025 10:16:17
[10:16:17] taskTimeSync | Heap: 1376
[10:16:17] time | stop task
[10:16:18] Network | [POST] result: 200
[10:16:18] Network | [POST] response: Previous records removed successfully.New record [on] created successfullyNew record [lo] created successfully
[10:16:18] stats | Statistics sent
[10:16:18] stats | Heap: 2460
[10:16:26] EventSender | [_handleNewClient] new client: 192.168.0.78
[10:16:41] EventSender | [_handleNewClient] new client: 192.168.0.22
[10:34:40] taskZB | New client: 192.168.0.87 id: 0
[10:35:00] taskZB | Client disconnected, id: 0

### Zigbee2MQTT version

2.30

### Adapter firmware version

20240710

### Adapter

SLZ06

### Setup

Z2M running as a LXC into Proxmox

JC38, 2025-05-16:
I'm going to close this ticket. After spending hours fighting with different firmware versions (radio and ESP), I was surprised to see Zigbee2MQTT running again.

closed 2025-05-16 by JC38

jojeju9428, 2025-05-25:
Hello
The SLZB-06 is currently operating with the 20240710 radio firmware and v2.8.5.dev12. Can you confirm whether the firmware is functioning correctly?
Thanks

JC38, 2025-05-25:
After my problem, i downgraded to 2.71 and 20240710. i will wait some weeks before trying to upgrade again.