Koenkk/zigbee2mqtt #27418: [New device support]: TS0049 or maybe differemt

Martijndutch, 2025-05-16:
### Link

https://nl.aliexpress.com/item/1005008585150701.html?srcSns=sns_WhatsApp&spreadType=socialShare&bizType=ProductDetail&social_params=61103857079&aff_fcid=f7a3241199e64700b1373af19ae98096-1747385497559-09291-_EIE3YRy&tt=MG&aff_fsk=_EIE3YRy&aff_platform=default&sk=_EIE3YRy&aff_trace_key=f7a3241199e64700b1373af19ae98096-1747385497559-09291-_EIE3YRy&shareId=61103857079&businessType=ProductDetail&platform=AE&terminal_id=36545328135048a5a8e302a0ecd0f351&afSmartRedirect=y

### Database entry

{"id":45,"type":"EndDevice","ieeeAddr":"0xa4c1385e824cec92","nwkAddr":36029,"manufId":4417,"manufName":"_TZ3000_ogjpfoyn","powerSource":"Battery","modelId":"TS0049","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[3,4,5,1,6,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_ogjpfoyn","zclVersion":3,"appVersion":72,"modelId":"TS0049","powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":1946894323},"lastSeen":1747384791751}

### Zigbee2MQTT version

2.3.0

### Comments

I've a water control valve, which identifies as TS0049 

[image]

It does pair, but whatever I do I can't get the device to respond. 

I'm not sure where to start troubleshooting

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['TS0049'],
    model: 'TS0049',
    vendor: '_TZ3000_ogjpfoyn',
    description: 'Automatically generated definition',
    extend: [m.battery(), m.onOff({"powerOnBehavior":false})],
    meta: {},
};
```

### What does/doesn't work with the external definition?

Nothing works.

FredrikM97, 2025-05-23:
Any other logs to share? Those logs contain no payload that external definition can map to.

Erol1982, 2025-06-15:
ich habe auch so ein änliches ventil. eingelernt geht  aber steuern kann mann es nicht. hat jemand eine lösung?

FredrikM97, 2025-06-20:
https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition

closed 2025-08-27 by github-actions[bot]