Koenkk/zigbee2mqtt #27427: Alarm _TZE204_hcxvyxa5 duration is reset in UI

ZTHawk, 2025-05-17:
### What happened?

I have the tuya _TZE204_hcxvyxa5 alarm. Everything is working fine (duration can be set, ringtone changed, ...).
But once an alarm is started then the duration field is reset to null (just UI).

When setting duration then state is:
```
{
    "alarm": "OFF",
    "last_seen": "2025-05-17T11:57:28+02:00",
    "linkquality": 120,
    "ringtone": "ringtone 1",
    "volume": "medium",
    "duration": 19
}
```

After alarm was palyed then state is:
```
{
    "alarm": "OFF",
    "last_seen": "2025-05-17T11:59:15+02:00",
    "linkquality": 132,
    "ringtone": "ringtone 1",
    "volume": "medium",
    "duration": null
}
```

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

7.4.5 [GA]

### Adapter

EmberZNet

### Setup

Addon on HA on Proxmox

ZTHawk, 2025-09-17:
Still applicable.
It is also resetting with the next status update.

ZTHawk, 2025-11-17:
Still applicable

ZTHawk, 2026-05-20:
> Hi. Can you obtain **debug** logs?

```
[20.5.2026, 07:00:02] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Türglocke', payload '{"alarm":"OFF","duration":18,"last_seen":"2026-05-20T06:59:49+02:00","linkquality":152,"ringtone":"ringtone 9","volume":"medium"}'
[20.5.2026, 07:00:02] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Türglocke', payload '{"alarm":"OFF","duration":18,"last_seen":"2026-05-20T06:59:59+02:00","linkquality":152,"ringtone":"ringtone 9","volume":"medium"}'
[20.5.2026, 07:01:30] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Türglocke', payload '{"alarm":"OFF","duration":null,"last_seen":"2026-05-20T07:01:28+02:00","linkquality":152,"ringtone":"ringtone 9","volume":"medium"}'
```

[z2m_bell_duration_log.zip](https://github.com/user-attachments/files/28041113/z2m_bell_duration_log.zip)

> Does the duration field work otherwise?

Yes

> Is it just for configuration?

Not sure if I understand this question. I set it and never need to touch it again. Well I do not see the value afterwards.

> Does it save "19s" on the device, and is that respected with every alarm?

The "19s" are respected everytime the ringtone is played.

> Just Z2M shows null?

Z2M and HA

andrei-lazarov, 2026-09-23:
Should be fixed by
- https://github.com/Koenkk/zigbee2mqtt/pull/32900