Koenkk/zigbee2mqtt #27432: [Feature request]: Enlarge device picture/photo/image on hover

wolph, 2025-05-18:
### Is your feature request related to a problem? Please describe

For some devices it's not always immediately clear what they are just by brand and model alone (at least not for me) and the picture can help. But by default the picture is really small and it would be useful if it would zoom in when you hover it so you can easily see what device you're looking at.

### Describe the solution you'd like

Enlarge images on hover so you can easily look at a bunch of devices in a few seconds.

I've made a little usercss that roughly does what I would like:

```css
#root > div.main > main > div > div > div > table > tbody > tr > td > div > img:hover{
    min-height: 200px;
    max-height: 500px !important;
    position: absolute;
    background-color: #3D4761;
    z-index: 1000;
    border: 2px dotted #fff;
}
```

[image]

The styling could use a little work but it's effective for now :)

### Describe alternatives you've considered

As an alternative I'm using a usercss right now, but I feel that this could be useful for more people.

### Additional context

None.

wolph, 2025-07-18:
Would still be useful :)

wolph, 2025-09-18:
Ping

Koenkk, 2025-09-18:
Please create frontend related issues in [this repo](https://github.com/Nerivec/zigbee2mqtt-windfront).