Koenkk/zigbee2mqtt #27437: [New device support]: Tuya Zigbee Mmwave Radar PIR Montion Sensor With Light Lumiance Temperature Humidity Detection

vdrahaniuk, 2025-05-18:
### Link

https://www.aliexpress.com/item/1005008903140286.html?spm=a2g0o.order_list.order_list_main.10.1f6e1802XiQY1T

### Database entry

{"id":14,"type":"EndDevice","ieeeAddr":"0xa4c138f03307f85b","nwkAddr":60357,"manufId":4742,"manufName":"_TZE200_uli8wasj","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280,1026,1029,1,1024],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"zclVersion":3}},"ssIasZone":{"attributes":{"iasCieAddr":"0x1cc089fffecddc3f","zoneState":0,"currentZoneSensitivityLevel":5}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2390,"tolerance":0}},"msRelativeHumidity":{"attributes":{"measuredValue":5360}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":27}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":32972}},"genIdentify":{"attributes":{"identifyTime":0}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x1cc089fffecddc3f","endpointID":1},{"cluster":1029,"type":"endpoint","deviceIeeeAddress":"0x1cc089fffecddc3f","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x1cc089fffecddc3f","endpointID":1},{"cluster":1024,"type":"endpoint","deviceIeeeAddress":"0x1cc089fffecddc3f","endpointID":1},{"cluster":1280,"type":"endpoint","deviceIeeeAddress":"0x1cc089fffecddc3f","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x1cc089fffecddc3f","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100},{"cluster":1029,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10},{"cluster":1024,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":5}],"meta":{}}},"appVersion":147,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1747564600094}

### Zigbee2MQTT version

2.3.0

### Comments

[image]

[image]

[image]

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_uli8wasj',
    description: 'Automatically generated definition',
    extend: [m.iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]}), m.temperature(), m.humidity(), m.battery(), m.illuminance()],
    meta: {},
};
```

### What does/doesn't work with the external definition?

I've tried add external definition, status changed to supported but only "humidity", "illuminance" and "temperature" send updates. 
Alarm_1 updates but there is some issue with it. It updates to OFF for few seconds and switches back to ON. Alarm_2 doesn’t switch. Temper also. 
Product doesn’t have specifications but based on description it has PIR sensor and also microwave sensor (detecting movement - alarm_2 ? )

kvazis, 2025-05-30:
Hi, try this converter - 

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_uli8wasj'}],
    model: 'TS0601_uli8wasj',
    vendor: 'Tuya',
    description: 'Tuya motion sensor with illuminance, temperature, humidity, battery, and calibration settings',
    fromZigbee: [tuya.fz.datapoints, fz.ias_occupancy_alarm_1],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.occupancy(),
        e.battery(),
        e.temperature(),
        e.humidity(),
        e.illuminance(),
        exposes.numeric('sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(19).withDescription('Motion detection sensitivity'),
        exposes.numeric('presence_time', ea.STATE_SET).withValueMin(5).withValueMax(28800).withUnit('s').withDescription('Presence clear timeout'),
        exposes.numeric('illuminance_interval', ea.STATE_SET).withValueMin(1).withValueMax(480).withUnit('min').withDescription('Illuminance reporting interval'),
        exposes.binary('led_enable', ea.STATE_SET, 'ON', 'OFF').withDescription('Enable or disable LED indication'),
        exposes.numeric('temperature_correction', ea.STATE_SET).withValueMin(-30).withValueMax(30).withUnit('°C').withDescription('Temperature correction x10'),
        exposes.numeric('humidity_correction', ea.STATE_SET).withValueMin(-10).withValueMax(10).withUnit('%').withDescription('Humidity correction'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'occupancy', tuya.valueConverter.trueFalse1],
            [101, 'humidity', tuya.valueConverter.raw],
            [106, 'illuminance', tuya.valueConverter.raw],
            [110, 'battery', tuya.valueConverter.raw],
            [111, 'temperature', tuya.valueConverter.divideBy10],
            [2, 'sensitivity', tuya.valueConverter.raw],
            [102, 'presence_time', tuya.valueConverter.raw],
            [107, 'illuminance_interval', tuya.valueConverter.raw],
            [108, 'led_enable', tuya.valueConverter.onOff],
            [104, 'humidity_correction', tuya.valueConverter.raw],
            [105, 'temperature_correction', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```

vdrahaniuk, 2025-05-31:
> Hi, try this converter -
> 
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const definition = {
>     fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_uli8wasj'}],
>     model: 'TS0601_uli8wasj',
>     vendor: 'Tuya',
>     description: 'Tuya motion sensor with illuminance, temperature, humidity',
>     fromZigbee: [tuya.fz.datapoints, fz.ias_occupancy_alarm_1],
>     toZigbee: [tuya.tz.datapoints],
>     configure: tuya.configureMagicPacket,
>     exposes: [
>         e.occupancy(),
>         e.battery(),
>         e.temperature(),
>         e.humidity(),
>         e.illuminance(),
>         exposes.numeric('sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(19).withDescription('Motion detection sensitivity'),
>         exposes.numeric('presence_time', ea.STATE_SET).withValueMin(5).withValueMax(28800).withUnit('s').withDescription('Presence clear timeout'),
>         exposes.numeric('illuminance_interval', ea.STATE_SET).withValueMin(1).withValueMax(480).withUnit('min').withDescription('Illuminance reporting interval'),
>         exposes.binary('led_enable', ea.STATE_SET, 'ON', 'OFF').withDescription('Enable or disable LED indication'),
>         exposes.numeric('temperature_correction', ea.STATE_SET).withValueMin(-30).withValueMax(30).withUnit('°C').withDescription('Temperature correction x10'),
>         exposes.numeric('humidity_correction', ea.STATE_SET).withValueMin(-10).withValueMax(10).withUnit('%').withDescription('Humidity correction'),
>     ],
>     meta: {
>         tuyaDatapoints: [
>             [1, 'occupancy', tuya.valueConverter.trueFalse1],
>             [101, 'humidity', tuya.valueConverter.raw],
>             [106, 'illuminance', tuya.valueConverter.raw],
>             [110, 'battery', tuya.valueConverter.raw],
>             [111, 'temperature', tuya.valueConverter.divideBy10],
>             [2, 'sensitivity', tuya.valueConverter.raw],
>             [102, 'presence_time', tuya.valueConverter.raw],
>             [107, 'illuminance_interval', tuya.valueConverter.raw],
>             [108, 'led_enable', tuya.valueConverter.onOff],
>             [104, 'humidity_correction', tuya.valueConverter.raw],
>             [105, 'temperature_correction', tuya.valueConverter.raw],
>         ],
>     },
> };
> 
> module.exports = definition;
> ```

Hi, @kvazis 
Thank you for response
In general it works, but with some issues.

1. Sensitivity works only for value <= 2. If I choose 3 or more then Occupancy status changes consistently (included Presence time). So if Presence time = 10 sec and Sensitivity > 2 then it switches off / on + 10 sec / off /  on + 10 sec /  (without any movements and I was in another room).

2. When Temperature correction or Humidity correction changed then Occupancy triggered. It seems like the occupancy status is triggered whenever any other status changes, such as illuminance / Temperature / Humidity  - after changes, Occupancy also changes to detected.

3. Might not be a "converter" issue, but the PIR sensor detects movement only within 1-1.5 meters. If the distance is greater, the device doesn’t detect movements.