Koenkk/zigbee2mqtt #27452: S60ZBTPG SONOFF iPlug Zigbee Smart Plug

ChrisThomas42, 2025-05-19:
### Link

https://itead.cc/product/sonoff-iplug-zigbee-smart-plug-s60-series/ref/3/

### Database entry

{"id":72,"type":"Router","ieeeAddr":"0xa4c13800e064ffff","nwkAddr":17213,"manufId":4742,"manufName":"SONOFF","powerSource":"Mains (single phase)","modelId":"S60ZBTPG","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,1794,2820,2821,64599,64529],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"S60ZBTPG","manufacturerName":"SONOFF","powerSource":1,"zclVersion":3,"appVersion":16,"stackVersion":2,"hwVersion":16,"dateCode":"20250411","swBuildId":"1.0.2"}},"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c9ee61","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":16,"stackVersion":2,"hwVersion":16,"dateCode":"20250411","swBuildId":"1.0.2","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1747677661040}

### Zigbee2MQTT version

2.3.0

### Comments

The Sonoff S60ZBTPG appears to be a UK version of S60ZBTPF, i.e. UK plug.  The S60ZBTPF entry has ```, m.electricityMeter()```  entry and when I look in the Dev console, I also have this entry.  However, this is missing from Sonoff.ts in line in line 1553, so I am making a naive assumption that add this line would make the device work?

I am having trouble understanding what I need to do to make the external definition (below) work, I haven't got it to be imported as yet.

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['S60ZBTPG'],
    model: 'S60ZBTPG',
    vendor: 'SONOFF',
    description: 'SONOFF iPlug Zigbee Smart Plug',
    extend: [m.onOff({"powerOnBehavior":false}), m.electricityMeter()],
};
```

### What does/doesn't work with the external definition?

Any guidance, advice or observations to help me understand would be gratefully received.

Koenkk, 2025-05-20:
> I haven't got it to be imported

Did you put the file in the `data/external_converters` folder?

ChrisThomas42, 2025-05-20:
This is where I have to show my ignorance, so please bear with me, I also on holiday with sporadic WiFi...  Under the config folder under zigbee2mqtt I created a ```data/external_converters``` folder and then created a file called ```sonoff_S60ZBTPG.mjs```.  Then I wasn't sure if I had to do anything else, I just restarted Z2M.  

I have limited access until the weekend, is this the right place to put the .mjs and do I need to do anything else for it to be read/used?

Koenkk, 2025-05-21:
That's the right place, check the z2m log if there are any issues when loading the ext converter.

ChrisThomas42, 2025-05-21:
Does the ext converter check that folder for files "automagically"?  I'm not seeing anything related to ext converter or this device in the logs (that I can see from holiday) does it only check on startup perhaps? I'm also not seeing the power consumption in the device.  Is there a way for me to see if that folder is being read and that that file is ok?

Koenkk, 2025-05-22:
>  does it only check on startup perhaps?

yes only on startup.

ChrisThomas42, 2025-05-23:
I'm back home, the device is connected and I can turn it on and off.  I've restarted Zigbee2MQTT (and HA) but I'm not seeing any change in device details.

If I look in the logs in Z2M or the log files under ```zigbee2mqtt/log``` I cannot see any entry relating to ```external_components```, the device is just reporting the same as before.

So what am I doing wrong or missing out!?

ChrisThomas42, 2025-05-23:
In this folder ```\\<Home Assistant IP Address>\config\zigbee2mqtt\data\external_converters``` I have a file called ```sonoff_S60ZBTPG.mjs```.

This file contains this
```
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['S60ZBTPG'],
    model: 'S60ZBTPG',
    vendor: 'SONOFF',
    description: 'SONOFF iPlug Zigbee Smart Plug',
    extend: [m.onOff({"powerOnBehavior":false}), m.electricityMeter()],
};
```
I have restarted Zigbee2mqtt from the add-on and from within the WebUI.  I have also restarted HA.

I have looked in the logs in the WebUI and in the ```\\<HA IP Address>\config\zigbee2mqtt\log``` folder at the logs files and I can't see any reference to ```external_converters```.

bverwijst, 2025-05-24:
Interestingly ZHA does see the power draw directly so it should absolutely be possible: https://www.youtube.com/watch?v=SwDkFeE2cl8.

Happy to help test too, I'm running Z2M and HA in Docker.

ChrisThomas42, 2025-06-09:
Well it's now working perfectly!  I see the documentation is also updated and I'm getting all of those settings reported!

Thank you!

lluisd, 2025-08-25:
> Hi guys, this device is NOT well supported yet. I discovered that with fw version 1.0.1 (latest), **current** (and so **power**) sensor remains blocked on the last read. If with switch ON the absorption is, for example, 0.5A, when I put on OFF the switch the current (and so the calculated power) stays on 0.5A, instead of descending to 0A.

this also happens to me and more people here: https://github.com/Koenkk/zigbee2mqtt/issues/27054

ChrisThomas42, 2025-10-27:
I don't think this is stale, at least not yet!

ChrisThomas42, 2025-10-31:
In a day at home between visiting family, I have the following extract from the log file.

[S60-log.txt](https://github.com/user-attachments/files/23265546/S60-log.txt)

I can spend more time from the middle of next week if this is not correct/good enough info.

ChrisThomas42, 2025-11-03:
Ah, somehow followed different instructions!  I will get on it as soon as we are home and time permits!

manusanudo, 2025-11-19:
Hello,

I am completly new to Zigbee2MQTT and I bought this smart plug as my first one to experiment and learn. I got it today and managed to run Z2M in a docker container, connect it with my coordinator and pair the plug. Very impressed with how easy the process is.

My disapointment come when I could turn the plug on and off, received the voltage level but current level, and hence power level, always shows 0.
My plug is running firmware 1.0.2, which I don't know if changes anything in the way it is reporting this data. I suppose the issue is somehow related with this case and the linked one.

As I said, I completly new to Z2M and Zigbee technology so I am not sure if the attached log is what you need. I tried to capture the output when clicking on "Reconfigure" the device:

[debug_S60ZBTPG_fw1_0_2.txt](https://github.com/user-attachments/files/23639316/debug_S60ZBTPG_fw1_0_2.txt)

bobo-cher, 2025-11-21:
Currently the device exposes energy toady, yesterday and month:

> [21.11.2025, 20:45:26] zhc:tz: Read result of 'customClusterEwelink': {"energyToday":30,"energyYesterday":50,"energyMonth":860}

Can you please add "total energy"? It is in a different cluster:

> [21.11.2025, 20:33:49] zhc:tz: Read result of 'seMetering': {"currentSummDelivered":3030}

This property is useful for energy dashboard in home assistant.

Thanks.

closed 2025-11-23 by Koenkk