Koenkk/zigbee2mqtt #27454: [New device support]: Support request for Tuya 6/8-button wall switch (_TZE204_tdhnhhiy / _TZE204_nvxorhcj)

francisdeabreu, 2025-05-20:
### Link

www.https://www.novadigitalsmart.com.br/

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xa4c1386da2b96f47","nwkAddr":57412,"manufId":4417,"manufName":"_TZE204_nvxorhcj","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�ټ/\u0013�ڼ/e�ڼ/\u0012","65506":56,"65508":0,"appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1747705341757} {"id":4,"type":"Router","ieeeAddr":"0xa4c13880a96e0794","nwkAddr":32513,"manufId":4417,"manufName":"_TZE204_tdhnhhiy","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�ټ/\u0013�ټ/e�ټ/\u0012�ټ/\u0016","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1747705397145}

### Zigbee2MQTT version

2.3.0 commit: unknown

### Comments

### Device support request for Tuya wall switch

I'm requesting support for the following Tuya Zigbee multi-button wall switches in Zigbee2MQTT. The devices pair and interview successfully but do not send any messages and are marked as "unsupported".

#### Device information

* **Device type:** Tuya wall switch (6-button and 8-button variants)
* **Model ID:** TS0601
* **Manufacturer IDs:**

  * `_TZE204_tdhnhhiy`
  * `_TZE204_nvxorhcj`

#### Environment

* **Zigbee2MQTT version:** 2.3.0
* **Coordinator firmware:** ZStack3x0 20230507

#### Logs during pairing

(Logs for both devices showing successful pairing and unsupported status)

*See full log output below.*

#### Description

These are Tuya multi-button switches. After successful pairing, no messages are emitted from button presses, and the devices appear as "unsupported". I'd like to request support so that the buttons can be recognized and used in automations.

---

**Logs during pairing:**

**info** 2025-05-18 17:48:15`z2m: Zigbee: allowing new devices to join.`

**info** 2025-05-18 17:48:15`z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254},"status":"ok","transaction":"4iwzq-4"}'`

**info** 2025-05-18 17:48:19`zh:controller: Interview for '0xa4c13880a96e0794' started`

**info** 2025-05-18 17:48:19`z2m: Device '0xa4c13880a96e0794' joined`

**info** 2025-05-18 17:48:19`z2m: Starting interview of '0xa4c13880a96e0794'`

**info** 2025-05-18 17:48:19`z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13880a96e0794","ieee_address":"0xa4c13880a96e0794"},"type":"device_joined"}'`

**info** 2025-05-18 17:48:19`z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13880a96e0794","ieee_address":"0xa4c13880a96e0794","status":"started"},"type":"device_interview"}'`

**info** 2025-05-18 17:48:19`z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13880a96e0794","ieee_address":"0xa4c13880a96e0794"},"type":"device_announce"}'`

**info** 2025-05-18 17:48:19`zh:controller: Succesfully interviewed '0xa4c13880a96e0794'`

**info** 2025-05-18 17:48:19`z2m: Successfully interviewed '0xa4c13880a96e0794', device has successfully been paired`

**warning** 2025-05-18 17:48:19`z2m: Device '0xa4c13880a96e0794' with Zigbee model 'TS0601' and manufacturer name '_TZE204_tdhnhhiy' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html`

**info** 2025-05-18 17:48:19`z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c13880a96e0794/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":1,"identifiers":["zigbee2mqtt_0xa4c13880a96e0794"],"manufacturer":"_TZE204_tdhnhhiy","model":"Automatically generated definition","model_id":"TS0601","name":"0xa4c13880a96e0794","via_device":"zigbee2mqtt_bridge_0x00124b0029e7ce9e"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"0xa4c13880a96e0794_linkquality","origin":{"name":"Zigbee2MQTT","sw":"2.3.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c13880a96e0794","unique_id":"0xa4c13880a96e0794_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'`

[... 23 lines cut ...]

**info** 2025-05-18 17:52:14`z2m: Successfully interviewed '0xa4c1386da2b96f47', device has successfully been paired`

**warning** 2025-05-18 17:52:14`z2m: Device '0xa4c1386da2b96f47' with Zigbee model 'TS0601' and manufacturer name '_TZE204_nvxorhcj' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html`

**info** 2025-05-18 17:52:14`z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c1386da2b96f47/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":1,"identifiers":["zigbee2mqtt_0xa4c1386da2b96f47"],"manufacturer":"_TZE204_nvxorhcj","model":"Automatically generated definition","model_id":"TS0601","name":"0xa4c1386da2b96f47","via_device":"zigbee2mqtt_bridge_0x00124b0029e7ce9e"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"0xa4c1386da2b96f47_linkquality","origin":{"name":"Zigbee2MQTT","sw":"2.3.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c1386da2b96f47","unique_id":"0xa4c1386da2b96f47_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'`

**info** 2025-05-18 17:52:14`z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Automatically generated definition","exposes":[{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0601","options":[],"supports_ota":false,"vendor":"_TZE204_nvxorhcj"},"friendly_name":"0xa4c1386da2b96f47","ieee_address":"0xa4c1386da2b96f47","status":"successful","supported":false},"type":"device_interview"}'`

### External definition

```shell
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_tdhnhhiy',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

No status and no commands

Koenkk, 2025-05-20:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-05-20 by Koenkk

Joel90300, 2025-05-22:
[image]

marcelobrandt, 2025-10-29:
> For the 6-gang:
> 
> ```
>     // Backlight (DP 16)
>     if (dp === 16) {
>         const value = data[0];
>         result.backlight = value === 1 ? 'ON' : 'OFF';
>     }
> 
>     return result;
> ```
> 
> And, when creating a Z2M group, even though we select L3 (for instance), it's always identified as endpoint 1.

Hey eduseiden,

I got the 6-gang (although its regonized as 8-gang in zigbee2mqtt). I'm trying to make the backlight function work. Do you have the full code for the external definition?

thanks a lot

marcelobrandt, 2025-10-29:
> Hi [@marcelobrandt](https://github.com/marcelobrandt)
> 
> https://seiden.tech/ha/6gang_tdhnhhiy.js

worked perfectly, thanks!!