Koenkk/zigbee2mqtt #27456: Ikea E206 TARKVIND air purifier table fails to configer properly

Keeze, 2025-05-20:
### What happened?

Yesterday my son bought a TARKVIND air purifier table and I tried to add it to Z2M. All seems to go well, and the table shows up in Z2M and Home assistant. Now I noticed that the table does very slow updates. I also have a 2007 Ikea Air purefier and this updates mosly instantly.

When I try to reconfigure the table i get the following erros:

```
info 2025-05-20 08:50:10z2m: Configuring 'Luchtreiniger Tim'
error 2025-05-20 08:50:10z2m: Failed to configure 'Luchtreiniger Tim', attempt 5 (TypeError: Bind 0x1c34f1fffebe2c4d/1 manuSpecificIkeaAirPurifier from '28/1' failed (value.slice is not a function) at BuffaloZdo.writeIeeeAddr (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:238:48) at BuffaloZdo.buildBindRequest (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/zspec/zdo/buffaloZdo.ts:1035:22) at Function.buildRequest (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/zspec/zdo/buffaloZdo.ts:796:32) at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:44) at Object.bind (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.36.0/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:40:24) at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.36.0/node_modules/zigbee-herdsman-converters/src/lib/ikea.ts:440:29) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.36.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:27) at processTicksAndRejections (node:internal/process/task_queues:105:5) at Configure.configure (/app/lib/extension/configure.ts:124:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:49:25))
info 2025-05-20 08:50:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{},"error":"Failed to configure (Bind 0x1c34f1fffebe2c4d/1 manuSpecificIkeaAirPurifier from '28/1' failed (value.slice is not a function))","status":"error","transaction":"mr6m3-1"}'

```

I updated the device to the latest available firmware. And home assistant and Z2M are on the latest (stable) versions.

### What did you expect to happen?

Rapid device updates.

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

### Setup

Z2M is running as a add-on in Home Assistant OS under Proxmox.

sjorge, 2025-05-21:
Assuming you are talking about a **S**TARKVIND, can you try reinverviewing/configuring the device.
You're probbably getting slow updates because some of the reporting is missing.

I just removed and rejoined mine and I had no errors, my reporting looks like

[image]

I had no weird slice error inside zigbee-herdsman though, but I am not using a ConBee2 but zstack, if you hit the same error there might be some weirdness in the driver for how it decodes the payload.

Keeze, 2025-05-21:
Thnx, it is the Strakvind indeed.  I already tried reinverviewing/configuring the device and removed and rejoined it. I also have a Starkvind E207, (the standing one) and this is configured correct (as shown above). A have about 70 devices in my Zigbee network, so I don't think it is a  driver issue. 

[image]

closed 2025-07-30 by github-actions[bot]