Koenkk/zigbee2mqtt #27465: Sonoff Zigbee Smart Water Valve: Incorrect `on_time` Duration (Payload Seconds Do Not Match Valve Response)

TapioJarnfors, 2025-05-21:
### What happened?

### Description

When using a Sonoff Zigbee Smart Water Valve (see: https://www.zigbee2mqtt.io/devices/SWV.html#sonoff-swv) with Zigbee2MQTT, sending a payload with the `on_time` key does not result in the expected valve open duration. For example, sending `on_time: 11` opens the valve for 110 seconds, and sending `on_time: 13` results in an unexpected duration with an odd value reported back.

#### Device Information

- **Device Type:** EndDevice
- **Zigbee Model:** SWV
- **Zigbee Manufacturer:** SONOFF
- **Description:** Zigbee smart water valve
- **Support status:** Supported
- **IEEE Address:** 0x7cc6b6fffe0d0509
- **Network address:** 0xC348 / 49992
- **Firmware build date:** 20240820
- **Firmware version:** 1.0.4
- **Manufacturer:** SONOFF
- **Model:** SWV
- **Power:** Battery 100%
- **Interview state:** Successful

#### Test Cases

**Case 1: 11 Seconds**

Home Assistant Button:
```yaml
- name: "Open Water Valve for testing 11seconds"
  unique_id: "open_water_valve_testing_11"
  command_topic: "zigbee2mqtt/sonoff_smart_water_valve1/set"
  payload_press: '{"state": "ON", "on_time": 11}'
  icon: mdi:water-pump
```

Zigbee2MQTT Log & Valve Status:
```json
{
  "cyclic_timed_irrigation": {
    "current_count": 1,
    "irrigation_duration": 110,
    "irrigation_interval": 0,
    "total_number": 1
  },
  ...
}
```

**Case 2: 13 Seconds**

Home Assistant Button:
```yaml
- name: "Open Water Valve for testing 13seconds"
  unique_id: "open_water_valve_testing_13"
  command_topic: "zigbee2mqtt/sonoff_smart_water_valve1/set"
  payload_press: '{"state": "ON", "on_time": 13}'
  icon: mdi:water-pump
```

Zigbee2MQTT Log & Valve Status:
```json
{
  "cyclic_timed_irrigation": {
    "current_count": 1,
    "irrigation_duration": 239,
    "irrigation_interval": -1078132736,
    "total_number": 1
  },
  ...
}
```
#### Zigbee2MQTT Configuration

- **Zigbee2MQTT version:** 2.3.0-1  
- **MQTT server:** core-mosquitto (Home Assistant add-on)  
- **Serial adapter:** Ember (`/dev/ttyUSB0`)  
- **Adapter firmware:** 2.3.0  
- **Adapter hardware:** EmberZNet 7.4.4 [GA]  
- **Channel:** 25  
- **Frontend:** enabled (port 8099)  
- **Home Assistant integration:** enabled  
- **Device:**  
  - **IEEE:** `0x7cc6b6fffe0d0509`  
  - **Friendly name:** `sonoff_smart_water_valve1`

### What did you expect to happen?

#### Expected Behavior

- Setting `on_time` to N should keep the valve ON for N seconds.

#### Actual Behavior

- The value reported back (and the actual valve operation) does not match the payload's `on_time` value.

### How to reproduce it (minimal and precise)

#### Steps to Reproduce

1. Send an MQTT payload with `{ "state": "ON", "on_time": <N> }` to the topic `zigbee2mqtt/sonoff_smart_water_valve1/set` (via Home Assistant or MQTT Explorer).
2. Observe Zigbee2MQTT logs and the status reported by the valve.
3. Note the mismatch between sent `on_time` and reported/actual operation duration.

#### Relevant Hardware/Software
- Device: Sonoff Smart Water Valve (SWV)
- Home Assistant & MQTT Explorer (version: 0.4.0-beta.6)
- Zigbee2MQTT version: 2.3.0-1

#### Additional Information
- [Device reference](https://www.zigbee2mqtt.io/devices/SWV.html#sonoff-swv)
- Example logs and payloads included above.

---

If you need any additional information, please let me know

### Zigbee2MQTT version

Zigbee2MQTT version: 2.3.0-1 

### Adapter firmware version

2.3.0

### Adapter

Adapter hardware: EmberZNet 7.4.4 [GA] (Zigbee 3.0 USB Dongle Plus)

### Setup

### Setup

- **How do you run Z2M:** Home Assistant add-on  
- **Machine:** Raspberry Pi 4 (replace with your hardware if different)  
- **OS:** Home Assistant OS  
- **Zigbee2MQTT version:** 4  
- **Coordinator:** Ember, port `/dev/ttyUSB0`  
- **MQTT:** Core Mosquitto add-on (`mqtt://core-mosquitto:1883`)  
- **Note:** Running as Home Assistant OS add-on, host shell access not available for `uname -a`.

### Debug log

More detailed logs are available upon request

Marius-Romanus, 2025-05-26:
The same thing happens to me, I set it to 10 minutes and I have to turn it off manually, this is the example of my automation:

```
alias: Riego - Antes del atardecer
description: >
  Activa el riego del patio al atardecer si no ha llovido y no se espera lluvia.
  El riego dura 20 minutos y se apaga automáticamente desde la válvula.
triggers:
  - event: sunset
    trigger: sun
conditions:
  - condition: numeric_state
    entity_id: sensor.aemet_rain
    below: 0.2
  - condition: numeric_state
    entity_id: sensor.aemet_rain_probability
    below: 20
actions:
  - data:
      topic: zigbee2mqtt/Riego patio/set
      payload: "{\"state\": \"ON\", \"on_time\": 600, \"off_wait_time\": 300}"
    action: mqtt.publish
mode: single
```

Firmware Date: 20240820
Firmware Version: 1.0.4

TapioJarnfors, 2025-05-27:
Thanks for co-support, there is a temporary workaround as you can set ex. 0.3 for 30 seconds, for 10 minutes you could try 10 x 60 = 600 -> pure 60 should generate 600 seconds

- name: "Open Water Valve **30sec**"
  unique_id: "open_water_valve_30sec"
  command_topic: "zigbee2mqtt/sonoff_smart_water_valve1/set"
  payload_press: '{"state": "ON", "on_time": **3**}'
  icon: mdi:water-pump

- name: "Open Water Valve **5min**"
  unique_id: "open_water_valve_5min"
  command_topic: "zigbee2mqtt/sonoff_smart_water_valve1/set"
  payload_press: '{"state": "ON", "on_time": **30**}'
  icon: mdi:water-pump

- name: "Open Water Valve **30min**"
  unique_id: "open_water_valve_30min"
  command_topic: "zigbee2mqtt/sonoff_smart_water_valve1/set"
  payload_press: '{"state": "ON", "on_time": **180**}'
  icon: mdi:water-pump

- name: "Open Water Valve **60min**"
  unique_id: "open_water_valve_60min"
  command_topic: "zigbee2mqtt/sonoff_smart_water_valve1/set"
  payload_press: '{"state": "ON", "on_time": **360**}'
  icon: mdi:water-pump

grhallenbeck, 2025-07-09:
Chiming in here to add some additional info. I'm using the payload shape that sets the timer with the `cyclic_timed_irrigation` object and its properties. I wanted to run an interval with a duration of 15 minutes which triggered this error, and I noticed this pattern while attempting to find durations that worked:

* Every time this error presents, there is a range of 128 numbers that trigger the error.
* Every number within the range causes the `irrigation_duration` to be set to the same number.
* The `irrigation_interval` is always set to what appears to be a max negative int.
* The first valid number before the range is the number of `irrigation_duration` after sending the payload minus 112.
* The first valid number after the range is the number of `irrigation_duration` after sending the payload plus 17.

For the 15-minute duration example, the payload is:
```json
{
  "cyclic_timed_irrigation": {
    "current_count": 0, 
    "total_number": 1,
    "irrigation_duration": 900,
    "irrigation_interval": 0
  }
}
```

After sending this payload, the valve immediately resets from `irrigation_duration=900, irrigation_interval=0` to `interval_duration=1007, irrigation_interval=-1078132736`. The negative interval appears to cause the valve to immediately complete the cycle without ever opening the water valve.

Given the error range of 128 and the apparent overflow of the `irrigation_interval` this seems to me like an error in bit math or byte conversion, but I don't know enough about the internals of either zigbee2mqtt or the Sonoff water valves to know which one it's happening in.

TapioJarnfors, 2025-07-09:
**Greetings from Finland to Canada!**

I wanted to share some additional findings and suggestions:

1. **Alternative `on_time` Values**:
   - Setting `on_time` to `90` resulted in a 15-minute duration during my tests. This might be worth trying for you.
   - You could also experiment with fractional values like `90.1`. While I haven't tested this, adding a decimal can provide more precise control (e.g., `{"state": "ON", "on_time": 0.3}` runs for 3 seconds in my case).

2. **Code Modifications**:
   - I believe there might be a need to modify the `sonoff.ts` functions to improve reporting and feedback display from the valve. This could help address the mismatch between the payload and the actual operation.

3. **Additional Details**:
   - I've attached a Wireshark log image and some minor details for background.

I am just learning this also, but let me know if this is helpful or if there's anything else I can provide!

[image]

TapioJarnfors, 2025-08-02:
Version 2.6.0-1 seems to working okay and the times are updating correctly - thanks about the fix

closed 2025-08-02 by TapioJarnfors