Koenkk/zigbee2mqtt #27471: [Wrong device]: https://www.zigbee2mqtt.io/devices/S1-R.html

daNick73, 2025-05-21:
### Link

https://www.ubisys.de/produkte/produkte-licht-verbraucher/produkte-licht-verbraucher-s1-r/

### Model

S1-R (Series 2)

### Description

Power switch S1-R (Series 2)

### Vendor

Ubisys

### Picture (link)

https://www.ubisys.de/wp-content/themes/yootheme/cache/a6/S1-R_neu_2-1-a63dc12c.webp

### Database entry

{"id":4,"type":"Router","ieeeAddr":"0x001fee000000a9a9","nwkAddr":18551,"manufId":4338,"manufName":"ubisys","powerSource":"Mains (single phase)","modelId":"S1-R (5601)","epList":[1,2,3,232,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"S1-R (5601)","manufacturerName":"ubisys","powerSource":1,"zclVersion":8,"appVersion":1,"stackVersion":1,"hwVersion":16,"dateCode":"20240614-DE-FB1","swBuildId":"2.5.0"}},"seMetering":{"attributes":{"multiplier":1,"divisor":10000,"instantaneousDemand":403,"currentSummDelivered":218}},"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":255}}},"binds":[{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0031dff40d","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0031dff40d","endpointID":1}],"configuredReportings":[{"cluster":1794,"attrId":1024,"minRepIntval":5,"maxRepIntval":3600,"repChange":1},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"2":{"profId":260,"epId":2,"devId":260,"inClusterList":[0,3],"outClusterList":[3,5,6,8,768,64514],"clusters":{},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x001fee000000a9a9","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":260,"inClusterList":[0,3],"outClusterList":[3,5,6,8,768,64514],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"232":{"profId":260,"epId":232,"devId":1287,"inClusterList":[0,61,64512,64599],"outClusterList":[3,25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":16,"dateCode":"20240614-DE-FB1","swBuildId":"2.5.0","zclVersion":8,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1747856542175}

### Notes

Hi

This is a bit complicated, so please excuse if I do anything wrong here:

The Ubisys S1-R has an older an a newer version (new S1-R (Series 2)).
The new one has bluetooth.

The new one has also separate OTA Firmwares

https://www.ubisys.de/en/support/firmware/ lists

[ubisys S1-R Firmware Version 2.6.0, ubisys Zigbee Stack 4.6.0](https://www.ubisys.de/wp-content/uploads/10F2-7B35-0000-0004-02600460-spo-rms.ota.zigbee)

[ubisys S1-R Series 2 (with Zigbee and Bluetooth) Firmware Version 2.6.1, ubisys Zigbee Stack 4.6.3](https://www.ubisys.de/wp-content/uploads/10F2-7B45-0100-0100-02610463-ubisys-s1r-qpg6105.ota.zigbee)

Current images include the below (which seems to be the one above, but it is not identified as an OTA.
images/Ubisys/10F2-7B45-0100-0100-02610463-ubisys-s1r-qpg6105.ota.zigbee

THX for checking!

chris-1243, 2025-05-21:
It might be related to

https://github.com/Koenkk/zigbee-herdsman-converters/issues/9356

closed 2025-07-28 by github-actions[bot]

Torte1976, 2025-09-09:
Please reopen and fix  this issue. I have also a Power switch S1-R (Series 2) and it was detected as "S1-R (5601)"