Koenkk/zigbee2mqtt #27475: [Wrong device]: Tuya TS0011

Patbonamy, 2025-05-22:
### Link

https://www.tongou.com/fr/produit/disjoncteur-intelligent-sur-rail-din-zigbee-avec-comptage-energetique-to-q-sy1-jzt-de-tuya/

### Model

TO-Q-SY1-JZT

### Description

Din smart relay (with power monitoring via polling)

### Vendor

Tongou

### Picture (link)

https://www.tongou.com/fr/wp-content/uploads/2023/03/%E4%B8%BB%E5%9B%BE2-1.jpg

### Database entry

{"id":24,"type":"EndDevice","ieeeAddr":"0xa4c1383270118d9b","nwkAddr":9694,"manufId":4417,"manufName":"_TZ3000_gzvniqjb","powerSource":"Mains (single phase)","modelId":"TS0011","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_gzvniqjb","zclVersion":3,"appVersion":80,"modelId":"TS0011","powerSource":3}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x847127fffedca69a","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1747926187305}

### Notes

I try severals times to pairred but no luck

chris-1243, 2025-05-22:
Could you try this converter and report if everything is working. Add it as an `*.mjs` file.

Check this link to know how to add it

https://www.zigbee2mqtt.io/advanced/more/external_converters.html

```javascript
import * as tuya from "zigbee-herdsman-converters/lib/tuya";
import * as fz from "zigbee-herdsman-converters/converters/fromZigbee";
import * as tz from "zigbee-herdsman-converters/converters/toZigbee";
import * as exposes from "zigbee-herdsman-converters/lib/exposes";
import * as reporting from "zigbee-herdsman-converters/lib/reporting";

const e = exposes.presets;
const ea = exposes.access;

export default {
    fingerprint: [
        tuya.fingerprint("TS011F", ["_TZ3000_qeuvnohg", "_TZ3000_6l1pjfqe", "_TZ3000_2iiimqs9"]),
        tuya.fingerprint("TS0011", ["_TZ3000_gzvniqjb"]),
    ],
    model: "TS011F_din_smart_relay_polling",
    description: "Din smart relay (with power monitoring via polling)",
    vendor: "Tuya",
    fromZigbee: [
        fz.on_off,
        fz.electrical_measurement,
        fz.metering,
        fz.ignore_basic_report,
        tuya.fz.power_outage_memory,
        fz.tuya_relay_din_led_indicator,
    ],
    toZigbee: [tz.on_off, tuya.tz.power_on_behavior_1, tz.tuya_relay_din_led_indicator],
    whiteLabel: [
        tuya.whitelabel("Tongou", "TO-Q-SY1-JZT", "Din smart relay (with power monitoring via polling)", ["_TZ3000_qeuvnohg", "_TZ3000_gzvniqjb"]),
        tuya.whitelabel("TOMZN", "TOB9Z-63M", "Din smart relay (with power monitoring via polling)", ["_TZ3000_6l1pjfqe"]),
        tuya.whitelabel("Nous", "DZ", "DIN Switch (with power monitoring via polling)", ["_TZ3000_2iiimqs9"]),
    ],
    ota: true,
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ["genOnOff", "haElectricalMeasurement", "seMetering"]);
        await reporting.rmsVoltage(endpoint, {change: 5});
        await reporting.rmsCurrent(endpoint, {change: 50});
        await reporting.activePower(endpoint, {change: 10});
        await reporting.currentSummDelivered(endpoint);
        endpoint.saveClusterAttributeKeyValue("haElectricalMeasurement", {
            acCurrentDivisor: 1000,
            acCurrentMultiplier: 1,
        });
        endpoint.saveClusterAttributeKeyValue("seMetering", {
            divisor: 100,
            multiplier: 1,
        });
        device.save();
    },
    exposes: [
        e.switch(),
        e.power(),
        e.current(),
        e.voltage(),
        e.energy(),
        e.enum("power_outage_memory", ea.ALL, ["on", "off", "restore"]).withDescription("Recover state after power outage"),
        e.enum("indicator_mode", ea.STATE_SET, ["off", "on_off", "off_on"]).withDescription("Relay LED indicator mode"),
    ],
    options: [exposes.options.measurement_poll_interval()],
    onEvent: (type, data, device, options) => tuya.onEventMeasurementPoll(type, data, device, options, true, false),
};
```

Patbonamy, 2025-05-22:
Thanks for your quick response.
But I don't know where to put the file in home assistant. Do I have to restart HA ? Do I have to delete my device? And paire it again?
Thanks for your help.

Patbonamy, 2025-05-22:
Sorry I close by mistake

Patbonamy, 2025-05-22:
I put the files in config/zigbee2mqtt/external_converters/
Restart HA
Delete the device, pair it again.
Same problem

Chartreusito, 2025-05-23:
> I put the files in config/zigbee2mqtt/external_converters/ Restart HA Delete the device, pair it again. Same problem

You're missing a directory in the path. It now expects config/zigbee2mqtt/data/external_converters as mentioned in the doc link stated above

Patbonamy, 2025-05-23:
> > I put the files in config/zigbee2mqtt/external_converters/ Restart HA Delete the device, pair it again. Same problem
> 
> You're missing a directory in the path. It now expects config/zigbee2mqtt/data/external_converters as mentioned in the doc link stated above

I put the file in the correct folder same problem.

chris-1243, 2025-05-23:
@Patbonamy Could you restart Z2M and provide the log of the startup sequence ?

Patbonamy, 2025-05-23:
> [@Patbonamy](https://github.com/Patbonamy) Could you restart Z2M and provide the log of the startup sequence ?

Here it is:

[2025-05-23 14:38:20] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-05-23 14:38:20] info: 	z2m: Disconnecting from MQTT server
[2025-05-23 14:38:20] info: 	z2m: Stopping zigbee-herdsman...
[2025-05-23 14:38:21] info: 	zh:controller: Wrote coordinator backup to '/config/zigbee2mqtt/coordinator_backup.json'
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   Total frames: RX=330, TX=421
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   DATA frames : RX=330, TX=91
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   DATA bytes  : RX=9534, TX=2458
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=330
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2025-05-23 14:38:21] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2025-05-23 14:38:21] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2025-05-23 14:38:21] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2025-05-23 14:38:21] info: 	z2m: Stopped zigbee-herdsman
[2025-05-23 14:38:21] info: 	z2m: Stopped Zigbee2MQTT
[14:38:21] INFO: Preparing to start...
[14:38:21] INFO: Socat not enabled
[14:38:21] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-05-23 14:38:22] info: 	z2m: Logging to console, file (filename: log.log)
[2025-05-23 14:38:22] info: 	z2m: Starting Zigbee2MQTT version 2.3.0 (commit #unknown)
[2025-05-23 14:38:22] info: 	z2m: Starting zigbee-herdsman (4.0.0)
[2025-05-23 14:38:22] info: 	zh:ember: Using default stack config.
[2025-05-23 14:38:22] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-05-23 14:38:22] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-05-23 14:38:22] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-05-23 14:38:22] info: 	zh:ember:uart:ash: Socket ready
[2025-05-23 14:38:22] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-05-23 14:38:24] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2025-05-23 14:38:24] info: 	zh:ember:uart:ash: ======== ASH started ========
[2025-05-23 14:38:24] info: 	zh:ember:ezsp: ======== EZSP started ========
[2025-05-23 14:38:24] info: 	zh:ember: Adapter EZSP protocol version (14) lower than Host. Switched.
[2025-05-23 14:38:25] info: 	zh:ember: Adapter version info: {"ezsp":14,"revision":"8.0.2 [GA]","build":397,"major":8,"minor":0,"patch":2,"special":0,"type":170}
[2025-05-23 14:38:25] info: 	zh:ember: [STACK STATUS] Network up.
[2025-05-23 14:38:25] info: 	zh:ember: [INIT TC] Adapter network matches config.
[2025-05-23 14:38:25] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1246ms until next broadcast.
[2025-05-23 14:38:25] info: 	z2m: zigbee-herdsman started (resumed)
[2025-05-23 14:38:25] info: 	z2m: Coordinator firmware version: '{"meta":{"build":397,"ezsp":14,"major":8,"minor":0,"patch":2,"revision":"8.0.2 [GA]","special":0,"type":170},"type":"EmberZNet"}'
[2025-05-23 14:38:25] info: 	z2m: BTZ2 (0xa4c138991c3f8b70): TS0044 - Tuya Wireless switch with 4 buttons (EndDevice)
[2025-05-23 14:38:25] info: 	z2m: BTZ1 (0xa4c1380a11688be3): SH-SC07 - Tuya Button scene switch (EndDevice)
[2025-05-23 14:38:25] info: 	z2m: CAP2Z Bureau (0x847127fffea847a6): TS0201 - Tuya Temperature & humidity sensor with display (EndDevice)
[2025-05-23 14:38:25] info: 	z2m: ZR2 Répéteur SDB (0x943469fffe542d62): TS0207_repeater - Tuya Repeater (Router)
[2025-05-23 14:38:25] info: 	z2m: CAP13Z Chambre 2 (0xa4c138b6317c1d36): TS0201 - Tuya Temperature & humidity sensor with display (EndDevice)
[2025-05-23 14:38:25] info: 	z2m: CAP11Z Hub Salon (0xa4c13834b49e675b): ZTH05 - Tuya Temperature and humidity sensor (EndDevice)
[2025-05-23 14:38:25] info: 	z2m: CAP12Z Douche (0xa4c1380560bde62f): ZTH05 - Tuya Temperature and humidity sensor (EndDevice)
[2025-05-23 14:38:25] info: 	z2m: ZR1 Répéteur Chambre 2 (0x943469fffe54557e): TS0207_repeater - Tuya Repeater (Router)
[2025-05-23 14:38:25] info: 	z2m: VR1 Chambre (0xa4c1382d246505a1): TRV06_1 - AVATTO Thermostatic radiator valve (EndDevice)
[2025-05-23 14:38:25] info: 	z2m: SWZ1 Entrée (0x842712fffe38a28b): ZBMINIL2 - SONOFF Zigbee smart switch (no neutral) (EndDevice)
[... cut ...]
[2025-05-23 14:38:25] info: 	z2m: SDZ1 (0xa4c1383270118d9b): TS0011 - Tuya Smart light switch - 1 gang (EndDevice)
[2025-05-23 14:38:25] info: 	z2m: VR2 Salon (0xa4c138e9a5504352): TRV06_1 - AVATTO Thermostatic radiator valve (EndDevice)
[2025-05-23 14:38:25] info: 	z2m: SPZ2 (0x70b3d52b60138c06): TS011F_plug_3 - Tuya Smart plug (with power monitoring by polling) (Router)
[2025-05-23 14:38:25] info: 	z2m: Currently 23 devices are joined.
[2025-05-23 14:38:25] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2025-05-23 14:38:26] info: 	z2m: Connected to MQTT server

chris-1243, 2025-05-23:
Your log is not complete.  I need to check after this sequence. Could you share more?

Patbonamy, 2025-05-23:
I put the file here; /homeassistant/zigbee2mqtt/external_converters/
And I have this in the log: [2025-05-23 15:46:01] info: 	z2m: Loaded external converter 'sdz1.mjs'.

Chartreusito, 2025-05-23:
> I put the file here; /homeassistant/zigbee2mqtt/external_converters/ And I have this in the log: [2025-05-23 15:46:01] info: z2m: Loaded external converter 'sdz1.mjs'. But no change for my device

Your path is still missing a "/data" subdirectory.

Patbonamy, 2025-05-23:
> > I put the file here; /homeassistant/zigbee2mqtt/external_converters/ And I have this in the log: [2025-05-23 15:46:01] info: z2m: Loaded external converter 'sdz1.mjs'. But no change for my device
> 
> Your path is still missing a "/data" subdirectory.

When I put in a "/data" subdirectory z2m don't load it...

chris-1243, 2025-05-23:
Try to force remove your device.

Then restart Z2M, check your log to ensure the converter is correctly loaded. Pair your device again and keep us informed

chris-1243, 2025-05-23:
@Patbonamy I might have done a small error in my previous converter. Try this one please

```javascript
import * as tuya from "zigbee-herdsman-converters/lib/tuya";
import * as fz from "zigbee-herdsman-converters/converters/fromZigbee";
import * as tz from "zigbee-herdsman-converters/converters/toZigbee";
import * as exposes from "zigbee-herdsman-converters/lib/exposes";
import * as reporting from "zigbee-herdsman-converters/lib/reporting";

const e = exposes.presets;
const ea = exposes.access;

export default {
    fingerprint: [
        ...tuya.fingerprint("TS011F", [
            "_TZ3000_qeuvnohg",
            "_TZ3000_6l1pjfqe",
            "_TZ3000_2iiimqs9",
        ]),
        ...tuya.fingerprint("TS0011", ["_TZ3000_gzvniqjb"]),
    ],
    model: "TS011F_din_smart_relay_polling",
    description: "Din smart relay (with power monitoring via polling)",
    vendor: "Tuya",
    fromZigbee: [
        fz.on_off,
        fz.electrical_measurement,
        fz.metering,
        fz.ignore_basic_report,
        tuya.fz.power_outage_memory,
        fz.tuya_relay_din_led_indicator,
    ],
    toZigbee: [tz.on_off, tuya.tz.power_on_behavior_1, tz.tuya_relay_din_led_indicator],
    whiteLabel: [
        tuya.whitelabel("Tongou", "TO-Q-SY1-JZT", "Din smart relay (with power monitoring via polling)", ["_TZ3000_qeuvnohg", "_TZ3000_gzvniqjb"]),
        tuya.whitelabel("TOMZN", "TOB9Z-63M", "Din smart relay (with power monitoring via polling)", ["_TZ3000_6l1pjfqe"]),
        tuya.whitelabel("Nous", "DZ", "DIN Switch (with power monitoring via polling)", ["_TZ3000_2iiimqs9"]),
    ],
    ota: true,
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ["genOnOff", "haElectricalMeasurement", "seMetering"]);
        await reporting.rmsVoltage(endpoint, {change: 5});
        await reporting.rmsCurrent(endpoint, {change: 50});
        await reporting.activePower(endpoint, {change: 10});
        await reporting.currentSummDelivered(endpoint);
        endpoint.saveClusterAttributeKeyValue("haElectricalMeasurement", {
            acCurrentDivisor: 1000,
            acCurrentMultiplier: 1,
        });
        endpoint.saveClusterAttributeKeyValue("seMetering", {
            divisor: 100,
            multiplier: 1,
        });
        device.save();
    },
    exposes: [
        e.switch(),
        e.power(),
        e.current(),
        e.voltage(),
        e.energy(),
        e.enum("power_outage_memory", ea.ALL, ["on", "off", "restore"]).withDescription("Recover state after power outage"),
        e.enum("indicator_mode", ea.STATE_SET, ["off", "on_off", "off_on"]).withDescription("Relay LED indicator mode"),
    ],
    options: [exposes.options.measurement_poll_interval()],
    onEvent: (type, data, device, options) => tuya.onEventMeasurementPoll(type, data, device, options, true, false),
};
```

chris-1243, 2025-05-23:
> Model
TO-Q-SY1-JZT

This what you specified in your first post. Try to be more precise next time.

> My Device is TO-Q-SY1-ZT

No metering on this device. Let's try a new converter....

chris-1243, 2025-05-23:
This should do the job...

```javascript
import * as tuya from "zigbee-herdsman-converters/lib/tuya";
import * as fz from "zigbee-herdsman-converters/converters/fromZigbee";
import * as tz from "zigbee-herdsman-converters/converters/toZigbee";
import * as exposes from "zigbee-herdsman-converters/lib/exposes";
import * as reporting from "zigbee-herdsman-converters/lib/reporting";

const e = exposes.presets;
const ea = exposes.access;

export default {
    fingerprint: [
        ...tuya.fingerprint("TS011F", ["_TZ3000_7issjl2q"]),
        ...tuya.fingerprint("TS0011", ["_TZ3000_gzvniqjb"]),
    ],
    model: "ATMS1601Z",
    description: "Din smart relay (without power monitoring)",
    vendor: "Tuya",
    fromZigbee: [
        fz.on_off,
        fz.ignore_basic_report,
        tuya.fz.power_outage_memory,
        fz.tuya_relay_din_led_indicator,
    ],
    toZigbee: [
        tz.on_off,
        tuya.tz.power_on_behavior_1,
        tz.tuya_relay_din_led_indicator,
    ],
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ["genOnOff"]);
        device.save();
    },
    whiteLabel: [tuya.whitelabel("Tongou", "TO-Q-SY1-ZT", "Din smart relay (without power monitoring)", ["_TZ3000_gzvniqjb"])],
    exposes: [
        e.switch(),
        e.enum("power_outage_memory", ea.ALL, ["on", "off", "restore"]).withDescription("Recover state after power outage"),
        e.enum("indicator_mode", ea.STATE_SET, ["off", "on_off", "off_on"]).withDescription("Relay LED indicator mode"),
    ],
};
```

Patbonamy, 2025-05-23:
Thank you Chris-1234.
All is OK for me
Have a nice evening.

closed 2025-05-23 by Patbonamy

Patbonamy, 2025-05-23:
> TO-Q-SY1-JZT is already supported

Perfect.
Thanks